libusb/hidapi #190: Travis CI integration issue

Youw, 2020-09-27:
hidapi Travis builds are passing, e.g.: [image]
but the build results are not getting back to Github, e.g.: https://github.com/libusb/hidapi/pull/189 
[image]

Probably something wrong with Travis settings, or something.

Youw, 2020-09-27:
@z3ntu do you happen to remember who is the owner of Travis builds configuration?

Youw, 2020-09-27:
> @z3ntu do you happen to remember who is the owner of Travis builds configuration?

I guess it doesn't really matter.

As an Admin of libusb/hidapi I pretty much can configure Travic-CI build itself.
The issue we face is well-known: https://travis-ci.community/t/known-issue-travis-ci-reports-expected-waiting-for-status-to-be-reported-on-the-github-status-api-but-the-status-never-arrives/1154

Just need to migrate Travis integration from Github Hooks to Github Apps.
Already created a request. @LudovicRousseau would you be able to approve?

LudovicRousseau, 2020-09-27:
Travis-CI should have Read and Write access to libusb/hidapi

Youw, 2020-09-27:
Yeap, everything seems to be working. Thanks.

closed 2020-09-27 by Youw