libusb/hidapi #197: Delay in Windows 10 build 2004

miguel-r, 2020-10-09:
Hi all,

After updating to Windows 10 build 2004, I'm observing a delay of about 1 second from moving the joystick until I see it on the output. Checking on Windows's joy.cpl, there's no delay.

I'm using this happens both with the latest release and cdc473df commit. My program is in Qt C++ compiling with MSVC 32 bit and the device I'm using is an U-HID.

If I reduce the sleep time, the rate of messages increases, but the delay is still there. I have 2 identical Windows computers  running exactly the same code with different outcomes. If I downgrade Windows on the 2004 and don't connect it to the internet, it works fine too, so it seems that is linked with a Windows driver, but I wanted it to work with the latest driver.

Any suggestions on how to figure out the problem? Snippet of the code below.

Thank you,

```
  pDev = hid_open_path(pDevice->path);
  if (!pDev)
    printf("Cannot open HMI device VID:%d PID:%d DEV:%d\n", m_vid, m_pid, m_nDev);
  else
  {
    qDebug() << "Opened HMI device";
    hid_set_nonblocking(pDev, 1);
    break;
  }

  while (IsActive()) {
    if (pDev) {
      memset(data, 0, 64);
      int r = hid_read(pDev, data, 64);
      QString s, t;
      if (r > 0) {
        for (int i = 0; i < r; i++) {
          t.sprintf("%02x", data[i]);
          s += t;
        }
        qDebug() << "Output: " QString::number(r) << s;
        /* parsing code */
      }
    }
    SleepFor(10ms);
  }
```

Youw, 2020-10-09:
This is interesting change in behavior.
Cannot check it deeply, at least because I don't have a machine that can be upgraded to build 2004.

After a quick overview, first thing that comes to my mind: can you confirm, that your device/joystick writes data in chunks of size exactly 64 bytes?

Try using `hid_get_input_report` instead of `hid_read`. Does it work? Does it give same result/issue?

miguel-r, 2020-10-12:
Thank you for the quick feedback.

`hid_read` returns 21 bytes, I've changed the 3rd parameter to 21 and the result is the same (delay).

I've tried `hid_get_input_report `with length argument set to 22, the function returns 9. Printing data array, it's all zeroes.

Any other suggestions?

Thank you,

Youw, 2020-11-24:
Nope, sorry, I don't have any more ideas.

gre, 2020-11-24:
No clue if it is due to this, but is there a possibility that upgrading hidapi would help on the matter?
I see in https://github.com/libusb/hidapi/releases/tag/hidapi-0.10.0 that they mention "Fixed a sometime hang in Windows hid_read;" 🤔

Youw, 2020-11-24:
Shurely you should try, if you haven't already.

mcuee, 2021-06-19:
@gre Any updates on your issue? Is it still there with the latest git?

mcuee, 2021-07-13:
Close this issue since no inputs from the OP for more than 6 months. Please reopen when you have more info.

closed 2021-07-13 by mcuee