libusb/hidapi #199: FD 0 is a valid file descriptor

mzyngier, 2020-10-15:
I may have reported this issue against a stale repository, so here it is again:

https://github.com/signal11/hidapi/issues/462

Short story: File descriptor 0 is perfectly valid, and shouldn't be considered as an error.

https://github.com/libusb/hidapi/blob/d2c3a9862eefe2d3f4db0f00c0238277bfb4e44b/linux/hid.c#L668-L671

z3ntu, 2020-10-15:
Relevant part from `open(2)`:

> RETURN VALUE
>        open(), openat(), and creat() return the new file descriptor (a nonnegative integer), or -1 if an error occurred (in which case, errno is set appropriately).

Youw, 2020-10-15:
> (a nonnegative integer)

so I guess 0 is a nonnegative integer, looks valid

@mzyngier would you care to submit a PR?

PR #200 (2020-10-20): linux/hidraw: Don't consider open returning 0 as an error
The open systcall is allowed to return 0 as a valid file descriptor,
and considering it an error breaks environments where unnecessary
fds are closed before executing untrusted processes.

Fixes: #199
commit ad985aea (2020-10-20): linux/hidraw: Don't consider open returning 0 as an error

The open systcall is allowed to return 0 as a valid file descriptor,
and considering it an error breaks environments where unnecessary
fds are closed before executing untrusted processes.

Fixes: #199
closed 2020-10-20 by Youw