libusb/hidapi #201: Android Close two hid devices crash

VansonChung, 2020-10-19:
I successfully opened two hid devices using hid_init/hid_open, but when I close it (hid_close), it will crash.

hid.c

```
static void *read_thread(void *param) {
    xxxxxxxx
    ...
    ...
    while (!dev->cancelled) {
        libusb_handle_events_completed(usb_context, &dev->cancelled); // crash
    }
}
```

Then I found a temporary solution

```
void HID_API_EXPORT hid_close(hid_device *dev) {
    xxxxxxxxx
    ...
    ...
    dev->cancelled = 1; // Add this
}
```
Is there a better solution?

Youw, 2020-10-19:
first - try latest master
if the issues is still there - we'll discuss more

VansonChung, 2020-10-20:
I try the latest master, it still crash.

```
static void *read_thread(void *param) {
    ...
    ...
    ...
    while (!dev->transfer_loop_finished)
        libusb_handle_events_completed(usb_context, &dev->transfer_loop_finished); // crash
}
```

Youw, 2020-10-20:
Of course it is.
You're not supposed to `hid_exit` until you close the last device.

For simplicity, it is considered OK the next approach:
```cpp
int main() {
    hid_init();

    // your entire application logic

    // at the very end
    hid_exit();
}
```

closed 2020-10-20 by Youw

VansonChung, 2020-10-21:
I encapsulated hid_api into a class (SDK), so the user only needs to new class and use the related API. As I said before, I can’t know how many devices users want to use (new many class), so I must do hid_exit when class destroy.

Youw, 2020-10-21:
But you can.
Make a global shared object, that is being destroyed, when destroyed the last instance of device object is destroyed.

VansonChung, 2020-10-21:
Yes, but this can only be implemented by users layer, not in SDK layer.
I will tell the user to do this special treatment.
Thanks !

Youw, 2020-10-21:
For example:
```cpp
// hid_device.h

#include <memory>

class HidContext;

std::shared_ptr<HidContext> getGlobalContext();

class HidDevice {
public:
    HidDevice();
    ~HidDevice();

    // other member functions

private:
    // because of this member, clients of this class don't have to do anything about context manually
    std::shared_ptr<HidContext> context{getGlobalContext()};
    // other members
};

// hid_device.cpp
#include "hid_device.h"

#include <mutex>

class HidContext {
public:
    HidContext() {
        ::hid_init();
    }
    ~HidContext() {
        ::hid_exit();
    }
};

// "lazy" singleton - object exists only when it is used
std::shared_ptr<HidContext> getGlobalContext() {
    // this two variables may be moved to global namespace if so desired
    static std::weak_ptr<HidContext> global_weak_context;
    static std::mutex global_weak_context_mutex;

    auto result = global_weak_context.lock();
    if (!result) {
        const std::lock_guard lck(global_weak_context_mutex);
        result = global_weak_context.lock();
        if (!result) {
            result = std::make_shared<HidContext>();
            global_weak_context = result;
        }
    }
    return result;
}

HidDevice::HidDevice() {
    // hid_open/hid_open_path
}

HidDevice::~HidDevice() {
    // hid_close
}
```

VansonChung, 2020-10-21:
I know, but it still need **user space** to take care.
HidDevice is SDK part.
For **user space**, maybe will new 2 HidDevice class (A, B)
For this case, global shared object can only handle and judged in the **user space**, not in SDK part, so **user space** need take care about shared_ptr.
The above is my understanding (My C++ is poor...)

Youw, 2020-10-21:
take another look on the `context` member of `HidDevice`
it has to be the first member declared in the class, and it takes care of everything for the user
check here: https://onlinegdb.com/HJIAjs6vP

VansonChung, 2020-10-21:
Sorry, I explain my situation clearly

I have a C++ class HidDevice.cpp and I package in a android class HID.java, then make it as .so or aar
So for android user space part.
HID a = new HID(xxx, xxx)
HID b = new HID(xxx, xxx)
and something like a.init(xxx) -> a.open(xxxx), b.init(xxx) -> b.open(xxxx)
https://onlinegdb.com/HJIAjs6vP still workable as my case ?

Youw, 2020-10-21:
don't know how exactly do you package C++ class under Java class, but if my guess is correct, and `a/b.init` creates an instance of the `HidDevice` C++ class, and later you have a/b.deinit() or similar, then yes, it will work perfectly fine

VansonChung, 2020-10-21:
a/b.init will create two  instance of HidDevice  C++ class, so I think it will not work.
I will try it. Thanks a lot.