libusb/hidapi #208: Segmentation fault on Mac OS Catalina

jonasmalacofilho, 2020-11-13:
An issue was reported in liquidctl that appears to come from hidapi.

As mentioned in jonasmalacofilho/liquidctl#216, executing liquidctl in the context of a global agent, during boot, results in a segmentation fault in `hid_enumerate`.  The issue does *not* reproduce in other contexts, like a regular terminal shell.

As the crash report bellow shows, the segfault is triggered in `CFSetGetCount`, which makes me think that issue is in the set returned by `IOHIDManagerCopyDevices`, in [`libusb/hidapi/mac/hid.c#L585-L588`](
https://github.com/libusb/hidapi/blob/8ed36853ba08d82ebce3bd27f7c510db87fa7b0a/mac/hid.c#L585-L588).  While not clearly documented behavior, perhaps it's necessary to check for `NULL`s?

For reference, the issue has been observed with hidapi 0.10.0 (302d1f24018d5a4a480f92a72307616f88d97651), being used from Python with cython-hidapi.

```
Process:               Python [253]
Path:                  /usr/local/Cellar/python@3.9/3.9.0_1/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               3.9.0 (3.9.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [96]
User ID:               0

Date/Time:             2020-11-11 15:50:10.735 -0500
OS Version:            Mac OS X 10.15.7 (19H2)
Report Version:        12
Anonymous UUID:        E20FAF83-55E4-C66A-3D0A-658B42A1F320

Time Awake Since Boot: 17 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [253]

VM Regions Near 0:
--> 
    __TEXT                 0000000105d6d000-0000000105d71000 [   16K] r-x/r-x SM=COW  /usr/local/Cellar/python@3.9/3.9.0_1/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x00007fff36f2fbb4 CFSetGetCount + 6
1   hid.cpython-39-darwin.so      	0x0000000106676943 hid_enumerate + 313
2   hid.cpython-39-darwin.so      	0x00000001066761ab __pyx_pw_3hid_1enumerate + 437
3   org.python.python             	0x0000000105de2487 cfunction_call + 122
4   org.python.python             	0x0000000105db1b47 _PyObject_MakeTpCall + 266
5   org.python.python             	0x0000000105e59bd7 call_function + 455
6   org.python.python             	0x0000000105e56d34 _PyEval_EvalFrameDefault + 27452
7   org.python.python             	0x0000000105dc00c5 gen_send_ex + 269
8   org.python.python             	0x0000000105e56629 _PyEval_EvalFrameDefault + 25649
9   org.python.python             	0x0000000105dc00c5 gen_send_ex + 269
10  org.python.python             	0x0000000105e56629 _PyEval_EvalFrameDefault + 25649
11  org.python.python             	0x0000000105dc00c5 gen_send_ex + 269
12  org.python.python             	0x0000000105dc967d list_extend + 169
13  org.python.python             	0x0000000105dccf5f list___init___impl + 157
14  org.python.python             	0x0000000105dca918 list_vectorcall + 121
15  org.python.python             	0x0000000105e59ba3 call_function + 403
16  org.python.python             	0x0000000105e56de0 _PyEval_EvalFrameDefault + 27624
17  org.python.python             	0x0000000105db21ec function_code_fastcall + 97
18  org.python.python             	0x0000000105e59ba3 call_function + 403
19  org.python.python             	0x0000000105e56de0 _PyEval_EvalFrameDefault + 27624
20  org.python.python             	0x0000000105e5a6eb _PyEval_EvalCode + 1998
21  org.python.python             	0x0000000105e5011d PyEval_EvalCode + 79
22  org.python.python             	0x0000000105e8b185 run_eval_code_obj + 110
23  org.python.python             	0x0000000105e8a57d run_mod + 103
24  org.python.python             	0x0000000105e89441 PyRun_FileExFlags + 241
25  org.python.python             	0x0000000105e88a31 PyRun_SimpleFileExFlags + 271
26  org.python.python             	0x0000000105ea094d Py_RunMain + 1839
27  org.python.python             	0x0000000105ea0c86 pymain_main + 306
[... 298 lines cut ...]
===========                     =======  ======= 
Activity Tracing                   256K        1 
Dispatch continuations            32.0M        1 
Kernel Alloc Once                    8K        1 
MALLOC                           114.1M       29 
MALLOC guard page                   24K        4 
MALLOC_MEDIUM (reserved)         360.0M        3         reserved VM address space (unallocated)
STACK GUARD                         12K        3 
Stack                             17.0M        3 
VM_ALLOCATE                       5632K       22 
__DATA                            22.3M      258 
__DATA_CONST                       328K       20 
__FONT_DATA                          4K        1 
__LINKEDIT                       389.4M       36 
__OBJC_RO                         32.3M        1 
__OBJC_RW                         1908K        2 
__TEXT                           157.0M      256 
__UNICODE                          564K        1 
shared memory                       12K        3 
===========                     =======  ======= 
TOTAL                              1.1G      645 
TOTAL, minus reserved VM space   772.6M      645 
```

Related: trezor/cython-hidapi#100.

Youw, 2020-11-14:
> While not clearly documented behavior, perhaps it's necessary to check for NULLs?

Agree, not clearly documented.
Agree, checking for NULL is a good idea. Would you care to make a PR?

As for the original issue, as reported for liquidctl - it won't really help. It will (likely) eliminate the crash, but in that context there will be no usable devices...

jonasmalacofilho, 2020-11-14:
> Agree, checking for NULL is a good idea. Would you care to make a PR?

Sure.

> As for the original issue, as reported for liquidctl - it won't really help. It will (likely) eliminate the crash, but in that context there will be no usable devices...

Yeap!

... which is weird, since `IOHIDManagerCopyDevices` appears to fail only at that time, and only in that machine.  AFAIK other users are running the same boot/global agent configuration without issues.

Do you have any idea of what could explain this?

Youw, 2020-11-14:
In my experience - that boot/global agent configuration without is not supposed to work at all :)

I have an application, that runs as an agent, and it starts seeing HID devices only after I login with my user (so there is an active user UI session).
At least I think so....
That was almost 2 years ago, and there is definitely some restrictions regarding access to HID/USB devices from an agent that starts before UI session.
Maybe the executable/script that is started by a deamon has to be owned by root (at least that's what I did in my project, even though I don't remember was it nesessary for it or not).

jonasmalacofilho, 2020-11-14:
> ... starts seeing HID devices only after I login with my user ...

Oh, I see...

> Maybe the executable/script that is started by a deamon has to be owned by root ...

My current instructions do say to start a daemon that's owned by root, but I didn't check if that's exactly what the user is doing.  Thanks for the hint!

Youw, 2020-11-23:
Fixed by https://github.com/libusb/hidapi/pull/209

closed 2020-11-23 by Youw

dumski, 2022-11-02:
Hi, sorry to comment on closed issue but I think I might be a regression bug. For me hid_enumerate still causes SEGFAULT sometime. My debugger lead me to this line: https://github.com/libusb/hidapi/blob/bb792a1f7eb7e9dff37d8e538219cc0a00207bf8/mac/hid.c#L653

I think sometimes `matching` variable os NULL, and I see there's no additional condition to match this situation. I suggest adding moving the `IOHIDManagerSetDeviceMatching(hid_mgr, matching);` call inside next scope to cover such case.

```
	if (matching != NULL) {
		IOHIDManagerSetDeviceMatching(hid_mgr, matching);
		CFRelease(matching);
	}
```

It works for me. What do you think?
Teo

Youw, 2022-11-03:
While this is distantly related to this one, I believe what you experience is a different issue.
Please open a separate issue, and lets continue the discussion there.