libusb/hidapi #211: enumerate returns wrong information on mac

jkelleyrtp, 2020-11-18:
`enumerate` returns a list of devices that I can't find using any other tools.

running 
```
❯ ioreg -p IOUSB -w0                                                                                       ─╯
```
returns
```
+-o Root  <class IORegistryEntry, id 0x100000100, retain 22>
  +-o AppleUSBVHCIBCE Root Hub Simulation@80000000  <class AppleUSBRootHubDevice, id 0x100000501, registered, matched, active, busy 0 (0 ms), retain 16>
  | +-o IOUSBHostDevice@80800000  <class AppleUSBDevice, id 0x100000503, registered, matched, active, busy 0 (0 ms), retain 11>
  | +-o Apple T2 Controller@80100000  <class AppleUSBDevice, id 0x100000507, registered, matched, active, busy 0 (0 ms), retain 13>
  | +-o FaceTime HD Camera (Built-in)@80200000  <class AppleUSBDevice, id 0x10000050c, registered, matched, active, busy 0 (0 ms), retain 14>
  | +-o Headset@80400000  <class AppleUSBDevice, id 0x100000511, registered, matched, active, busy 0 (0 ms), retain 11>
  | +-o Touch Bar Display@80600000  <class AppleUSBDevice, id 0x100000516, registered, matched, active, busy 0 (0 ms), retain 13>
  | +-o Ambient Light Sensor@80300000  <class AppleUSBDevice, id 0x100000520, registered, matched, active, busy 0 (0 ms), retain 11>
  | +-o Touch Bar Backlight@80700000  <class AppleUSBDevice, id 0x100000524, registered, matched, active, busy 0 (0 ms), retain 13>
  | +-o Apple Internal Keyboard / Trackpad@80500000  <class AppleUSBDevice, id 0x10001a06c, registered, matched, active, busy 0 (7 ms), retain 19>
  +-o AppleUSBXHCI Root Hub Simulation@01000000  <class AppleUSBRootHubDevice, id 0x100002c65, registered, matched, active, busy 0 (0 ms), retain 9>
  | +-o USB3.1 Hub@01100000  <class AppleUSBDevice, id 0x100021743, registered, matched, active, busy 0 (0 ms), retain 11>
  +-o AppleUSBXHCI Root Hub Simulation@14000000  <class AppleUSBRootHubDevice, id 0x100002c6e, registered, matched, active, busy 0 (0 ms), retain 9>
    +-o USB2.0 Hub@14100000  <class AppleUSBDevice, id 0x100021752, registered, matched, active, busy 0 (0 ms), retain 13>
      +-o USB-C Digital AV Multiport Adapter@14120000  <class AppleUSBDevice, id 0x100021761, registered, matched, active, busy 0 (2 ms), retain 11>
      +-o Bluefruit nRF52840 DFU@14110000  <class AppleUSBDevice, id 0x100022ef1, registered, matched, active, busy 0 (3 ms), retain 15>
```

I'm trying to connect to the nRF52840. It seems that `enumerate` 's list just doesn't make sense.

Here's the VIDs and PIDs for everything that comes out of enumerate:
```
1452 34304
1452 832 Apple Internal Keyboard / Trackpad
1452 832 Apple Internal Keyboard / Trackpad
1452 33026 Touch Bar Backlight
1452 33026 Touch Bar Backlight
1452 832 Apple Internal Keyboard / Trackpad
1452 832 Apple Internal Keyboard / Trackpad
1452 832 Apple Internal Keyboard / Trackpad
1452 832 Apple Internal Keyboard / Trackpad
1452 33538 Touch Bar Display
1452 33538 Touch Bar Display
1452 33538 Touch Bar Display
1452 832 Apple Internal Keyboard / Trackpad
1452 832 Apple Internal Keyboard / Trackpad
1452 33028
4096 6 ATH-M50xBT
1452 33378 Ambient Light Sensor
1452 33027 Headset
1133 50732 ConnexionHID
```

It doesn't seem to match the system profiler list:

[image]

Any thoughts on how to fix this?

todbot, 2020-11-18:
What specifically are you not seeing from hidapi? 
DFU is not HID.  Many HID devices have multiple usages, which show up as separate "devices" in hidapi.

todbot, 2020-11-18:
Oh I see you are looking for HF2 devices, not DFU.  From https://github.com/jacobrosenthal/hf2-rs/issues/34 it appears that HF2 isn't correctly identifying as a HID device.

jkelleyrtp, 2020-11-18:
Ah, it seems that HF2 doesn't define interface identification rules and HID isn't picking it up?

https://github.com/microsoft/uf2/issues/25

Is this something that HID should be concerned about, perhaps by adding ad-hoc support, or is the lack of spec around HF2 mean that HID won't support HF2 devices?

Youw, 2020-11-24:
> enumerate returns a list of devices that I can't find using any other tools.
> running
> ioreg -p IOUSB -w0

HID devices not necessary USB devices.

> Any thoughts on how to fix this?

There is nothing to fix here. VID/PID does not uniquely identifies a device.
VID + PID + SERIAL_NUMBER + USAGE + USAGE_PAGE does (in most cases).
Not all USB devices are HID devices, and some HID devices are not USB devices.
hidapi enumerates exactly what Operating System threats as a HID device (at least on macOS). Nothing more, nothing less.

closed 2020-11-24 by Youw