libusb/hidapi #212: MacOS Big Sur IOError on open

smartavionics, 2020-11-19:
Hi, I am the developer of a plugin (https://github.com/smartavionics/RawMouse) that lets people use HID mouse devices with the Cura slicer program. It makes use of https://github.com/trezor/cython-hidapi which then uses https://github.com/libusb/hidapi (thanks for providing that). It has all been working splendidly for many months but now a user has reported a problem on MacOS Big Sur. The RawMouse plugin is reporting this...

```
2020-11-18 11:46:33,570 - DEBUG - [RawMouse] RawMouse.RawMouse._run_hid [219]: Trying to open IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/PEG2@1,2/IOPP/UPSB@0/IOPP/DSB1@1/IOPP/UPS0@0/IOPP/pci-bridge@1/IOPP/pci1b73,1100@0/AppleUSBXHCIFL1100@00000000/AppleUSB20XHCIPort@00100000/USB2.1 Hub@00100000/AppleUSB20Hub@00100000/AppleUSB20HubPort@00130000/SpaceMouse Wireless@00130000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice
2020-11-18 11:46:33,571 - ERROR - [RawMouse] RawMouse.RawMouse._run_hid [246]: IOError while reading HID events: open failed
```

Is the hidapi library compatible with Big Sur? And if so, do I need to change the way it is being used? Is it a permissions thing?

Youw, 2020-11-19:
Mice are special devices, I wouldn't expect hidapi to be able to open one of those at all.
Can't say for sure what's wrong.

smartavionics, 2020-11-19:
Hi, I don't think it is actually seen as a mouse, rather it's a multi-axis controller? It does work on Linux/Windows/MacOS just fine but Big Sur has a problem.

Youw, 2020-11-19:
so it is not a "real" mouse? just "some" hid device?

smartavionics, 2020-11-19:
> so it is not a "real" mouse? just "some" hid device?

Correct, but the product is actually called SpaceMouse.

Youw, 2020-11-19:
Current hidapi implementation tries to open the device for exclusive access. Maybe there is another application/service, that opens that device.

smartavionics, 2020-11-19:
I have asked the person who reported the issue to see if the vendor's driver is active. I only use Linux and don't have the vendor's software installed. On Windows, the RawMouse plugin works irrespective of whether the 3dconnexion driver is active or not. Perhaps on MacOS, it needs to be disabled for the HID access to work. I know that people have been using RawMouse on MacOS up till now without problems.

Youw, 2020-11-19:
Also see:
https://github.com/libusb/hidapi/issues/27
https://github.com/libusb/hidapi/pull/148

mcuee, 2020-12-04:
FYI hidapi and libusb work fine under my Mac Mini M1 (Apple Silicon M1 chip, Big Sur). 

The issue here is that the exclusive access for device like a mouse under macOS.

mcuee, 2021-01-26:
From the original ticket we can see that there is a 3dconnexion driver enabled for this device and there is no way to disable it under macOS, in that case, HIDAPI can not be used.
https://github.com/smartavionics/RawMouse/issues/9

smartavionics, 2021-01-26:
So how come RawMouse was being used OK on MacOS before Big Sur arrived? Did it only work then because the users did not have the 3dconnexion driver enabled? I'm pretty sure that no one had mentioned that was necessary.

mcuee, 2021-01-26:
Yes that is the point. Please check with the user if they have 3dconnection driver prior to Big Sur.

mcuee, 2021-01-26:
https://3dconnexion.com/uk/support/faq/beta-driver-for-macos-11-big-sur/

It is possible the previous driver is not a kernel driver but the new one is a kernel driver.

smartavionics, 2021-01-26:
Understood. I am trying to get confirmation that it worked OK before without disabling the 3dconnexion driver.

smartavionics, 2021-01-26:
My user just replied...

>Yes I was using the RawMouse plugin also before BigSur and it just worked
fine without disabling the 3dconnexion driver. Unfortunately disabling the
driver in MacOS is not so easy, I can only deinstall the 3dconnection
driver package, but then I miss the functionality of it for Fusion360.

So I guess there's nothing that can be done here in hidapi to solve this problem.

mcuee, 2021-01-26:
Looks like either it is the change in 3dconnection driver or macOS Big Sur which changes the behavior. 

One way is to wait for 3dconnection to release the driver for Big Sur (right now it is under beta) to see if that has different behavior. Still all in all, there is not much hidapi can do as of now.

Youw, 2021-01-26:
> So I guess there's nothing that can be done here in hidapi to solve this problem.

I guess not. Two applications (regardless - a user applications, or a kernel driver) cannot access same HID device on macOS at the same time if at least one of them tries to open the device in exclusive mode.

You could try to apply https://github.com/libusb/hidapi/pull/148, even though this could be a long shoot.

smartavionics, 2021-01-26:
> You could try to apply #148, even though this could be a long shoot.

Thanks, I will do the experiment and let you know what happens.

smartavionics, 2021-01-26:
Sorry to report that my helper tried a build with hidapi not using exclusive mode and reports that it still doesn't work.

Youw, 2021-01-26:
That only means, that 3dconnection driver openes the device in exclusive mode too, and until it is active - no other application will able to access it.

closed 2021-06-10 by mcuee