libusb/hidapi #217: Where is hidapi-externals.zip?

chrisgr88, 2020-12-04:
This file is described in the documentation as being available "from the download site", presumably the same place where hidapi was downloaded, but it's nowhere on the github  or anywhere else I can find with a Google search.  Is the documention incorrect?  Where is it available?

Youw, 2020-12-04:
Hm, interesting.
Apparently, Github has deprecated the "Download" page, that used to exist. Now it is not reachable from Github (at least I'm not aware how to get there 'legally').

I gues Google loves me more, and I was able to find it here: https://github.com/signal11/hidapi/downloads

Just note: the chances, that it will work are minimal (check the TestUI-related issues list), and no one is willing to support that.

And yes, the readme is a bit not up-to-date.

chrisgr88, 2020-12-05:
Hi Ihor, thanks for the reply, and for the reference to the zip, and the
warning about the issues!  From the issues it looks like it will be
difficult to get working.  I see that someone had success with HIDPyToy so
I think I'll give that a try.

- Chris

mcuee, 2021-06-14:
Close this for now. Please use hidpytoy and hidapitester (command line).

mcuee, 2021-06-20:
Keep this open as people may still want to look for this.

mcuee, 2021-06-20:
BTW, I have tried the zip file with the latest VS2019 but it does not work.

Then I tried to build fox with VS2019 and it also failed. So I kind of give up.

mcuee, 2021-09-13:
MSYS2 MinGW build binary is okay -- please refer to #68.