libusb/hidapi #222: travis-ci.org -> travis-ci.com migration

Youw, 2020-12-12:
Apparently, the integration with travis-ci.org is broken again. The last build was 9 days ago, and no more builds are triggered since then.
Taking into account the fact, that travis-ci encourage developer to migrate from travis-ci._org_ to travis-ci._com_, I think better to put effort into migrating to new infra, instead of trying to fix the old one.

I've created request to libusb organisation to enable access from travis-ci._com_ to hidapi repo.

@LudovicRousseau I know that I'm bothering you regarding this matter from time to time, but only a libusb organisation administrator can approve the access request. Can you assist one more time?

todbot, 2020-12-12:
I've been having very good luck recently with Github Actions as a replacement for my uses of Travis CI, Appveyor, and Cirrus CI.

Just yesterday I added Github Actions CI to `hidapitester`: https://github.com/todbot/hidapitester/tree/master/.github/workflows

Maybe that could work here?  No need for extra logins on separate service.

Youw, 2020-12-12:
This sounds temping. It might just work out nicely.
I'll give it a try.

LudovicRousseau, 2020-12-13:
I approved the request.

PR #223 (2020-12-13): macOS: use GitHub action CI instead of Travis-CI
Closes: #222
closed 2020-12-13 by Youw