libusb/hidapi #230: hid_read() is there any way to return the correct length?

WanManito, 2021-01-05:
"hid_ Read () "is there any way to return the correct length? Like serial port and TCP communication, each read will return the actual length. Now the length is always 48, and there is a lot of irrelevant data.

Youw, 2021-01-05:
On which platform?

WanManito, 2021-01-05:
@Youw window10

Youw, 2021-01-05:
> Now the length is always 48

That's probably a "maximum length" of the input report. That's a current limitation of a Windows backend implementation.

WanManito, 2021-01-05:
[image]

I set 64 because I'm not sure how long to read. Is there a way to return the actual length

Youw, 2021-01-05:
Not as of today.

WanManito, 2021-01-08:
Is this problem considered to be solved in the next version? thank you

Youw, 2021-01-08:
Sure, if someone cares to prepare a PR.
But I cannot give any guarantee.

Youw, 2021-02-06:
I've checked the API/implementation again, and can confirm that on Windows, hidapi returns as accurate result as _possible_ for this platform.
In other words: WinAPI cannot give any better, nor hidapi.

The way hidapi is designed and originally purposed - if you want to communicate with HID device, you need to know protocol/specs of your device, including details like the size of the input/output report, in advance.

closed 2021-02-06 by Youw

JoergAtGithub, 2021-02-06:
Thanks for investigating! But doesn't the Windows API return the size of the HID caps? If you sum up the cap sizes per report, shouldn't you get the size of the report. Or I'm getting something wrong here?

Youw, 2021-02-06:
In theory, it really is possible to get capabilities of a specific report (its size) and _assume_ that successfully read report is (only) of size, as specified in capabilities.

By doing so, we would start doing something what we don't do for any other platform. hidapi is designed to be a thin proxy between a user and underlying backend with _unifor API_ on each platform.