libusb/hidapi #233: Missing `__x86_64__` in CFLAGS for Darwin (Unsupported architecture / architecture not supported)

Sup3rGeo, 2021-01-20:
Follow up for https://github.com/trezor/cython-hidapi/issues/96:

This causes build to fail on OSX with:

```
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/machine/limits.h:8:2: error:
      architecture not supported
#error architecture not supported
```

Addressed by the following change:

```
 if sys.platform.startswith('darwin'):
     macos_sdk_path = subprocess.check_output(['xcrun', '--show-sdk-path']).decode().strip()
-     os.environ['CFLAGS'] = '-isysroot "%s" -framework IOKit -framework CoreFoundation' % macos_sdk_path
+     os.environ['CFLAGS'] = ' -D__x86_64__ -isysroot "%s" -framework IOKit -framework CoreFoundation' % macos_sdk_path
```

> Especially with the upcoming macOS for arm64 arch. I think this should be fixed in hidapi upstream by including the correct include file

z3ntu, 2021-01-20:
How can this be reproduced? Neither the CI nor any other user seems to have experienced this.

z3ntu, 2021-01-20:
I've found https://github.com/ycm-core/YouCompleteMe/issues/3770#issuecomment-700841907 now, this doesn't look like a hidapi bug.

Sup3rGeo, 2021-01-20:
Sorry I am quite a noob in this and just wanted to move things forward, so I don't know too much.
The way I reproduce it is to run `pip install hidapi` in my macbook pro with Catalina.

Maybe @prusnak can help?

Youw, 2021-01-20:
I don't believe this is a hidapi issue.
hidapi provides its own autotools based build system (and a CMake build system work-in-progress, but not officially released).
The build system handles this kind of things.

What cython-hidapi does - it compiles hidapi source(s) on its own, using its own rules/tools, and apparently not all required compilation parameters are passed, which causes a build failure.

> this should be fixed in hidapi upstream by including the correct include file

If someone can suggest a specific additional include file for macOS backend (with motivation, please), I'd happily add it, to make things easier for those, who is using its own build system(s).

prusnak, 2021-01-20:
Okay, it seems this is indeed an issue within cython-hidapi. Feel free to close this for now. If I'll find something that needs to be sent upstream, I'll send a patch.

Youw, 2021-01-20:
Awsome, thanks.

closed 2021-01-20 by Youw