libusb/hidapi #234: Windows version doesn't enumerate devices when exactly one is connected

jm4R, 2021-01-20:
I don't know if this is coincidence, but this is my case:

I plug UPS HID device (CyberPower or APC) as an only HID device in the whole system, and `hid_enumerate` returns `nullptr`.

When I plug USB keyboard as a second device, `hid_enumerate` enumerates both - keyboard and UPS.

On Linux and exactly the same machine it works fine.

Youw, 2021-01-20:
I assume you're using build 0.10.1.

Can you step-trace a "single device" scenario with a debugger and tell what do you get from [SetupDiGetClassDevsA](https://github.com/libusb/hidapi/blob/master/windows/hid.c#L330)/[SetupDiEnumDeviceInterfaces](https://github.com/libusb/hidapi/blob/master/windows/hid.c#L339) (or after those)?

jm4R, 2021-01-20:
driver_name is "Battery" which is weird as in windows device manager the device class shows "HIDClass" as expected.

Youw, 2021-01-20:
And when you have two devices connected?

todbot, 2021-01-20:
Yes, I believe on Windows, with the HID Battery driver, that removes it from the list of devices hidapi can access.  (as what happens to gamepads) http://batcmd.com/windows/10/services/hidbatt/

todbot, 2021-01-20:
I just tested an APC Back-UPS ES 550 on Windows 10 Pro. It appears as both a HID device and as a "Battery" to device driver (see screenshot below).  And it appears to hidapi's `hid_enumerate` just fine (see `hidapitester` output below).  Perhaps you have an additional, non-Microsoft driver on the device?

[image]
[image]

Youw, 2021-01-20:
I think the problem comes if you'd have only this UPS connected, and no other HID devices in the system.

todbot, 2021-01-20:
Yes I understand that better now. :)  Trying to figure out how to test this since RDP just decided to stop working with a recent Win10 update...

jm4R, 2021-01-21:
@Youw when I plug the keyboard and the mouse in, all iterations returns "Keyborad" as a device name. Something is definetely wrong in this code. I found old, open but abandoned pull request that claims that the index passed to the `SetupDiEnumDeviceInfo` is wrong:

https://github.com/signal11/hidapi/pull/291/files/f453bda38ff17ea28b7b1d59afd8d09e74e5efb5#diff-8162c58f4630fb0f3a6a21f2e094e883L354

It seems that after simillar change the driver_name shows more apriopriate values to the ones that are shown in Windows Device Manager. To sumarise that, it looks like
**hid_enumerate enumerates all USB devices if the first on the list is HIDClass or Mouse or Keyboard**

This is coincidence that the code works with my UPS because `SetupDiEnumDeviceInfo(device_info_set, device_index, &devinfo_data);` after the "quickfix" still returns "Battery", but we should somehow retreive the value which is shown by the Windows device manager as on the screenshot:
[image]

This way, we can expect that we could get rid of the "Keyboard" and "Mouse" exceptions, because for those devices the device manager also shows "HIDClass" value. Yet, I am not very familiar with WinApi so I don't know (yet) how to retrive it using WinApi.

jm4R, 2021-01-21:
How about solution straight from Microsoft:

https://github.com/microsoft/Windows-driver-samples/blob/master/hid/hclient/pnp.c#L32

?

It is simmilar but instead of checking device class name, it just tries to open it. I am just not sure where it checks if it is a HID device, but it is a code straight from MS so worth taking a look.

Youw, 2021-01-21:
I'm afraid, opening a device might cause issue, like described here: https://github.com/libusb/hidapi/issues/225#issuecomment-751252471
I'll take a look at this closely

jm4R, 2021-01-21:
We already do this:
https://github.com/libusb/hidapi/blob/master/windows/hid.c#L411

It is being open with `open_rw` set to `FALSE`, maybe this doesn't make the issue. I won't change it.

Youw, 2021-01-21:
Hmm, right, fair enough.

jm4R, 2021-01-21:
The PR works for me, please review. I didn't use `HidD_GetHidGuid` directly as it would require linking to `hid.lib`. The hard-coded GUID is the same with the one returned by this function which means the enumeration already filters only HID-class devices.

PR #235 (2021-01-30): Windows: Fix devices enumeration
Based on
https://github.com/microsoft/Windows-driver-samples/blob/master/hid/hclient/pnp.c#L32

Fixes wrong detection of the class of devices.

Fixes: #234
commit 3cd17427 (2021-01-30): Windows: Fix devices enumeration (#235)

Based on
https://github.com/microsoft/Windows-driver-samples/blob/d40232638faaab19de557e70a3ee938da7a35295/hid/hclient/pnp.c#L32

Fixes wrong detection of the class of devices.

Fixes: #234
closed 2021-01-30 by Youw