libusb/hidapi #237: Drop Windows XP support

mcuee, 2021-01-26:
From https://github.com/libusb/hidapi/issues/133.
It may be a good idea to drop Windows XP support. libusb has done that for Windows.

Youw, 2021-01-26:
Maybe.
It just don't really solve anything in context of #133. #146 - does. I've described it in [details](https://github.com/libusb/hidapi/issues/133#issuecomment-567247547).

mcuee, 2021-01-26:
Yes it does not sort the issue of #133. 

But then it may still be a good idea to upgrade from CancelIo to CancelIoEx.

Youw, 2021-01-26:
What I afraid the most - it will hide the #133 issue deep enough so some users will start thinking that there is no issue.
I always favor an explicit crash/hang/error (so it is reproducible, can be easily debugged/fixed) instead of an "undefined behavior" that _may_ hit you one in a million times.

In other words: `hid_close` better not try to interrupt `hid_read` (explicitly or implicitly) unless it is going to wait for its completion (which is not the case for any of the backends).

mcuee, 2021-04-20:
In this case, I will close this one.

mcuee, 2021-06-14:
I will keep this open even if this does not help on #133. Probably this can be considered after #133.

mcuee, 2022-04-25:
I think it is better to close this as it is clear that hidapi will support old OS when there are no extra efforts.

https://github.com/libusb/hidapi/issues/376#issuecomment-1033756841
> > does it make sense to still support Windows XP and Vista? Or old macOS versions? Or very old Linux versions?
> 
> For as long as it doesn't require the actually effort / workarounds on a code level to support those.

closed 2022-04-25 by mcuee

aitap, 2023-03-07:
When the use of [`CM_Get_DevNode_PropertyW`](https://learn.microsoft.com/en-us/windows/win32/api/cfgmgr32/nf-cfgmgr32-cm_get_devnode_propertyw) (Vista and later) has been introduced in 11cc0e8a86b33f5b760934eef32e09c55f2c8eae, it was made optional. In 2551a5d84baca323ca8d61db06464cd1b963c289, failure to resolve `CM_Get_DevNode_PropertyW` becomes a fatal error, which effectively drops Windows XP support.

This is not a complaint (though if you're interested, I could submit a pull request); I'm just providing information for the sake of someone else whose code will fail to initialise on an old machine and who will go searching for "Windows XP" in this repo.

Youw, 2023-03-07:
Thanks for observation.
If you're interested in Windows XP support and can test it - a PR with a fix is welcome.

As for general direction - there're still older releases of HIDAPI that build&run on Windows XP, and I'm sure there'are projects that are still using it.
But until I hear about them here on Github (as an issue or a PR), I say that we can effectively abandone WindowsXP support, even if not intentionally, but as a result of a refactoring/improvement.