libusb/hidapi #240: windows：How to interact with a composite device with multiple identical HID interfaces

uzsunliming1, 2021-01-29:
I plan to design a composite device with 4 interfaces. These 4 interfaces are all general HID devices. Each device has an interrupt input and an interrupt output endpoint.
It is not clear how to open these four devices and send and receive data with their eight endpoints;
If we use the writefie method to send data to four output endpoints at the same time, can we achieve greater data throughput?
res1 = WriteFile(dev1-&gt;Device_handle...
res2 = WriteFile(dev2-&gt;Device_handle...
res3 = WriteFile(dev3-&gt;Device_handle...
res4 = WriteFile(dev4-&gt;Device_handle...
If using asynchronous mode, can four sets of data arrive at the device end in one polling cycle?Assume that all endpoints on the device side have a polling period of 1 millisecond (they are working in USB 2.0 full speed mode).

uzslm

todbot, 2021-01-29:
If you're interested in data throughput, do not use HID. It is not designed for that.

jdk, 2021-01-29:
It’s not going to all happen at once like that. You write to the same device pointer with different endpoint addresses. If they have different interfaces. They can each get their own handle. (Different apps can open and read/write to the device)

Host will poll 1ms intervals and your throughout will be divided by however many endpoints you are wanting to write concurrently.  

Just a note, HID 1ms polls are broken in Linux (may only be arm).  Workarounds exist for keyboards and mice, not generic. 

Very Sincerely,
Jason

ninouing, 2021-03-02:
Hi, I'm also interested in something similar. 

I have one device with multiple interfaces. I modified the code so that when I call hid_open it now checks also for the interface number, together with VID and PID. So until I open one interface, I communicate with it, then I close it and move to the next interface, all is ok. My problem is that I would like to communicate with all 4 of them without having to continuously open and close the communication. If I try to open multiple interfaces at the same time, creating 4 different handle variables and calling hid_open multiple times it doesn't work. 

Jason, you wrote that you can give each of them their own handle. Can you explain with more details how to have the various devices open at the same time, without having to open and close each time the device to switch to the other one?

Thank you in advance

jdk, 2021-03-02:
Can you share your code? I do this all the time just like you described. 

Very Sincerely,
Jason

ninouing, 2021-03-02:
> Can you share your code? I do this all the time just like you described. Very Sincerely, Jason
> […](#)

Hi Jason,

Forgive me, it was a stupid typo on my side. I found it after posting and I deleted my comment to not disturb you anymore but apparently you received it nonetheless. 

Thank you anyway!

closed 2021-06-10 by mcuee