libusb/hidapi #243: Can't import hidapi on my script

sonizef, 2021-02-06:
Hello ! 👋 

I'm trying to import hidapi on my script and I get this error :

`File "/usr/local/lib/python3.9/site-packages/hidapi.py", line 4, in <module>
    ffi.cdef("""
  File "/usr/local/lib/python3.9/site-packages/cffi/api.py", line 112, in cdef
    self._cdef(csource, override=override, packed=packed, pack=pack)
  File "/usr/local/lib/python3.9/site-packages/cffi/api.py", line 126, in _cdef
    self._parser.parse(csource, override=override, **options)
  File "/usr/local/lib/python3.9/site-packages/cffi/cparser.py", line 390, in parse
    self._internal_parse(csource)
  File "/usr/local/lib/python3.9/site-packages/cffi/cparser.py", line 395, in _internal_parse
    ast, macros, csource = self._parse(csource)
  File "/usr/local/lib/python3.9/site-packages/cffi/cparser.py", line 339, in _parse
    self.convert_pycparser_error(e, csource)
  File "/usr/local/lib/python3.9/site-packages/cffi/cparser.py", line 368, in convert_pycparser_error
    raise CDefError(msg)
cffi.CDefError: cannot parse "ULONG_PTR Internal;"
<cdef source string>:4:5: before: ULONG_PTR`

I've installed Python (3.9.1) with Hombrew on MacOS BigSur.

It's short but I think you know everything about my issue 😅

Thank you!

Youw, 2021-02-06:
I must say that I know too few of the issue you have.
What is obvious, is that you're using python wrapper over the hidapi library (this project) and as far as I can tell, the issue is wrapper/python related.
I suggest asking here: https://github.com/trezor/cython-hidapi
Maybe @prusnak would be more of help.

Youw, 2021-02-06:
What come to my mind: make sure you install hidapi itself with Homebrew too.

sonizef, 2021-02-06:
@Youw Thank for the answer!

I think hidapi is correctly installed with Hombrew

[image]

And my script just contains one line 

import hidapi

I will make some search on the cython-hidapi, thank's !

Youw, 2021-02-06:
> So I don't know where I used cython

How do you think it is possible to `import hidapi` from python? Python doesn't support importing C libraries. Only a special python wrappers over C libraries, and `cython-hidapi` is one of those. That's what you get, when you `pip install hidapi`.

Youw, 2021-02-06:
After taking a closer look, it doesn't look like you're using `cythin-hidapi` - they don't use `ffi`.
Have you been installing [hidapi-cffi](https://github.com/jbaiter/hidapi-cffi)? It doesn't seem like up-to-date or actively supported.

sonizef, 2021-02-07:
YAY !  🥳 Thank you! I just installed hidapi-cffi and everything works!

I'm not familiar with hidapi & python, so thank again for your patience!

closed 2021-02-07 by sonizef