libusb/hidapi #249: Need API to get the HID Report Descriptor or a similar data structure

JoergAtGithub, 2021-02-26:
We need a function to read the HID Report Descriptor or an equal data structure with information about the fields in the different HID reports.

Background:
For the cross platform Mixxx DJ software we need a solution, to allow the end user to map HID controller buttons, faders, knobs, RGB LEDs, etc. These are different on any DJ controller. Therefore this mapping it must be customizeable by end users.
As input we need information about the field, e.g.: an 12 bit at position n of Input report with ID0x02 with a logical minimum of -2048 and a maximum of +2047.

Be-ing, 2021-02-26:
Potentially helpful links:
https://github.com/DIGImend/usbhid-dump
https://github.com/DIGImend/hidrd
Those are both licensed under the GPL, so they couldn't be used directly in hidapi unless the maintainers are willing to change the hidapi license.

https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/

Youw, 2021-02-28:
Go check:
https://github.com/todbot/win-hid-dump
https://github.com/todbot/mac-hid-dump

tschiemer, 2021-03-16:
Actually, it somewhat comes for free as the basic usage page and usage are extracted from the report descriptor which is read here:

https://github.com/libusb/hidapi/blob/0ea5e9502f31a1e4dbc71567c0f9a781930f1d38/libusb/hid.c#L644

In my eyes it'd make sense to directly store this descriptor either
-  by (optionally) allocating memory linked through a new field in `struct hid_device_info` (simpler) or 
- by postponing allocation of found device structs [here](https://github.com/libusb/hidapi/blob/0ea5e9502f31a1e4dbc71567c0f9a781930f1d38/libusb/hid.c#L582) until the size of the report descriptor is found, allocating the base size along with the descriptor size and appending an indefinite size field to `struct hid_device_info` (feels more elegant)

That said, it might be nice or meaningful to somehow incorporate this quite nice [descriptor to string parser](https://github.com/uint32tMnstr/USB-HID-Report-Parser) based on which I'm momentarily trying to make a [generic parser](https://github.com/tschiemer/USB-HID-Report-Parser/tree/usbhid_map) to help extract meaningful values from given reports without having to know the report layout beforehand (primarily trying to make it work with my use cases, but for simple devices should be fine).

Be-ing, 2021-03-16:
Thanks for that info @tschiemer. I have not looked into what it would take to implement this. If you have a branch of hidapi you are working on, please open a pull request for this upstream repository and I'd be happy to test it with my Native Instruments Traktor Kontrol S4 Mk3.

tschiemer, 2021-03-16:
Ok, can see to it, but might take a moment

Youw, 2021-03-16:
> Actually, it somewhat comes for free as the basic usage page and usage are extracted from the report descriptor

While this indeed already available and easy to get for libusb and hidraw backends, one of the first things about HIDAPI is being cross-platform.

I agree that this would be very useful, but Windows backend doesn't have an API to get a descriptor at all.
The descriptor could be partially reconstructed, but still, it is not an easy task.
I know one successfull attempt to reconstruct a HID descriptor using WinAPI calls, but that one is written on C# (https://github.com/todbot/win-hid-dump is using it). If someone could port it to C - I'd gladly introduce a new API (to get HID descriptor) for all _platforms/backends_.

JoergAtGithub, 2021-03-16:
There's also the approach to use the generic WinUSB driver (comes with Win7 or newer) instead of Windows High-Level-HID-API: https://github.com/libusb/hidapi/issues/61#issuecomment-638397298
As far as I understood it, the WinUSB driver gives a userspace Windows program full access to an USB device, but locks it exclusive.

Youw, 2021-03-16:
> gives a userspace Windows program full access to an USB device, but locks it exclusive

Yeah, and someone actually needs to make a driver for each specific device. And the problem not even with making a driver - a template inf file can be generated. But signing a driver properly - that requires some effort.

And you'd need to compile a libusb backend on Windows, which isn't compilable of-the-box yet.

JoergAtGithub, 2021-03-16:
Ok, that doesn't sound like what we're looking for.

Be-ing, 2021-03-17:
I think it could be a good idea to start with libusb or hidraw to figure out what the API should be then implement it for other backends.

tschiemer, 2021-03-22:
@Be-ing so not quite a hidapi branch, but as a proof of concept I implemented [this](https://github.com/tschiemer/hid-pd-external) puredata external to parse input reports based on the report descriptor, works for me on macos and a rpi4/linux with a joystick and mouse (see [releases](https://github.com/tschiemer/hid-pd-external/releases/tag/v0.1.0) for binaries).

It's a bit of a mess as it uses libusb to find devices according to search criteria (more than hidapi provides out of the box, but meaningful in my opinion and based on hidapi-libusb internals) and reads/parses the report descriptor but relies on hidapi to actually receive input reports (and thus on linux the device listing is set to detach devices as in the hidapi driver..) - should have had a deeper look at hidapi's (or hidraw) interface before..

At the moment only the raw values as given in the input report are extracted without making use of any of the logical/physical min/max, unit, exponential etc values but adding that shouldn't be too big of a deal..
Could not test this with any feature/output reports or more complicated report descriptors, as I have neither the devices nor much more capacity..
The hid report parser just creates a datastructure that's more easy to handle but I would guess it won't work for all descriptors (collection items are somewhat ignored; note that the handling of output and feature items likely will need some fixing in the parser..).
As external it "publishes" any changed values (usage page, usage, value), similar to the apple HIDIOManager value callbacks.
I guess construction of output reports can be achieved by inverting the logic of the value extractor.

Maybe this is helpful in some form

JoergAtGithub, 2021-04-06:
> I agree that this would be very useful, but Windows backend doesn't have an API to get a descriptor at all.
> The descriptor could be partially reconstructed, but still, it is not an easy task.
> I know [...]

I implemented such a reconstructor in C and will start a Pull Request for HIDAPI soon. Stay tuned!

JoergAtGithub, 2021-04-10:
I started PR #262 with a C implementation of the Windows Report-Descriptor code.

mcuee, 2021-06-13:
I think libusb HID backend has already the codes to partially reconstruct the descriptors. Some from the HUB driver, some from the HID driver.

> While this indeed already available and easy to get for libusb and hidraw backends, one of the first things about HIDAPI is being cross-platform.
>
> I agree that this would be very useful, but Windows [...]

mcuee, 2021-06-17:
FYI I have just tested win-hid-dump. It seems I got some problems with it.
https://github.com/todbot/win-hid-dump/issues/2

mcuee, 2021-06-17:
But HIDSharp seems to work. I have a long list of HID device in the system (Windows 10, Dell Laptop with a docking station).

<details>
<summary> click to expand </summary>

```
/c/work/hid/HidSharp
$ ./bin/HidSharp.Test.exe
All device list:
(unnamed manufacturer) (unnamed product) (no serial number) (VID 32903, PID 2590, version 2.0) @ \\?\hid#intc816&col01#3&36a7043c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Microsoft HIDI2C Device 9999 (VID 1160, PID 4639, version 4.13) @ \\?\hid#dell091a&col01#5&99b72d3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Microsoft HIDI2C Device 9999 (VID 1160, PID 4639, version 4.13) @ \\?\hid#dell091a&col02#5&99b72d3&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col03#7&1ebb799e&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
   (VID 1102, PID 4626, version 0.0) @ \\?\hid#vid_044e&pid_1212&col01&col02#7&290aacae&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col04#7&1ebb799e&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
Plantronics Plantronics Blackwire 3220 Series D1CEC32927974D5F9BD6B2AEBF2EA8E3 (VID 1151, PID 49238, version 2.10) @ \\?\hid#vid_047f&pid_c056&mi_03&col01#f&39e6f119&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col05#7&1ebb799e&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
(unnamed manufacturer) (unnamed product) (no serial number) (VID 1118, PID 0, version 0.0) @ \\?\hid#converteddevice&col02#5&379854aa&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
(unnamed manufacturer) (unnamed product) (no serial number) (VID 32903, PID 2590, version 2.0) @ \\?\hid#intc816&col02#3&36a7043c&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
   (VID 1102, PID 4626, version 0.0) @ \\?\hid#vid_044e&pid_1212&col01&col01#7&290aacae&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Logitech USB Optical Mouse (no serial number) (VID 1133, PID 49271, version 72.0) @ \\?\hid#vid_046d&pid_c077#e&113cd935&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Plantronics Plantronics Blackwire 3220 Series D1CEC32927974D5F9BD6B2AEBF2EA8E3 (VID 1151, PID 49238, version 2.10) @ \\?\hid#vid_047f&pid_c056&mi_03&col03#f&39e6f119&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
Microsoft HIDI2C Device 9999 (VID 1160, PID 4639, version 4.13) @ \\?\hid#dell091a&col04#5&99b72d3&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
(unnamed manufacturer) (unnamed product) (no serial number) (VID 1118, PID 0, version 0.0) @ \\?\hid#converteddevice&col03#5&379854aa&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
DELL Dell USB Entry Keyboard (no serial number) (VID 16700, PID 8455, version 1.78) @ \\?\hid#vid_413c&pid_2107#e&11005b8d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
Microsoft HIDI2C Device 9999 (VID 1160, PID 4639, version 4.13) @ \\?\hid#dell091a&col05#5&99b72d3&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col01#7&1ebb799e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Plantronics Plantronics Blackwire 3220 Series D1CEC32927974D5F9BD6B2AEBF2EA8E3 (VID 1151, PID 49238, version 2.10) @ \\?\hid#vid_047f&pid_c056&mi_03&col02#f&39e6f119&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
(unnamed manufacturer) (unnamed product) (no serial number) (VID 1118, PID 0, version 0.0) @ \\?\hid#converteddevice&col01#5&379854aa&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col02#7&1ebb799e&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_00#7&51bc424&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
Microsoft HIDI2C Device 9999 (VID 1160, PID 4639, version 4.13) @ \\?\hid#dell091a&col03#5&99b72d3&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
(unnamed manufacturer)  (no serial number) (VID 16700, PID 45166, version 1.1) @ \\?\hid#vid_413c&pid_b06e#c&37ff1248&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Microchip Technology Inc. Simple HID Device Demo (no serial number) (VID 1240, PID 63, version 0.2) @ \\?\hid#vid_04d8&pid_003f#e&24ba9f4d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
(unnamed manufacturer)  (no serial number) (VID 16700, PID 45167, version 1.1) @ \\?\hid#vid_413c&pid_b06f#d&3624b04c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
COM3 (\\.\COM3) @ \\.\COM3
BLE device list:

Press any key

Complete device list (took 0 ms to get 26 devices):
\\?\hid#intc816&col01#3&36a7043c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
(unnamed manufacturer) (unnamed product) (no serial number) (VID 32903, PID 2590, version 2.0)
Max Lengths: Input 2, Output 0, Feature 0
Serial Ports:
Report Descriptor:
  05 01 09 0C A1 01 85 08 09 C6 15 00 25 01 35 00 45 01 65 00 55 00 75 01 95 01 81 06 95 07 81 03 C1 00 (34 bytes)
  UsagePage 1
  Usage 12
  Collection 1
    ReportID 8
    Usage 198
    LogicalMinimum 0
    LogicalMaximum 1
    PhysicalMinimum 0
    PhysicalMaximum 1
    Unit 0
    UnitExponent 0
    ReportSize 1
    ReportCount 1
    Input 6
    ReportCount 7
    Input 3
  EndCollection 0
Usage: 1000C 65548
Input: ReportID=8, Length=2, Items=2
[... cut ...]
Usage: FFDA00DA 4292477146
Input: ReportID=0, Length=193, Items=2
  1 Elements x 8 Bits, Units: None, Expected Usage Type: 0, Flags: Variable, Usages: FFDA00DA 4292477146
  1528 Elements x 1 Bits, Units: None, Expected Usage Type: 0, Flags: Constant, Variable, Usages:
Output: ReportID=0, Length=193, Items=1
  1536 Elements x 1 Bits, Units: None, Expected Usage Type: 0, Flags: Constant, Variable, Usages:
Opening device for 20 seconds...
Opened device.
Closed device.

Press a key to exit...

```

</details>

mcuee, 2021-06-17:
mac-hid-dump output seems to be a bit too simple.

```
./mac-hid-dump 
mac-hid-dump:
0000 0000: Apple - Headset
DESCRIPTOR:
  05  0c  09  01  a1  01  05  0c  09  cd  09  ea  09  e9  15  00 
  25  01  95  03  75  01  81  02  95  05  81  05  c0  
  (29 bytes)
1915 1025: ZY.Ltd - ZY Control Mic
DESCRIPTOR:
  05  01  09  02  a1  01  85  04  09  01  a1  00  05  09  19  01 
  29  05  15  00  25  01  95  03  75  01  81  02  95  01  75  05 
  81  03  05  01  09  30  09  31  09  38  15  81  25  7f  75  08 
  95  03  81  06  c0  c0  05  0c  09  01  a1  01  85  01  75  10 
  95  01  15  00  26  ff  03  19  00  2a  ff  03  81  00  c0  05 
  01  09  80  a1  01  85  02  75  01  95  02  15  00  25  01  09 
  81  09  82  81  02  75  01  95  0e  81  03  c0  06  00  ff  09 
  00  a1  01  85  06  15  00  26  ff  00  75  08  95  1f  09  00 
  b1  02  c0  
  (131 bytes)
0000 0000: Apple - 
DESCRIPTOR:
  06  00  ff  0a  ff  00  a1  01  15  00  26  ff  00  75  08  95 
  01  81  02  c0  
  (20 bytes)
046D C52B: Logitech - USB Receiver
DESCRIPTOR:
  06  00  ff  09  01  a1  01  85  10  75  08  95  06  15  00  26 
  ff  00  09  01  81  00  09  01  91  00  c0  06  00  ff  09  02 
  a1  01  85  11  75  08  95  13  15  00  26  ff  00  09  02  81 
  00  09  02  91  00  c0  06  00  ff  09  04  a1  01  85  20  75 
  08  95  0e  15  00  26  ff  00  09  41  81  00  09  41  91  00 
  85  21  95  1f  15  00  26  ff  00  09  42  81  00  09  42  91 
  00  c0  
  (98 bytes)
046D C52B: Logitech - USB Receiver
DESCRIPTOR:
  05  01  09  02  a1  01  85  02  09  01  a1  00  05  09  19  01 
  29  10  15  00  25  01  95  10  75  01  81  02  05  01  16  01 
  f8  26  ff  07  75  0c  95  02  09  30  09  31  81  06  15  81 
  25  7f  75  08  95  01  09  38  81  06  05  0c  0a  38  02  95 
  01  81  06  c0  c0  05  0c  09  01  a1  01  85  03  75  10  95 
  02  15  01  26  ff  02  19  01  2a  ff  02  81  00  c0  05  01 
  09  80  a1  01  85  04  75  02  95  01  15  01  25  03  09  82 
  09  81  09  83  81  60  75  06  81  03  c0  06  bc  ff  09  88 
  a1  01  85  08  19  01  29  ff  15  01  26  ff  00  75  08  95 
  01  81  00  c0  
  (148 bytes)
046D C52B: Logitech - USB Receiver
DESCRIPTOR:
  05  01  09  06  a1  01  05  07  19  e0  29  e7  15  00  25  01 
  75  01  95  08  81  02  81  03  95  05  05  08  19  01  29  05 
  91  02  95  01  75  03  91  01  95  06  75  08  15  00  26  a4 
  00  05  07  19  00  2a  a4  00  81  00  c0  
  (59 bytes)
0000 0000: APPL - BTM
DESCRIPTOR:
  06  00  ff  0a  48  00  a1  01  06  29  ff  85  01  25  7f  95 
  01  75  08  09  01  b1  02  09  02  b1  02  09  25  a1  03  85 
  02  24  76  98  3e  09  03  81  22  09  04  76  38  02  b1  02 
  c0  c0  
  (50 bytes)
1915 1025: ZY.Ltd - ZY Control Mic
DESCRIPTOR:
  05  01  09  06  a1  01  05  07  19  e0  29  e7  15  00  25  01 
  75  01  95  08  81  02  95  01  75  08  81  01  95  03  75  01 
  05  08  19  01  29  03  91  02  95  05  75  01  91  01  95  06 
  75  08  15  00  26  ff  00  05  07  19  00  2a  ff  00  81  00 
  c0  
  (65 bytes)
```

JoergAtGithub, 2021-06-17:
> 
> 
> @DJm00n
> I have rewritten my code that is trying to get HID Report Descriptor from USB devices. Can you give it a try?
> https://github.com/DJm00n/RawInputDemo/tree/hid_report_descriptor_dump
> It successfully dumps descriptor from my keyboard, dualshock4, dualsense, stadia controller etc into debug output in MSVS. > > Now without memleaks and crashes :)
> Cannot dump descriptor from my mouse because it supports only Boot Interface Subclass (see 4.2 Subclass in HID spec).
> 
> It uses IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION under the hood.
> 
> PS: Bluetooth and Bluetooth LE another story that I'll try to investigate.

I tested your new code, on my Windows 7 system. unfortunateley it always fails with:
```
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\psapi.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\cryptbase.dll'. 
[RawInputDevice::QueryDeviceInfo:34] Cannot get USB device info from '\\?\USB#VID_17CC&PID_1130#C86013EA#{a5dcbf10-6530-11d2-901f-00c04fb951ed}' interface.
[RawInputDeviceHid::RawInputDeviceHid:25] New HID device[VID:0000,PID:0000][UP:0000,U:0000]: Manufacturer: '(StandardsystemgerÃ¤te)', Product: 'HID-konformes GerÃ¤t', Interface: `\\?\HID#VID_17CC&PID_1130&MI_04#9&11d406cd&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}`
[RawInputDeviceHid::RawInputDeviceHid:37]   ->Its USB Device[VID:0000,PID:0000,VER:0000]: Manufacturer: '', Product: '', Serial Number: ``, Interface: `\\?\USB#VID_17CC&PID_1130#C86013EA#{a5dcbf10-6530-11d2-901f-00c04fb951ed}`
[RawInputDeviceManager::RawInputManagerImpl::OnInputDeviceChange:309] Invalid device: '255270837'
[RawInputDeviceHid::~RawInputDeviceHid:42] Removed HID device: 'HID-konformes GerÃ¤t', Interface: `\\?\HID#VID_17CC&PID_1130&MI_04#9&11d406cd&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}`
Debug Error!

Program: ...oerg\source\repos\RawInputDemo\bin\x64\Debug\RawInputDemo.exe

abort() has been called

(Press Retry to debug the application)
[RawInputDeviceManager::RawInputManagerImpl::OnInput:355] Device 0xb0039 not found
[RawInputDeviceManager::RawInputManagerImpl::OnInput:355] Device 0xb0039 not found
[RawInputDeviceManager::RawInputManagerImpl::OnInput:355] Device 0xb0039 not found
```

I would be glad if your approach would work, because gather the real Report Descriptor would be better than any reconstruction.

mcuee, 2021-06-18:
@DJm00n I tried RawInputDemo under Windows 10, no problem running it but the RawInputDemo.exe does not seem to do anything. How do I use it?

mcuee, 2021-06-18:
#262 seems to running fine at my Dell Windows 10 laptop with a docking edition and quite some HID devices. The only problem is that I do not know the original HID report descriptor of the devices other than the device I have access to the FW (say Microchip based Simple HID Demo, PICkit 2, Lakeview Research HID demo) and they are on the simpler side.

mcuee, 2021-06-18:
> But HIDSharp seems to work. I have a long list of HID device in the system (Windows 10, Dell Laptop with a docking station).

It is not so stable for my system, sometime it hangs.

DJm00n, 2021-06-18:
@mcuee sorry. I forgot to mention - it outputs info to debug console window in msvs.

mcuee, 2021-06-18:
> @mcuee sorry. I forgot to mention - it outputs info to debug console window in msvs.

This is what I got in  the output window.
```
'RawInputDemo.exe' (Win32): Loaded 'C:\work\libusb\RawInputDemo\bin\x64\Debug\RawInputDemo.exe'. Symbols loaded.
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\apphelp.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\cfgmgr32.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\hid.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\msvcp_win.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140_1d.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140d.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\msvcp140d.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbased.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbased.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbased.dll'. 
'RawInputDemo.exe' (Win32): Unloaded 'C:\Windows\System32\ucrtbased.dll'
'RawInputDemo.exe' (Win32): Unloaded 'C:\Windows\System32\ucrtbased.dll'
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\imm32.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\umppc13702.dll'. 
'RawInputDemo.exe' (Win32): Unloaded 'C:\Windows\System32\umppc13702.dll'
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\umppc13702.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\kernel.appcore.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\uxtheme.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\msctf.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\oleaut32.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\TextShaping.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\TextInputFramework.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\CoreUIComponents.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\SHCore.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\CoreMessaging.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\ntmarta.dll'. 
'RawInputDemo.exe' (Win32): Loaded 'C:\Windows\System32\WinTypes.dll'. 
[RawInputDeviceHid::RawInputDeviceHid:25] New HID device[VID:047F,PID:C056][UP:000B,U:0005]: Manufacturer: 'Plantronics', Product: 'Plantronics Blackwire 3220 Series', Interface: `\\?\HID#VID_047F&PID_C056&MI_03&Col02#f&39e6f119&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}`
[RawInputDeviceHid::RawInputDeviceHid:37]   ->Its USB Device[VID:047F,PID:C056,VER:0210]: Manufacturer: 'Plantronics', Product: 'Plantronics Blackwire 3220 Series', Serial Number: `D1CEC32927974D5F9BD6B2AEBF2EA8E3`, Interface: `\\?\USB#VID_047F&PID_C056#D1CEC32927974D5F9BD6B2AEBF2EA8E3#{a5dcbf10-6530-11d2-901f-00c04fb951ed}`
[RawInputDeviceHid::RawInputDeviceHid:25] New HID device[VID:047F,PID:C056][UP:FFA0,U:0003]: Manufacturer: 'Plantronics', Product: 'Plantronics Blackwire 3220 Series', Interface: `\\?\HID#VID_047F&PID_C056&MI_03&Col03#f&39e6f119&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}`
[RawInputDeviceHid::RawInputDeviceHid:37]   ->Its USB Device[VID:047F,PID:C056,VER:0210]: Manufacturer: 'Plantronics', Product: 'Plantronics Blackwire 3220 Series', Serial Number: `D1CEC32927974D5F9BD6B2AEBF2EA8E3`, Interface: `\\?\USB#VID_047F&PID_C056#D1CEC32927974D5F9BD6B2AEBF2EA8E3#{a5dcbf10-6530-11d2-901f-00c04fb951ed}`
[RawInputDeviceHid::RawInputDeviceHid:25] New HID device[VID:047F,PID:C056][UP:000C,U:0001]: Manufacturer: 'Plantronics', Product: 'Plantronics Blackwire 3220 Series', Interface: `\\?\HID#VID_047F&PID_C056&MI_03&Col01#f&39e6f119&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}`
[RawInputDeviceHid::RawInputDeviceHid:37]   ->Its USB Device[VID:047F,PID:C056,VER:0210]: Manufacturer: 'Plantronics', Product: 'Plantronics Blackwire 3220 Series', Serial Number: `D1CEC32927974D5F9BD6B2AEBF2EA8E3`, Interface: `\\?\USB#VID_047F&PID_C056#D1CEC32927974D5F9BD6B2AEBF2EA8E3#{a5dcbf10-6530-11d2-901f-00c04fb951ed}`
[RawInputDeviceHid::RawInputDeviceHid:25] New HID device[VID:04D8,PID:0033][UP:FF00,U:0001]: Manufacturer: 'Microchip Technology Inc.', Product: 'PICkit 2 Microcontroller Programmer', Interface: `\\?\HID#VID_04D8&PID_0033#e&bd5b18b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}`
[... cut ...]
[RawInputDeviceMouse::RawInputDeviceMouse:21]   ->Its USB Device[VID:046D,PID:C077,VER:7200]: Manufacturer: 'Logitech', Product: 'USB Optical Mouse', Serial Number: ``, Interface: `\\?\USB#VID_046D&PID_C077#d&3e263b&0&4#{a5dcbf10-6530-11d2-901f-00c04fb951ed}`
[RawInputDeviceMouse::RawInputDeviceMouse:15] New Mouse device: '', Interface: `\\?\HID#Vid_044E&Pid_1212&Col01&Col01#7&290aacae&0&0000#{378de44c-56ef-11d1-bc8c-00a0c91405dd}`
[RawInputDeviceMouse::RawInputDeviceMouse:18]   ->Its HID Device[VID:044E,PID:1212]: Interface: `\\?\HID#Vid_044E&Pid_1212&Col01&Col01#7&290aacae&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}`

```

mcuee, 2021-06-18:
I found that pull request #262 is not correct for Microchip PICKit 2 and then I run HIDSharp again and it is also not correct.

Original HID report descriptor
```
rom struct{byte report[29];}hid_rpt01={
    0x06, 0x00, 0xFF,       // Usage Page (Vendor Defined)
    0x09, 0x01,             // Usage (Vendor Usage)
    0xA1, 0x01,             // Collection (Application)
    0x19, 0x01,             //      Usage Minimum (Vendor Usage = 1)
//    0x29, 0x08,             //      Usage Maximum (Vendor Usage = 8)
    0x29, 0x40,             //      Usage Maximum (Vendor Usage = 64)
    0x15, 0x00,             //      Logical Minimum (Vendor Usage = 0)
    0x26, 0xFF, 0x00,       //      Logical Maximum (Vendor Usage = 255)
    0x75, 0x08,             //      Report Size (Vendor Usage = 8)
//    0x95, 0x08,             //      Report Count (Vendor Usage = 8)
    0x95, 0x40,             //      Report Count (Vendor Usage = 64)
    0x81, 0x02,             //      Input (Data, Var, Abs)
    0x19, 0x01,             //      Usage Minimum (Vendor Usage = 1)
//    0x29, 0x08,             //      Usage Maximum (Vendor Usage = 8)
    0x29, 0x40,             //      Usage Maximum (Vendor Usage = 64)
    0x91, 0x02,             //      Output (Data, Var, Ads)
    0xC0};                  // End Collection
```

Dumped by pull request #262
```
 06  00  FF  09  01  A1  01  19  01  29  40  15  00  26  FF  00  75  08  95  01  81  02  19  01  29  40  15  00  26  FF  00  75  08  95  01  91  02  C0 
```

https://eleccelerator.com/usbdescreqparser/ output.
```
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x40,        //   Usage Maximum (0x40)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x01,        //   Report Count (1)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x40,        //   Usage Maximum (0x40)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x01,        //   Report Count (1)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 38 bytes
```

As a comparison, this is what I get from HIDSharp which is also not correct.
```
\\?\hid#vid_04d8&pid_0033#e&bd5b18b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Microchip Technology Inc. PICkit 2 Microcontroller Programmer pk2new (VID 1240, PID 51, version 0.2)
Max Lengths: Input 65, Output 65, Feature 0
Serial Ports:
Report Descriptor:
  06 00 FF 09 01 A1 01 15 00 25 01 35 00 45 01 65 00 55 00 75 01 96 00 02 81 03 91 03 C1 00 (30 bytes)
  UsagePage 65280
  Usage 1
  Collection 1
    LogicalMinimum 0
    LogicalMaximum 1
    PhysicalMinimum 0
    PhysicalMaximum 1
    Unit 0
    UnitExponent 0
    ReportSize 1
    ReportCount 512
    Input 3
    Output 3
  EndCollection 0
Usage: FF000001 4278190081
Input: ReportID=0, Length=65, Items=1
  512 Elements x 1 Bits, Units: None, Expected Usage Type: 0, Flags: Constant, Variable, Usages:
Output: ReportID=0, Length=65, Items=1
  512 Elements x 1 Bits, Units: None, Expected Usage Type: 0, Flags: Constant, Variable, Usages:
Opening device for 20 seconds...
Opened device.
Closed device.

```

todbot, 2021-06-18:
The HID Report Descriptors that HIDSharp (and `win-hid-dump`) emit are reconstructions based on what is available via Windows HID APIs, it is *not* the actual descriptor sent by the device, but rather a "consolation-of-a-parsed-data-structure".

mcuee, 2021-06-18:
> The HID Report Descriptors that HIDSharp (and `win-hid-dump`) emit are reconstructions based on what is available via Windows HID APIs, it is _not_ the actual descriptor sent by the device, but rather a "consolation-of-a-parsed-data-structure".

I understand, but in this case, it is quite a bit OFF.

mcuee, 2021-06-18:
@todbot Maybe this device is indeed strange, it has two USB configurations but that should not matter as Windows will ignore the second configuration.

Full USB Descriptor C source code for reference.

<details>
<summary> click to expand </summary>

```
/*********************************************************************
 *
 *                Microchip USB C18 Firmware Version 1.0
 *
 * This file is generated by Microchip USB Wizard Version 1.0, 2004
 *********************************************************************
 * FileName:        usbdsc.c
 * Dependencies:    See INCLUDES section below
 * Processor:       PIC18
 * Compiler:        C18 3.00
 * Company:         Microchip Technology, Inc.
 *
 * Software License Agreement
 *
 * The software supplied herewith by Microchip Technology Incorporated
 * (the “Company”) for its PICmicro® Microcontroller is intended and
 * supplied to you, the Company’s customer, for use solely and
 * exclusively on Microchip PICmicro Microcontroller products. The
 * software is owned by the Company and/or its supplier, and is
 * protected under applicable copyright laws. All rights are reserved.
 * Any use in violation of the foregoing restrictions may subject the
 * user to criminal sanctions under applicable laws, as well as to
 * civil liability for the breach of the terms and conditions of this
 * license.
 *
 * THIS SOFTWARE IS PROVIDED IN AN “AS IS” CONDITION. NO WARRANTIES,
 * WHETHER EXPRESS, IMPLIED OR STATUTORY, INCLUDING, BUT NOT LIMITED
 * TO, IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A
 * PARTICULAR PURPOSE APPLY TO THIS SOFTWARE. THE COMPANY SHALL NOT,
 * IN ANY CIRCUMSTANCES, BE LIABLE FOR SPECIAL, INCIDENTAL OR
 * CONSEQUENTIAL DAMAGES, FOR ANY REASON WHATSOEVER.
 *
 ********************************************************************/

/*********************************************************************
 * Descriptor specific type definitions are defined in:
 * system\usb\usbdefs\usbdefs_std_dsc.h
 *
 * Configuration information is defined in:
 * autofiles\usbcfg.h
 ********************************************************************/
 
/** I N C L U D E S *************************************************/
#include "system\typedefs.h"
#include "system\usb\usb.h"

/** C O N S T A N T S ************************************************/
#pragma romdata

/* Device Descriptor */
rom USB_DEV_DSC device_dsc=
{    
    sizeof(USB_DEV_DSC),    // Size of this descriptor in bytes
    DSC_DEV,                // DEVICE descriptor type
    0x0200,                 // USB Spec Release Number in BCD format
    0x00,                   // Class Code
    0x00,                   // Subclass code
    0x00,                   // Protocol code
    EP0_BUFF_SIZE,          // Max packet size for EP0, see usbcfg.h
    0x04D8,                 // Vendor ID = 1240
    0x0033,                 // Product ID = 51
    0x0002,                 // Device release number in BCD format
[... 124 lines cut ...]
//    0x95, 0x08,             //      Report Count (Vendor Usage = 8)
    0x95, 0x40,             //      Report Count (Vendor Usage = 64)
    0x81, 0x02,             //      Input (Data, Var, Abs)
    0x19, 0x01,             //      Usage Minimum (Vendor Usage = 1)
//    0x29, 0x08,             //      Usage Maximum (Vendor Usage = 8)
    0x29, 0x40,             //      Usage Maximum (Vendor Usage = 64)
    0x91, 0x02,             //      Output (Data, Var, Ads)
    0xC0};                  // End Collection

rom const unsigned char *rom USB_CD_Ptr[]={&cfg01,&cfg02};
//rom const unsigned char *rom USB_SD_Ptr[]={&sd000,&sd001,&sd002,&sd003,&sd004};
rom const unsigned char *rom USB_SD_Ptr[]={&sd000,&sd001,&sd002};

rom pFunc ClassReqHandler[1]=
{
    &USBCheckHIDRequest
};

#pragma code

/** EOF usbdsc.c ****************************************************/

```

</details>

todbot, 2021-06-18:
Also, @mcuee since this is a PICkit2 and not your own device, have you uninstalled all the Microchip Windows drivers? Vendor drivers usually really mess up USB HID libraries. (Some even create fake HID devices that mask the real device. Not what's going on here I think)

mcuee, 2021-06-18:
@todbot This is a standard HID device and no vendor driver is needed. I know this Microchip PICKit 2 very well as I was helping Microchip to develop the cross platform pk2cmd applications.

mcuee, 2021-06-18:
> The HID Report Descriptors that HIDSharp (and `win-hid-dump`) emit are reconstructions based on what is available via Windows HID APIs, it is _not_ the actual descriptor sent by the device, but rather a "consolation-of-a-parsed-data-structure".
> 
> It should be semantically (to the OS) the same, however. If it's not, it usually means the descriptor sent by the device is wrong in some subtle way. For vendor usages, this doesn't much matter, as then all that really matters is REPORT_ID, REPORT_COUNT, and REPORT_SIZE

@todbot This is a good point. In that case, it seems #262 is actually okay for Microchip PICkit 2. Report size of 8 bytes is correct. Report count of 1 is actually correct as well. Report ID is in fact also correct (0, or no report ID).

mcuee, 2021-06-18:
The following raw report is reported by mac-hid-dump (https://github.com/todbot/mac-hid-dump).
```
04D8 0033: Microchip Technology Inc. - PICkit 2 Microcontroller Programmer
DESCRIPTOR:
  06  00  ff  09  01  a1  01  19  01  29  40  15  00  26  ff  00 
  75  08  95  40  81  02  19  01  29  40  91  02  c0  
  (29 bytes)
```

Translation from https://eleccelerator.com/usbdescreqparser/ 

```
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x40,        //   Usage Maximum (0x40)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x40,        //   Report Count (64)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x40,        //   Usage Maximum (0x40)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 29 bytes
```

mcuee, 2021-06-18:
The following is from USB Prober under macOS Big Sur (Mac Mini M1) 

```
Full Speed device @ 15 (0x02145700): .............................................   Composite device: "PICkit 2 Microcontroller Programmer"
    Port Information:   0x0018
           Not Captive
           External Device
           Connected
           Enabled
    Number Of Endpoints (includes EP0):   
        Total Endpoints for Configuration 1 (current):   3
        Total Endpoints for Configuration 2 :   3
    Device Descriptor   
        Descriptor Version Number:   0x0200
        Device Class:   0   (Composite)
        Device Subclass:   0
        Device Protocol:   0
        Device MaxPacketSize:   8
        Device VendorID/ProductID:   0x04D8/0x0033   (Microchip Technology Inc.)
        Device Version Number:   0x0002
        Number of Configurations:   2
        Manufacturer String:   1 "Microchip Technology Inc."
        Product String:   2 "PICkit 2 Microcontroller Programmer"
        Serial Number String:   3 "pk2new"
    Configuration Descriptor (current config): ......................   "PICkit 2 Microcontroller Programmer"
        Length (and contents):   41
            Raw Descriptor (hex)    0000: 09 02 29 00 01 01 02 80  32 09 04 00 00 02 03 00  
            Raw Descriptor (hex)    0010: 00 00 09 21 01 00 00 01  22 1D 00 07 05 81 03 40  
            Raw Descriptor (hex)    0020: 00 01 07 05 01 03 40 00  01 
        Number of Interfaces:   1
        Configuration Value:   1
        Attributes:   0x80 (bus-powered)
        MaxPower:   100 mA
        Interface #0 - HID   
            Alternate Setting   0
            Number of Endpoints   2
            Interface Class:   3   (HID)
            Interface Subclass;   0
            Interface Protocol:   0
            HID Descriptor   
                Descriptor Version Number:   0x0001
                Country Code:   0
                Descriptor Count:   1
                Descriptor 1   
                    Type:   0x22  (Report Descriptor)
                    Length (and contents):   29
                        Raw Descriptor (hex)    0000: 06 00 FF 09 01 A1 01 19  01 29 40 15 00 26 FF 00  
                        Raw Descriptor (hex)    0010: 75 08 95 40 81 02 19 01  29 40 91 02 C0 
                    Parsed Report Descriptor:   
                          Usage Page    (Vendor defined 0) 
                          Usage 1 (0x1)    
                              Collection (Application)    
                                Usage Minimum...........    (1)  
                                Usage Maximum...........    (64)  
                                Logical Minimum.........    (0)  
                                Logical Maximum.........    (255)  
                                Report Size.............    (8)  
                                Report Count............    (64)  
                                Input...................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Bitfield) 
                                Usage Minimum...........    (1)  
                                Usage Maximum...........    (64)  
                                Output..................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Nonvolatile, Bitfield) 
                              End Collection     
            Endpoint 0x81 - Interrupt Input   
                Address:   0x81  (IN)
                Attributes:   0x03  (Interrupt)
                Max Packet Size:   64
                Polling Interval:   1 ms
            Endpoint 0x01 - Interrupt Output   
                Address:   0x01  (OUT)
                Attributes:   0x03  (Interrupt)
                Max Packet Size:   64
                Polling Interval:   1 ms
    Configuration Descriptor: .......................................   ""
        Length (and contents):   32
            Raw Descriptor (hex)    0000: 09 02 20 00 01 02 04 80  32 09 04 00 00 02 FF 00  
            Raw Descriptor (hex)    0010: 00 00 07 05 81 03 40 00  01 07 05 01 03 40 00 01  
            Unknown Descriptor   0020: 
        Number of Interfaces:   1
        Configuration Value:   2
        Attributes:   0x80 (bus-powered)
        MaxPower:   100 mA
        Interface #0 - Vendor-specific   
            Alternate Setting   0
            Number of Endpoints   2
            Interface Class:   255   (Vendor-specific)
            Interface Subclass;   0   (Vendor-specific)
            Interface Protocol:   0
            Endpoint 0x81 - Interrupt Input   
                Address:   0x81  (IN)
                Attributes:   0x03  (Interrupt)
                Max Packet Size:   64
                Polling Interval:   1 ms
            Endpoint 0x01 - Interrupt Output   
                Address:   0x01  (OUT)
                Attributes:   0x03  (Interrupt)
                Max Packet Size:   64
                Polling Interval:   1 ms
```

mcuee, 2021-06-18:
mac-hid-dump is very accurate accurate. It matches exactly my FW codes.

```
0925 7001: Lakeview Research - Generic HID
DESCRIPTOR:
  06  00  ff  09  01  a1  01  15  00  26  ff  00  75  08  95  02 
  85  01  09  00  82  02  01  85  02  09  00  92  02  01  85  03 
  09  00  b2  02  01  85  04  09  00  b2  02  01  c0  
  (45 bytes)
```

FW:
```
ROM struct{BYTE report[HID_RPT01_SIZE];}hid_rpt01={
 {
   0x06, 0x00, 0xFF,               // Usage page (Vendor Defined Page1)
   0x09, 0x01,                     // Usage (Vendor Usage 1)
   0xA1, 0x01,                     // Collection (Application)

   // Global items apply to all reports below

   0x15, 0x00,                     // Logical Minimum (0)
   0x26, 0xFF, 0x00,               // Logical Maximum (255)
   0x75, 0x08,                     // Report Size (8 bits)
   0x95, 0x02,                     // Report Count (2 fields)

   // Input report
   0x85, 0x01,                     // Report ID (1)
   0x09, 0x00,                     // Usage (Undefined)
   0x82, 0x02, 0x01,               // Input (Data,Var,Abs,Buf)

   // Output report
   0x85, 0x02,                     // Report ID (2)
   0x09, 0x00,                     // Usage (Undefined)
   0x92, 0x02, 0x01,               // Output (Data,Var,Abs,Buf)

   // Feature report
   0x85, 0x03,                     // Report ID (3)
   0x09, 0x00,                     // Usage (Undefined)
   0xB2, 0x02, 0x01,               // Feature (Data,Var,Abs,Buf)

   // Feature report
   0x85, 0x04,                     // Report ID (4)
   0x09, 0x00,                     // Usage (Undefined)
   0xB2, 0x02, 0x01,               // Feature (Data,Var,Abs,Buf)

   0xC0                            // end collection
 }```

mcuee, 2021-06-18:
> > Actually, it somewhat comes for free as the basic usage page and usage are extracted from the report descriptor
> 
> While this indeed already available and easy to get for libusb and hidraw backends, one of the first things about HIDAPI is being cross-platform.

So we seem to have a good solution for Linux/FreeBSD already. For macOS, it seems to me mac-hid-dump is a good base.

Only Windows may be challenging. But it seems to me #262 works reasonably well based on my limited testing.

mcuee, 2021-06-18:
usbhid-dump under Linux is indeed very accurate too. It matched exactly the FW codes.

For 0925 7001: Lakeview Research - Generic HID
```
001:009:000:DESCRIPTOR         1624018582.659913
 06 00 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 02
 85 01 09 00 82 02 01 85 02 09 00 92 02 01 85 03
 09 00 B2 02 01 85 04 09 00 B2 02 01 C0
```

JoergAtGithub, 2021-06-18:
> @todbot This is a good point. In that case, it seems #262 is actually okay for Microchip PICkit 2. Report size of 8 bytes is correct. Report count of 1 is actually correct as well. Report ID is in [...]

Did you change the Report Descriptor of your test device between the tests? Otherwise I can't understand why Report count of 1 is Ok here.

JoergAtGithub, 2021-06-18:
MacOS and Linux provide APIs for reading the Report descriptor of an HID device. Therefore you get the unmodified Report Descriptor on these operating systems.
Unfortunately the Windows HID subsystem doesn't provide such an API. What I did in #262 is to reconstruct a Report Descriptor from the available high-level HID API (same technical approach as win-hid-dump). This reconstructed Report Descriptor has the same logical meaning in most aspects, as the original Report Descriptor, but is bytewise different.
There is another promissing approach by @DJm00n , which unfortunately doesn't work for me yet, to download the real HID report Descriptor using Windows USB API: https://github.com/DJm00n/RawInputDemo/tree/hid_report_descriptor_dump

mcuee, 2021-06-19:
I have asked in the libusb-devel to see if Tim Roberts has some suggestions.

mcuee, 2021-06-19:
No I did not change. So looks like there are still issues.

BTW, I am not that confident of the report parser website or Linux hidrd for the devices. I will probably check the output from Apple USB Prober  under macOS.

> Did you change the Report Descriptor of your test device between the tests? Otherwise I can't understand why Report count of 1 is Ok here.

mcuee, 2021-06-19:
Answer from Tim in libusb-devel mailing list

On Sat, Jun 19, 2021 at 12:14 PM Tim Roberts wrote:
>
> On 6/18/21 6:56 PM, Xiaofan Chen wrote:
>
> Hi Tim,
>
> Switch to another question. Hopefully you can shed some light on this as well. Thanks.
>
> MacOS and Linux provide APIs for reading the HID Report descriptor of an HID device. Therefore we can get the unmodified HID Report Descriptor on these operating systems. Unfortunately the Windows HID subsystem doesn't provide such an API.
>
> Yes.  This is a huge irritation to me.  I have done two projects recently where I needed to imitate another device.  If I could have grabbed the original report descriptor, this task would have been way easier.  In both cases, I ended up hard-coding the report descriptor, which is distasteful.
>
>
> What will be the good approach to reconstruct the HID report descriptor?
>
> I spent some time with this, and I don't think it is possible.  There's just not enough information in the "parsed descriptors" to do that.
>
> MAYBE (I haven't proven this to myself) you could use the same hub backdoor that "usbview" uses, and fetch the report descriptor that way.
>
>
> If we just limit to USB HID device and not Bluetooth HID device, will it be a bit easier?
>
> I don't think it makes much difference.  The "usbview" trick would only be for USB HID devices.
>
> -- 
> Tim Roberts
> Providenza & Boekelheide, Inc.

mcuee, 2021-06-19:
So it seems to me the only possibility is probably the usbview trick. It is only for USB HID devices though, not Bluetooth.
https://github.com/microsoft/Windows-driver-samples/tree/master/usb/usbview
https://github.com/DJm00n/Windows-driver-samples/commits/usb-hid-descriptor/usb/usbview

I need to check again the following.
https://github.com/DJm00n/RawInputDemo/tree/hid_report_descriptor_dump

JoergAtGithub, 2021-06-19:
There is also the reconstructor from Google Chromiums WebHID implementation, which accesses the not official documented struct _HIDP_PREPARSED_DATA directly instead of using the official functions for this. 
https://chromium.googlesource.com/chromium/src/+/73fdaaf605bb60caf34d5f30bb84a417688aa528/services/device/hid/hid_preparsed_data.cc

This approach colud solve the remaining issues with #262 and win_hid_dump, where HidP_SetData returns HIDP_STATUS_IS_VALUE_ARRAY ( button array with usage 0). I think this case is an error in the HidP_SetData function, which Google circumvented by accessing the _HIDP_PREPARSED_DATA direct, which is the input for HidP_SetData.

JoergAtGithub, 2021-06-19:
> No I did not change. So looks like there are still issues.
>
> BTW, I am not that confident of the report parser website or Linux hidrd for the devices. I will probably check the output from Apple USB [...]

This should be solveable, it is the second case in rd_determine_value_bitpositions, where I commented "// EXPERIMENTAL - No device available for test": https://github.com/libusb/hidapi/blob/b54a041cad6dee24703bc5baa0db0aa71f068e51/windows/hid.c#L870
win_hid_dump does this better here, but therefore fails with the report_size, which #262 determines correct . But it proves that the needed data are available. The win_hid_dump implementation of this case is here: https://github.com/IntergatedCircuits/HidSharp/blob/e8359220e0cdb1541844835055f5f19af02a9f4c/HidSharp/Platform/Windows/WinHidDevice.ReportDescriptorReconstructor.cs#L161

mcuee, 2021-06-19:
Personally I think using unofficiallly documented internal struct _HIDP_PREPARSED_DATA is not the way to go for HIDAPI. But anyway, I have asked in the libusb-devel mailing list again. 

I will check the general question in OSR forum as well.

JoergAtGithub, 2021-06-19:
I think the best possible approach would be, to try the USB hub trick to read the real Report Descriptor, and if it fails return the reconstructed Report Descriptor instead. The later is needed anyway, because the USB hub trick will not work with Bluetooth HID devices.

mcuee, 2021-06-19:
Just tried this one and it does not seem to work at all for USB HID Report Descriptor, tested on my home Acer Windows 10 laptop.
https://github.com/DJm00n/Windows-driver-samples/tree/usb-hid-descriptor/usb/usbview

For example, this is the output from my mode of Lakeview Research Demo.

```
       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x02
wTotalLength:                    0x0029  -> Validated
bNumInterfaces:                    0x01
bConfigurationValue:               0x01
iConfiguration:                    0x00
bmAttributes:                      0xC0  -> Self Powered
MaxPower:                          0x32 = 100 mA

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x04
bInterfaceNumber:                  0x00
bAlternateSetting:                 0x00
bNumEndpoints:                     0x02
bInterfaceClass:                   0x03  -> HID Interface Class
bInterfaceSubClass:                0x00
bInterfaceProtocol:                0x00
iInterface:                        0x00

          ===>HID Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x21
bcdHID:                          0x0111
bCountryCode:                      0x00
bNumDescriptors:                   0x01
bDescriptorType:                   0x22 (Report Descriptor)
wDescriptorLength:               0x002D
*!*ERROR:  No HID Descriptor for interface 0!

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x01

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x01

```

mcuee, 2021-06-19:
Then I tried RawInputDemo here with my home Acer Windows 10 laptop and it crashed.
https://github.com/DJm00n/RawInputDemo/tree/hid_report_descriptor_dump

```
abort() has been called

(Press Retry to debug the application)
RawInputDemo.exe has triggered a breakpoint.
```

Call stack
```
 	ucrtbased.dll!00007ffb691e7c75()	Unknown
 	ucrtbased.dll!00007ffb691e7e13()	Unknown
 	ucrtbased.dll!00007ffb691fe01d()	Unknown
>	RawInputDemo.exe!LogMessageFatal::~LogMessageFatal() Line 176	C++
 	RawInputDemo.exe!`anonymous namespace'::GetUsbHidReportDescriptor(const std::unique_ptr<void,ScopedHandleDeleter> & usbHubHandle, unsigned long connectionIndex, std::vector<unsigned char,std::allocator<unsigned char>> & outReportDescriptor) Line 315	C++
 	RawInputDemo.exe!RawInputDevice::QueryUsbDeviceInfo() Line 367	C++
 	RawInputDemo.exe!RawInputDevice::QueryDeviceInfo() Line 32	C++
 	RawInputDemo.exe!RawInputDeviceHid::QueryDeviceInfo() Line 54	C++
 	RawInputDemo.exe!RawInputDeviceHid::RawInputDeviceHid(void * handle) Line 23	C++
 	RawInputDemo.exe!RawInputDeviceFactory<RawInputDeviceHid>::Create(void * handle) Line 96	C++
 	RawInputDemo.exe!RawInputDeviceManager::RawInputManagerImpl::CreateRawInputDevice(unsigned long deviceType, void * handle) Line 371	C++
 	RawInputDemo.exe!RawInputDeviceManager::RawInputManagerImpl::OnInputDeviceChange() Line 306	C++
 	RawInputDemo.exe!RawInputDeviceManager::RawInputManagerImpl::ThreadRun() Line 140	C++

```

Be-ing, 2021-06-19:
Thanks for working on this @mcuee!

Youw, 2021-06-19:
There is one general issue with "usbview" or "IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION" way:

Those would give the descriptor of the entire HID device (usb interface) which is  basically what we can get on Linux/macOS.
But that wouldn't be accurate in case of Windows, because Windows creates a unique virtual HID device for each usage_page/usage combination of the HID device.

mcuee, 2021-06-20:
Windows HID Top-Level Collections:
https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/hid-collections

In Windows, the HID device setup class (HIDClass) generates a unique physical device object (PDO) for each Top Level Collection described by the Report Descriptor. Microsoft defines a top-level collection as a HID collection that is not nested within another collection. An unnested collection is always a top-level collection, regardless of its HID type. In particular, a top-level collection does not have to be an Application collection, as defined by the USB HID Standard.

mcuee, 2021-06-20:
>There is one general issue with "usbview" or "IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION" way

There are some arguments about this.

Ref: https://community.osr.com/discussion/287808/ioctl-usb-get-descriptor-from-node-connection
For a HID report descriptor, the Recipient bits of bmRequestType must be set to Interface (1).
IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION sets bmRequestType to 0x80.

From https://github.com/DJm00n/RawInputDemo/commit/026261520bfcf93ed9559c482fd36cd66960d9b4
```

<!--StartFragment-->            // According to HID spec we need to do Report Descriptor request with            // bmRequest set to 0x81 (Interface_In) but seems its also working with 0x80 for some reason            const USHORT hidReportDescriptorSize = hidDescriptor-&gt;DescriptorList[0].wReportLength;            std::unique_ptr&lt;uint8_t[]&gt; hidReportDescriptor = GetUsbDescriptor(usbHubHandle, connectionIndex, HID_REPORT_DESCRIPTOR_TYPE, hidReportDescriptorSize, 0, interfaceDescriptor-&gt;bInterfaceNumber);<br class="Apple-interchange-newline"><!--EndFragment-->

```

mcuee, 2021-06-20:
Example Logitech USB Receiver. It is a USB Composite Device, under Windows Device Manager, it will show as 9 devices, one USB Composite Device parent, then six devices in Human Interface Devices section, one in mouse and one in keyboard.

HIDSharp will show it as six HID devices, one for Interface 0, one for each top level collection of Interface 1 (col 1/2/3/4/5)..

```
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col02#7&2fa52f82&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}

Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_00#7&73a81b9&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd

Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col01#7&2fa52f82&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col04#7&2fa52f82&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}

Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col03#7&2fa52f82&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}

Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col05#7&2fa52f82&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}

```

Full run-log of HIDSharp under my Acer laptop with the Logitech USB receiver installed.

<details>
<summary> click to expand </summary>

```
All device list:
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col02#7&2fa52f82&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_00#7&73a81b9&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col01#7&2fa52f82&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col04#7&2fa52f82&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col03#7&2fa52f82&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
Microsoft HIDI2C Device 9999 (VID 1267, PID 12384, version 0.3) @ \\?\hid#elan0501&col02#5&10870616&1&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1) @ \\?\hid#vid_046d&pid_c534&mi_01&col05#7&2fa52f82&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
Microsoft HIDI2C Device 9999 (VID 1267, PID 12384, version 0.3) @ \\?\hid#elan0501&col01#5&10870616&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Microsoft HIDI2C Device 9999 (VID 1267, PID 12384, version 0.3) @ \\?\hid#elan0501&col03#5&10870616&1&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
Microsoft HIDI2C Device 9999 (VID 1267, PID 12384, version 0.3) @ \\?\hid#elan0501&col04#5&10870616&1&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
Acer Inc. Launch Manager Wireless Device (no serial number) (VID 11254, PID 41110, version 0.1) @ \\?\hid#hidclass&col02#1&2d595ca7&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
Acer Inc. Launch Manager Wireless Device (no serial number) (VID 11254, PID 41110, version 0.1) @ \\?\hid#hidclass&col01#1&2d595ca7&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
BLE device list:

Press any key

Complete device list (took 0 ms to get 12 devices):
\\?\hid#vid_046d&pid_c534&mi_01&col02#7&2fa52f82&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1)
Max Lengths: Input 5, Output 0, Feature 0
Serial Ports:
System.NotSupportedException: Unable to reconstruct the report descriptor.
   at HidSharp.Platform.Windows.WinHidDevice.GetRawReportDescriptor() in C:\work\hid\HidSharp\HidSharp\Platform\Windows\WinHidDevice.cs:line 195
   at HidSharp.Test.Program.Main(String[] args) in C:\work\hid\HidSharp\HidSharp.Test\Program.cs:line 249
\\?\hid#vid_046d&pid_c534&mi_00#7&73a81b9&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
Logitech USB Receiver (no serial number) (VID 1133, PID 50484, version 29.1)
Max Lengths: Input 9, Output 2, Feature 0
Serial Ports:
Report Descriptor:
  05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01 35 00 45 01 65 00 55 00 75 01 95 08 81 02 95 38 81 03 05 08 19 01 29 05 95 05 91 02 95 03 91 03 C1 00 (50 bytes)
  UsagePage 1
  Usage 6
  Collection 1
    UsagePage 7
    UsageMinimum 224
    UsageMaximum 231
    LogicalMinimum 0
    LogicalMaximum 1
    PhysicalMinimum 0
    PhysicalMaximum 1
    Unit 0
    UnitExponent 0
    ReportSize 1
    ReportCount 8
[... 499 lines cut ...]
    LogicalMinimum 0
    LogicalMaximum 255
    PhysicalMinimum 0
    PhysicalMaximum 0
    Unit 0
    UnitExponent 0
    ReportSize 8
    ReportCount 16
    Input 2
    Usage 2
    Output 2
  EndCollection 0
Usage: FF000001 4278190081
Input: ReportID=64, Length=17, Items=1
  16 Elements x 8 Bits, Units: None, Expected Usage Type: 0, Flags: Variable, Usages: FF000002 4278190082
Output: ReportID=64, Length=17, Items=1
  16 Elements x 8 Bits, Units: None, Expected Usage Type: 0, Flags: Variable, Usages: FF000002 4278190082
Opening device for 20 seconds...
Opened device.
Closed device.

Press a key to exit...
```

</deails>

JoergAtGithub, 2021-06-20:
The HidSharp report descriptor reconstructor doesn't implement reconstruction of the collection structure, but #262 does.
Can you please upload the output for this device with #262 too.

Youw, 2021-06-20:
> Windows creates a unique virtual HID device for each usage_page/usage combination of the HID device

> In Windows, the HID device setup class (HIDClass) generates a unique physical device object (PDO) for each Top Level Collection described by the Report Descriptor

I guess I wasn't technically correct, as usage_page/usage combination isn't always a Top Level Collection (I was referencing to my knowledge/experience at the time, instead of check with the documentation), but that doesn't change the root of the problem.

mcuee, 2021-06-20:
> The HidSharp report descriptor reconstructor doesn't implement reconstruction of the collection structure, but #262 does.
> Can you please upload the output for this device with #262 too.

This is the run log from #262. I just use the following branch.
https://github.com/JoergAtGithub/hidapi/tree/windows_report_descriptor_reconstructor

<details>
<summary> click to expand <summary>

```
C:\work\hid\hidapi_test\windows\Debug> .\hidtest.exe
hidapi test/example tool. Compiled with hidapi version 0.10.1, runtime version 0.10.1.
Compile-time version matches runtime version of hidapi.

05 01            Usage Page (1)
09 02            Usage  (2)
A1 01            Collection (Application)
09 01            Usage  (1)
A1 00            Collection (Physical)
85 02            Report ID (2)
05 09            Usage Page (9)
19 01            Usage Minimum (1)
29 10            Usage Maximum (16)
15 00            Logical Minimum (0)
25 01            Logical Maximum (1)
75 01            Report Size (1)
95 10            Report Count (16)
81 02            Input (0x02)
05 01            Usage Page (1)
09 30            Usage (48)
09 31            Usage (49)
16 01 F8         Logical Minimum (-2047)
26 FF 07         Logical Maximum (2047)
75 0C            Report Size (12)
95 02            Report Count (2)
81 06            Input (0x06)
09 38            Usage (56)
15 81            Logical Minimum (-127)
25 7F            Logical Maximum (127)
75 08            Report Size (8)
95 01            Report Count (1)
81 06            Input (0x06)
05 0C            Usage Page (12)
0A 38 02         Usage (568)
15 81            Logical Minimum (-127)
25 7F            Logical Maximum (127)
75 08            Report Size (8)
95 01            Report Count (1)
81 06            Input (0x06)
C0               End Collection
C0               End Collection
### Wrote report descriptor file: hid_report_descriptor_0_0_dev0.txt ###
05 0C            Usage Page (12)
09 01            Usage  (1)
A1 01            Collection (Application)
85 03            Report ID (3)
19 01            Usage Minimum (1)
2A 8C 02         Usage Maximum (652)
15 01            Logical Minimum (1)
26 8C 02         Logical Maximum (652)
75 10            Report Size (16)
95 02            Report Count (2)
81 00            Input (0x00)
C0               End Collection
### Wrote report descriptor file: hid_report_descriptor_0_0_dev1.txt ###
05 01            Usage Page (1)
09 06            Usage  (6)
A1 01            Collection (Application)
05 07            Usage Page (7)
19 E0            Usage Minimum (224)
[... 442 lines cut ...]

Device Found
  type: 2bf6 a096
  path: \\?\hid#hidclass&col01#1&2d595ca7&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: Acer Inc.
  Product:      Launch Manager Wireless Device
  Release:      1
  Interface:    -1
  Usage (page): 0x1 (0xff00)

Device Found
  type: 2bf6 a096
  path: \\?\hid#hidclass&col02#1&2d595ca7&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: Acer Inc.
  Product:      Launch Manager Wireless Device
  Release:      1
  Interface:    -1
  Usage (page): 0xc (0x1)

unable to open device
```

</details>

mcuee, 2021-06-20:
BTW there are no answers here in OSR forum. I guess this is a real difficult question.
https://community.osr.com/discussion/292947/how-to-get-hid-report-descriptor-using-windows-api

JoergAtGithub, 2021-06-20:
> 
> 
> I guess I wasn't technically correct, as usage_page/usage combination isn't always a Top Level Collection (I was referencing to my knowledge/experience at the time, instead of check with the documentation), but that doesn't change the root of the problem.

This is a serious issue for cross platform software based on HIDAPI in general, because such an device behaves different on Linux/Mac vs. Windows. The report descriptor is only one aspect were this unsolved issue appears again.
A proper fix would require, to group these virtual devices during device enumeration and represent them to the API as the same single HID device as on other platforms.

mcuee, 2021-06-20:
One potential way to sort out the issue is to create a libusb backend for Windows. It will have limited usage for people who are willing to use WinUSB driver to replace the Windows HID driver (and even the USB Composite Device Parent driver if that is the case)..

Youw, 2021-06-20:
> A proper fix would require, to group these virtual devices during device enumeration

We're doing qite the opposite for other platforms: we same device multiple times, so each such unique device has usage_page/usage filled up.

And it kind of bits off a general desing of HIDAPI - to be as simple as possible and just be a thin proxy over an OS-specific API/implementation, just enough to keep the uniform API across different platforms. (libusb backend is kind-of the most complicated in this matter, as it practically implements its own USB HID driver, but let it be the only exception.)

---
What I want to suggest is start with a question:
- _Why would we need to report a device descriptor at all?_

If I understand the initial problem correctly, a user application just wants to know the information about the reports, e.g.:
- number of reports;
- reports IDs;
- report details (size, min/max values, etc.);

This a bit out of scope of the initial HIDAPI design as well, as Alan always mentioned, that if you want to use HIDAPI, you probably already know the device you want to work with, and its API/reports etc. Clearly there are cases when a used still needs more than that (I guess `Mixxx DJ SW` is an example of such), and maybe we thing about this problem from a different angle:
- _Why don't we just provide an API to provide reports information?_

E.g. as "_HIDAPI specific data structure_", with a number of input/output reports and its characteristics.

That would require parsing a HID descriptor on all platforms except Windows (parsing is already partially implemented, maybe not for everything, but still), and on Windows we would fill such a structure directly from WinAPI. I haven't looked at the details, but unless there is something Windows doesn't provide, that a used application might need (the "_HIDAPI specific data structure_" would need to contain), that should do it in a much simpler manner.

It would also make things simpler for all user applications, as there is no need to perform HID descriptor parsing.

mcuee, 2021-06-20:
To me the goal may need to be a bit modest: to get REPORT_ID, REPORT_COUNT, and REPORT_SIZE correct.

[75 more replies]

PR #451 (2023-03-12): RFC: add hid_get_report_descriptor API function
- [x] Add [ASAN](https://docs.microsoft.com/en-us/cpp/sanitizers/asan?view=msvc-170) checks for utest/MSVC build;

Resolves: #249
closed 2023-03-12 by Youw