libusb/hidapi #258: hidraw support for hid_get_input_report + missing other combinations

smunaut, 2021-04-07:
Looking at the hidraw API I see thos ioctl :

HIDIOCSINPUT
HIDIOCGINPUT
HIDIOCSOUTPUT
HIDIOCGOUTPUT

which is exactly the functions I'd need for a device I have here but : 

 * hid_get_input_report  is the only function I see in the API and it's not implemented for hidraw
 * There is no function for the other combination like hid_set_input_report / hid_get_output_report / hid_set_output_report

Youw, 2021-04-07:
hid_set_output_report = hid_write

hid_get_output_report/hid_set_input_report - I don't believe HID standard defines this operations. USB implementation defines it, but others may not. Windows backend definitely doesn't support those.
macOS backend technically could implement such thing, but I'm not sure whether it is functional.
And of course, the device itself has to support such operation.

Please keep in mind, that hidapi is a cross-platform library in a first place, and it is not intended to implement platform-specific things or non-standard extensions.

---
As for get_input_report implementation on hidraw backend - Linux kernel itself didn't support it as far as [4 months ago](https://github.com/torvalds/linux/commit/f43d3870cafa2a0f3854c1819c8385733db8f9ae), so that one is yet to be added to hidapi.

smunaut, 2021-04-07:
hid_set_output_report - But won't hid_write the OUT interrupt EP if a device has such a thing instead of doing a control transfer ?
hid_get_input_report  - Yeah, right it's recent, I didn't check the history of the kernel to see when it was done, I just opened the latest hidraw doc.

Anyway, feel free to close, turns out that I misread the trace ... (the raw hw capture usb trace showed a SET/GET_REPORT but I missed the fact the report type was set to feature  :facepalm: ).

Youw, 2021-04-07:
> But won't hid_write the OUT interrupt EP if a device has such a thing instead of doing a control transfer ?

That's what I called "more flexible". I find interrupt endpoints more prefereble for cases where they can be used.

---
Anyway, thanks for pointing out to a new Kernel API - I haven't checked for it recently.

closed 2021-04-07 by Youw