libusb/hidapi #264: configure.ac:16: error: AC_CONFIG_MACRO_DIR can only be used once

autoantwort, 2021-05-01:
When running `./bootstrap` I get the following error:

```
➜  api-0.10.1-7bb92304db.clean git:(master) ✗ ./bootstrap         
+ autoreconf --install --verbose --force
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
configure.ac:16: error: AC_CONFIG_MACRO_DIR can only be used once
./lib/autoconf/general.m4:1970: AC_CONFIG_MACRO_DIR is expanded from...
configure.ac:16: the top level
autom4te: error: /usr/local/opt/m4/bin/m4 failed with exit status: 1
aclocal: error: /usr/local/Cellar/autoconf/2.71/bin/autom4te failed with exit status: 1
autoreconf: error: aclocal failed with exit status: 1
```
Was first discovered in microsoft/vcpkg#17231 but I can reproduce this locally.

JackBoosY, 2021-05-02:
Related: https://gitlab.com/gnuwget/wget2/-/issues/547

Youw, 2021-05-02:
This already has been fixed in the latest master: https://github.com/libusb/hidapi/pull/226

autoantwort, 2021-05-02:
> This already has been fixed in the latest master: #226

Ok cool!  
Sorry I have only searched in the issues and not in the PRs to see if the issue exists already.

closed 2021-05-02 by autoantwort

bentwookie, 2022-08-08:
I'm still seeing it.

ozoid, 2024-10-13:
problem still there in 2024.

mcuee, 2024-10-14:
> problem still there in 2024.

Hmm, you may want to post the build log. Are you using the latest version of hidapi? Which OS and version are you using?

ozoid, 2024-10-15:
> > problem still there in 2024.
> 
> Hmm, you may want to post the build log. Are you using the latest version of hidapi? Which OS and version are you using?

I completed a git clone of the repository, 
git clone https://github.com/Dashlane/hidapi.git
then ./bootstrap
then ./configure --enable-static --disable-shared

but this errored with the message: AC_CONFIG_MACRO_DIR can be used only once

I then went and commented line 22 of configure.ac 
AC_CONFIG_MACRO_DIR([m4])

Everything then went swimmingly and was able to configure and build successfully.

Thank you

mcuee, 2024-10-15:
> git clone https://github.com/Dashlane/hidapi.git

That is a wrong repo and super old verion of hidapi.