libusb/hidapi #265: Sourcehut CI transition to paid service

Qbicz, 2021-05-01:
Today I received an email from Sourcehut (we use it for building Linux and FreeBSD versions of hidapi):

```
Hello! I am the sysadmin at sourcehut, and you're receiving this email
because you submitted a build job at builds.sr.ht in the past 30 days
which would have been affected by an upcoming service change.

In short, we are requiring users to have a paid account before
submitting jobs with builds.sr.ht going forward. There is a 30 day grace
period where existing accounts (like yours) can still submit builds, but
from June 1st forward you will be required to have a paid sr.ht account
to submit builds.

You can set up billing here:

https://meta.sr.ht/billing

Please note that it is NOT our intention to price anyone out! We have
made this change to combat abuse on our services. If you require
financial assistance to afford your sr.ht plan, please reply to this
email explaining your circumstances and you WILL be given free service.

Feel free to reply to this email if you have any other questions
besides. For additional background on the change, details are available
here:

https://man.sr.ht/ops/builds.sr.ht-migration.md

Thank you for using sourcehut!
```

My reply:
```
Hi!

It's nice to hear from you. We are using sr.ht for building an open-source project HIDAPI: https://github.com/libusb/hidapi.

HIDAPI is an open-source, multi-platform library which allows an application to interface with USB and Bluetooth HID-Class devices on Windows, Linux, FreeBSD, and macOS.

Since this is an open source non-profit project maintained by a community of volunteers, we would like to continue building with sourcehut for free. The account qbicz on sr.ht is used only for this.

Please let us know on the next steps.

Kind regards,
Filip
```
 
I will keep you informed what's the outcome.

Youw, 2021-05-01:
Thanks for tha update.

If anything, it will be a trigger to migrate to Github actions. A quick googling brought up this: https://github.com/marketplace/actions/freebsd-vm

Qbicz, 2021-05-01:
```
Hi Filip!

Most projects on sr.ht are open source :) we still expect them to pay
for the service if they are able. If your maintainers can afford to,
they should pay for an account. Just one, though, if you pay for ~qbicz
I'll add a note that it's used for your whole organization.

If the maintainers cannot afford an account, I will issue you free
service.
```

Youw, 2021-05-02:
I say migrating to some other service, like Github Actions, would be a simplest step for an OpenSource project like hidapi.

z3ntu, 2021-05-03:
If you want to, I do have a paid sr.ht account that I can provide for hidapi. I also plan to continue paying for sr.ht for the next couple of years, but it's understandable if you don't want to use it anymore.

The abuse they're talking about in the first email is unfortunately crypto miners that spam the service with builds...

Youw, 2021-05-09:
I couldn't find any reasons why we wouldn't want to accept the offer of hosting sr.ht builds by you. It definitely saves us time of migrating the builds to a different infrastructure (at least for now).

Qbicz, 2021-05-10:
To me this sounds like a good idea, we can move the builds to your paid account.

@z3ntu How do we do it? There are no changes to do in `.builds` directory, so I understand you should just login to your sr.ht account and add the hidapi (as a hidapi admin). This will create a webhook in the Github repository (`Settings -> Webhooks`). Then I can remove the build from my sr.ht free account.

Correct?

z3ntu, 2021-05-10:
Yes I believe that should work.
I do need admin permission to the repo afaik https://git.sr.ht/~sircmpwn/dispatch.sr.ht/tree/master/item/dispatchsrht/tasks/github/github_commit_to_build.py#L102

z3ntu, 2021-05-10:
I've removed all sr.ht webhooks and the old tasks on dispatch.sr.ht and reconfigured the hooks for commits and PRs again.

Everything seems to work, for PRs see #268 and a direct push (branch is deleted again now) also works: https://builds.sr.ht/~z3ntu/job/504409

And I've "Removed yourself as a collaborator of libusb/hidapi"

PR #269 (2021-05-11): Switch builds.sr.ht from ~qbicz to ~z3ntu
Closes: #265
closed 2021-05-11 by Youw