libusb/hidapi #266: Mac OS vendor defined usage page not recognized

adriencarbonaro, 2021-05-08:
Hi,

I use node-hid on Windows 10 + Mac OS 11 (Big Sur) to recognize a HID device. On Windows, everything is working fine but on Mac OS, `HID.devices()` only return the 2 standard usage pages, not my vendor defined usage page `0xFF00`.

I've seen information saying that keyboard devices are owned by OS for security reasons, I confirm my device is not keyboard or mouse, and the usage page `0xFF00` is vendor defined.

Here are more information about the device :

**HID Descriptor**
```
Descriptor Version Number:   0x0111
Country Code:   0
Descriptor Count:   1

Type:   0x22  (Report Descriptor)
Length (and contents):   148
Raw Descriptor (hex)    0000: 05 0C 09 01 A1 01 85 0A  15 00 25 01 75 01 95 01  
Raw Descriptor (hex)    0010: 09 E9 81 06 09 EA 81 06  09 CD 81 06 75 05 81 01  
Raw Descriptor (hex)    0020: C0 05 0B 09 05 A1 01 85  0B 75 01 09 20 81 22 09  
Raw Descriptor (hex)    0030: 2F 81 06 09 21 81 06 09  07 05 09 75 01 95 01 81  
Raw Descriptor (hex)    0040: 02 75 04 81 01 05 08 15  00 25 01 09 17 09 09 09  
Raw Descriptor (hex)    0050: 18 09 20 75 01 95 04 91  22 05 0B 15 00 25 01 09  
Raw Descriptor (hex)    0060: 9E 75 01 95 01 91 22 75  01 95 03 91 01 C0 06 AC  
Raw Descriptor (hex)    0070: FF 09 00 A1 FF 09 01 15  00 26 FF 00 75 08 85 0C  
Raw Descriptor (hex)    0080: 95 3F 81 22 09 02 15 00  26 FF 00 75 08 85 0D 95  
Raw Descriptor (hex)    0090: 3F 91 22 C0 

Parsed Report Descriptor:
Usage Page    (Consumer)
[...] (This usage page is found both Windows and MacOS)

Usage Page    (Telephony Device) 
[...] (This usage page is found both Windows and MacOS)

Usage Page (Vendor defined 0xFF00) (This usage page is not found on MacOS)
Usage 0 (0x0)
Collection (Collection )    
    Usage 1 (0x1)    
    Logical Minimum.........    (0)  
    Logical Maximum.........    (255)  
    Report Size.............    (8)  
    ReportID................    (0x0C)  
    Report Count............    (63)  
    Input...................    (Data, Variable, Absolute, No Wrap, Linear, No Preferred, No Null Position, Bitfield) 
    Usage 2 (0x2)    
    Logical Minimum.........    (0)  
    Logical Maximum.........    (255)  
    Report Size.............    (8)  
    ReportID................    (0x0D)  
    Report Count............    (63)  
    Output..................    (Data, Variable, Absolute, No Wrap, Linear, No Preferred, No Null Position, Nonvolatile, Bitfield) 
End Collection

Endpoint 0x83 - Interrupt Input
Address: 0x83  (IN)
Attributes: 0x03  (Interrupt)
Max Packet Size: 64
Polling Interval: 1 ms

Endpoint 0x02 - Interrupt Output
Address: 0x02  (OUT)
Attributes: 0x03  (Interrupt)
Max Packet Size: 64
Polling Interval: 1 ms
```

**`HID.devices()` return on MacOS**
```
{
    vendorId: 123,
    productId: 456,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/MyDevice@14200000/IOUSBHostInterface@3/AppleUserUSBHostHIDDevice',
    serialNumber: 'ABCDEF',
    manufacturer: 'Me',
    product: 'MyDevice',
    release: 512,
    interface: 3,
    usagePage: 12,
    usage: 1
  },
  {
    vendorId: 123,
    productId: 456,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14200000/MyDevice@14200000/IOUSBHostInterface@3/AppleUserUSBHostHIDDevice',
    serialNumber: 'ABCDEF',
    manufacturer: 'Me',
    product: 'MyDevice',
    release: 512,
    interface: 3,
    usagePage: 11,
    usage: 5
  },
```
**`HID.devices()` return on Windows**
```
{
    vendorId: 123,
    productId: 456,
    path: '\\\\?\\hid#vid_0d8c&pid_0100&mi_03&col01#9&2ea62a40&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: 'ABCDEF',
    manufacturer: 'Me',
    product: 'MyDevice',
    release: 512,
    interface: 3,
    usagePage: 12,
    usage: 1
  },
  {
    vendorId: 3468,
    productId: 256,
    path: '\\\\?\\hid#vid_0d8c&pid_0100&mi_03&col02#9&2ea62a40&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: 'ABCDEF',
    manufacturer: 'Me',
    product: 'MyDevice',
    release: 512,
    interface: 3,
    usagePage: 11,
    usage: 5
  },
  {
    vendorId: 123,
    productId: 456,
    path: '\\\\?\\hid#vid_0d8c&pid_0100&mi_03&col03#9&2ea62a40&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: 'ABCDEF',
    manufacturer: 'Me',
    product: 'MyDevice',
    release: 512,
    interface: 3,
    usagePage: 65280
  }
```
Thank you for your help and for this lib.

Youw, 2021-05-09:
That's interesting. I have several devices with vendor-defined usage/pages and those work properly on Windows/Linux/macOS.

Can you confirm which version of hidapi do you use? the biggest euqstion is either https://github.com/libusb/hidapi/commit/fd53f39f2594e56d63881bf873f69425a566a158 included or not.

> HID Descriptor

How exactly did you read it?

Can you try https://github.com/todbot/hidapitester to see if there're any difference?

There is platform-specific tools to read HID descriptor: https://github.com/todbot/mac-hid-dump, https://github.com/todbot/win-hid-dump. Can you share its output for you device?

adriencarbonaro, 2021-05-10:
> Can you confirm which version of hidapi do you use? the biggest euqstion is either fd53f39 included or not.

I'll check and let you know.

> How exactly did you read it?

I read it with TDD on windows, and USBProbber on macOS

Youw, 2021-05-12:
> Do you know what could cause this ?

Don't know. Maybe the way your custom Usage/Page is defined is unparsable by macOS driver, or something else.

Do you have the ability to debug step-by-step C code? Would be interesting to see what's going on inside [hid_enumerate](https://github.com/libusb/hidapi/blob/master/mac/hid.c#L596) and [create_device_info](https://github.com/libusb/hidapi/blob/master/mac/hid.c#L514). If it is HIDAPI code that is failing to recognize custom usage/page - we can fix that. But if it doesn't reported by the macOS driver - you'll need to shuffle your descriptor definition to the point when it is recognized by macOS driver correctly.

---
P.S.: when you edit your existing comment no one is getting notifications. I've noticed it by accident only because I had a chrome tab opened here.

mcuee, 2021-08-02:
@adriencarbonaro You can always post the output of USB Prober vs hidtest output.

BTW, TDD under Windows will not be able to dump the HID report descriptor. So you have to use macOS USB Prober.

Youw, 2023-03-12:
I do not see any indication that HIDAPI implementation is related to the issue originally described by the author.
And don't see how can we help here.
Closing.

closed 2023-03-12 by Youw