libusb/hidapi #272: Installing hidapi on Raspberry Pi

simonsolnes, 2021-05-14:
After following the instructions:
```bash
git clone https://github.com/libusb/hidapi/
cd hidapi
./bootstrap
./configure
# then sudo apt install the libs that configure tells you to install
./configure
make
sudo make install
```
also installing hidapi for python:
```
pip3 install hid
```

I get this error when hidapi for python tries to 'access' hidapi:
```bash
$ python3 -c 'import hid'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/pi/.local/lib/python3.7/site-packages/hid/__init__.py", line 30, in <module>
    raise ImportError(error)
ImportError: Unable to load any of the following libraries:libhidapi-hidraw.so libhidapi-hidraw.so.0 libhidapi-libusb.so libhidapi-libusb.so.0 libhidapi-iohidmanager.so libhidapi-iohidmanager.so.0 libhidapi.dylib hidapi.dll libhidapi-0.dll
```

I know this is not a tech support forum, but I the reason I'm posting the issue, is that this is not documented as far as I know.

simonsolnes, 2021-05-14:
hidtest works perfectly, so this is just a case of me not understanding how to 'install'/'link' the libraries. I know this isn't StackOverflow, but some directions would be greatly appreciated.

todbot, 2021-05-14:
Python is not looking in `/usr/local/lib` for shared libraries for some reason, because this does work:
`LD_LIBRARY_PATH=/usr/local/lib python3 -c 'import hid'`

I'm not very Python knowledgable to know if this is expected or not.  You can "fix" hidapi to install in `/usr` instead:
```
cd hidapi
sudo make uninstall # remove installation in /usr/local
./configure --prefix=/usr
make
sudo make install
python3 -c 'import hid'
```

simonsolnes, 2021-05-18:
Thank you for the reply, but I've since abandoned the project because of new hardware.

closed 2021-06-10 by mcuee