libusb/hidapi #275: [hidapi] Broken Pipe on Unifying Receiver

Ledz96, 2021-05-17:
Hi everyone,
I am currently attempting to send HID reports to a Unifying Receiver through libusb, but I am having no success.

I tried both with hidapi (code compiled from the 'hidtest.cpp' file given) and with plain libusb (using Set_Report Request on endpoint 0), but what I get in both cases is a BROKEN_PIPE error.
Other kind of HID control transfers seem to work just fine with the device, so the HID protocol seems to be the problem here.

What could the problem be?

Youw, 2021-05-17:
Are you trying to communicate with Unifying Receiver itself, or with a Mouse/Keyboard connected thru it?

Ledz96, 2021-05-17:
> Are you trying to communicate with Unifying Receiver itself, or with a Mouse/Keyboard connected thru it?
> 
> I assume you mean https://en.wikipedia.org/wiki/Logitech_Unifying_receiver

Exactly, this is the device. I am trying to communicate with the receiver itself, not a mouse/keyboard connected to it.

Youw, 2021-05-17:
And you know the protocol that device uses? I haven't seen any public API for it. Only Windows/macOS applications.

Ledz96, 2021-05-17:
Apparently it uses some proprietary protocol (hidpp) that is built on top of HID. I could successfully communicate with it through [this tool](https://github.com/cvuchener/hidpp). I am however interested in using plain and simple hid.
I simply put the payload of the HID++ command in the HID data field, and filled the rest according to 7.2.2 of [this document](https://www.usb.org/sites/default/files/hid1_11.pdf). I am not too familiar with HID per se, so I might be missing something here.

Youw, 2021-05-17:
Do you have a sample of hidpp code that works, vs your sample of hidapi that doesn't?

mcuee, 2021-06-15:
@Ledz96 

By right you should be able to do the same with HIDAPI juding from the following codes.
https://github.com/cvuchener/hidpp/blob/master/src/libhidpp/hid/RawDevice_windows.cpp

You may try hidapitester which uses hidapi to play with the device.
https://github.com/todbot/hidapitester

mcuee, 2021-06-21:
@Ledz96 Close this for now. It is a pretty complicated device and it may not be easy to get it right if you are not familiar with HIDAPI.

Please reopen if you find issues with HIDAPI.

closed 2021-06-21 by mcuee