libusb/hidapi #278: MacOS panic when hid_write called with a zero length buffer.

JnyJny, 2021-05-19:
## Synopsis: writing a zero (although wrong) results in a MacOS panic

### Software Versions
```console
$ sw_vers
ProductName:	macOS
ProductVersion:	11.3.1
BuildVersion:	20E241

$ python --version
Python 3.9.4

# https://github.com/trezor/cython-hidapi
$ poetry show | grep hid
hidapi               0.10.1    A Cython interface to the hidapi from https://github.com/libusb/hidapi
```

### Reproducer
```python
#!/usr/bin/env python3

import hid

path = hid.enumerate()[0]["path"]
device = hid.device()
device.open_path(path)
device.write(0)
# panic
```

### Discussion

I know this is a case of "don't do that" but I was completely surprised that writing a zero caused a panic (several panics since I did it **a bunch** to confirm the cause of the panic).

I realize this problem begins in `python` but I'm reasonably confident the panic happens in `hidapi`.

Starting at [hid.pyx:device.write](https://github.com/trezor/cython-hidapi/blob/b94c8a09582681aff6edd145a3bfd3948ff7e5ca/hid.pyx#L136), a zero is transliterated into a `bytes` object with zero length and calls [hid_write](https://github.com/libusb/hidapi/blob/c7a2339b0530aa63814aee824c04872e89613a02/mac/hid.c#L961), a wrapper around [set_report](https://github.com/libusb/hidapi/blob/c7a2339b0530aa63814aee824c04872e89613a02/mac/hid.c#L896), which then dereferences (a possibly NULL) buffer pointer. Presuming the buffer point is not null,  the buffer pointer and length are then passed to `IOHIDDeviceGetReport`. The panic report (see below) doesn't seem to give much clue whether it panic'ed in `set_report` or `IOHIDDeviceGetReport`.  My _guess_ is `IOHIDDeviceGetReport`. 

### Desired Outcome 
I would prefer to be gently chided with a return value of `-1 ` and `errno=EBADPROGRAMMER` when I call `hid_write` with a zero length buffer. 

### Panic Report 
[Full panic report](https://gist.github.com/JnyJny/7ad2bed77183555b8d1db065ef6b7ad8) 

#### Extracted panicString
```
panic(cpu 0 caller 0xfffffff026ebf448): macOS panic detected
Debugger message: panic
Memory ID: 0x6
OS release type: User
OS version: 18P4556
macOS version: 20E241
Kernel version: Darwin Kernel Version 20.4.0: Thu Mar  4 23:34:36 PST 2021; root:xnu-7195.101.1~1/RELEASE_ARM64_T8010
Kernel UUID: 3FA744A7-0F34-36DA-AA6D-F6503A3176ED
iBoot version: iBoot-6723.101.4
secure boot?: YES
x86 EFI Boot State: 0x16
x86 System State: 0x0
x86 Power State: 0x0
x86 Shutdown Cause: 0xec
x86 Previous Power Transitions: 0x70707060400
PCIeUp link state: 0x1614
Paniclog version: 13
Kernel slide:      0x000000001ef74000
Kernel text base:  0xfffffff025f78000
mach_absolute_time: 0xf015ab22b
Epoch Time:        sec       usec
  Boot    : 0x60a53289 0x0002bba3
  Sleep   : 0x00000000 0x00000000
  Wake    : 0x00000000 0x00000000
[... 218 lines cut ...]
|IOUSBMass!SDriver	184.101.1
|IOSCSIBlockCommandsDevice	436.100.4
|IO!S!F	2.1
|IOSCSIArchitectureModel!F	436.100.4
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!ACredentialManager	1.0
>KernelRelayHost	1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
>!AACPIPlatform	6.1
>!ASMC	3.1.9
|IOPCI!F	2.9
|IOACPI!F	1.4
>watchdog	1
@kec.pthread	1
@kec.corecrypto	11.1
@kec.Libm	1
```

Youw, 2021-05-19:
> https://github.com/libusb/hidapi/blob/master/hidapi/hidapi.h#L246

> The first byte of @p data[] must contain the Report ID. For			devices which only support a single report, this must be set to 0x0.

So this _is_ a (documented) case of "don't do that", but if you'd care to prepare a PR for an explicit check that would be great too.

JnyJny, 2021-05-19:
Ouch, I missed that in my googling. Thank you for the link. I've also started an issue with the cython-hidapi project to put in a guard at the python level and will knock together a PR here too.

mcuee, 2021-06-13:
This may be related to my experiences as well using hidapitester and hidpytoy (as replacement for the fox based Test GUI).
Ref: https://github.com/libusb/hidapi/issues/187

mcuee, 2021-06-13:
cython-hidapi issue here.
https://github.com/trezor/cython-hidapi/issues/116

PR #279 (2021-06-14): Guard against zero length buffers in hid_write
This is my proposed solution for avoiding a macOS panic in `set_report` when the user supplies a zero length buffer (indicated by either a NULL data or a length of 0 or less).

Looking back at the source, I see that `length` is of type `size_t` which my foggy brain remembers being unsigned so the test for `less than or equal to zero` should likely be `equal zero`. 

Fixes: #278
closed 2021-06-14 by Youw