libusb/hidapi #281: hid_write return -1

natxopedreira, 2021-05-27:
Sorry to ask but im not able to write to device..... and i dont find why.

Im able to read perfect and works great but im not able to make write works

any clue? error verbose did not help much here

`unsigned char data[65];
data[0] = 0x00;
data[1] = 0x10;
data[2] = 0x01;
data[3] = 0x09;
data[4] = 0x1a;
data[5] = 0x00;
data[6] = 0xe8;
data[7] = 0x80;
       
res = hid_write(handle_boton, data, 65);
cout << "vibbbra" << res << endl;`

Youw, 2021-05-27:
What makes you think it shouldn't fail?
What platform are you on?
What version of hidapi do you use?
What kind of device is this?
Do you have spec for the device/its API?

natxopedreira, 2021-05-27:
Device is a logitech remote spotlight, there is a project based on that device https://github.com/jahnf/Projecteur and there it writes to that one so im trying to do that on hidapi

`What makes you think it shouldn't fail?` 
I dont know i think at least i will be able to write to device... maybe im wrong

` What platform are you on?`
linux

` What version of hidapi do you use?`
fresh built from repo

` Do you have spec for the device/its API?`
Nope, im getting the command from the other repo, i assume the command is common independent of the lib is using

Youw, 2021-05-27:
> linux

libhidapi-hidraw or libhidapi-libusb?

>  im getting the command from the other repo

Are you repeating the command exactly? Do you have a reference how the other project does it?

```c
unsigned char data[65];
data[0] = 0x00;
// ...
hid_write(handle_boton, data, 65);
```
this means you're sending an output report to the device of size 64 bytes (and device should expect this), which doesn't use indexed reports (`data[0]` is a report ID, and 0 means IDs aren't used).

natxopedreira, 2021-05-27:
sorry libhidapi-hidraw 

`Are you repeating the command exactly? Do you have a reference how the other project does it?`
Yep im repeating the command exactly

`unsigned char vibrate[] = {0x10, 0x01, 0x09, 0x1a, 0x00, 0xe8, 0x80};`

[https://github.com/jahnf/Projecteur/blob/dfb52bb0b86189ac5cc9383011e0028d2271ccb4/src/device-vibration.cc#L425](https://github.com/jahnf/Projecteur/blob/dfb52bb0b86189ac5cc9383011e0028d2271ccb4/src/device-vibration.cc#L425)

i think that to be literal i should do that (still returns -1) 0x050 is the length

```
const uint8_t vibrateCmd[] = {0x10, 0x01, 0x09, 0x1a, 0x050, 0xe8, 0x80};
 res = hid_write(handle_boton, vibrateCmd, sizeof(vibrateCmd));
 cout << "vibbbra" << res << endl;
```

In my mind i expect to get hid_write to return non -1 even the command do not work, but maybe im wrong

I dont know Qt, he does like that
```
// unsigned char vibrate[] = {0x10, 0x01, 0x09, 0x1a, 0x00, 0xe8, 0x80};
const uint8_t vlen = m_sbLength->value();
const uint8_t vint = m_sbIntensity->value();
const uint8_t vibrateCmd[] = {0x10, 0x01, 0x09, 0x1a, vlen, 0xe8, vint};
const auto notifier = m_subDeviceConnection->socketNotifier();
const auto res = ::write(notifier->socket(), &vibrateCmd[0], sizeof(vibrateCmd));
```

Youw, 2021-05-27:
> const uint8_t vibrateCmd[] = {0x10, 0x01, 0x09, 0x1a, 0x050, 0xe8, 0x80};
 res = hid_write(handle_boton, vibrateCmd, sizeof(vibrateCmd));

This looks more like what you should do.

If it still fails, and you are shure that you've opened the correct device, all I can suggest is to check `errno` after `hid_write` (just to check what the failure is).

Other than that, I don't have other suggestions.

natxopedreira, 2021-05-27:
ok, thank you!!! will do what you say

natxopedreira, 2021-05-31:
thanks for your help
is working!!!! problem was that device showed several interfaces and only one worked for vibration

closed 2021-05-31 by natxopedreira