libusb/hidapi #282: No product name when using Bluetooth classic HID in Windows10

LeeChunHei, 2021-05-28:
I am using the python wrapper of this library, [pyhidapi](https://github.com/apmorton/pyhidapi)
when I check the product_string of the Bluetooth HID device, it returns None.
I noticed that in windows this library is using [HidD_GetProductString](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/hidsdi/nf-hidsdi-hidd_getproductstring) to get the name of the HID device.
So I had asked in Microsoft community and they [reply](https://docs.microsoft.com/en-us/answers/questions/401236/hidd-getproductstring-with-ble-hid-device.html) this function will not work on BLE device but will work on the classic Bluetooth.
However it still not work after I change to use Bluetooth classic HID device.
Therefore I would like to ask if anyone successfully get the product string of a Bluetooth classic HID Device with Windows10 before.
Or I just messed up with some other stuff.

Youw, 2021-05-28:
I think this may be relevant to you:
https://stackoverflow.com/questions/65970214/where-does-bluetooth-and-other-devices-get-its-information-from#comment116656875_65970214

The other thing that comes to my mind: is your device a Mouse-/Keyboard-like device? In other words - does hid_open_path falls-back to no read/write access (`open_device(path, FALSE)`) ?

LeeChunHei, 2021-05-28:
> The other thing that comes to my mind: is your device a Mouse-/Keyboard-like device? In other words - does hid_open_path falls-back to no read/write access (`open_device(path, FALSE)`) ?

My device is not the mouse-/keyboard-like device, I set the report descriptor to a custom report, and when I try to read write the device, the python program did not warn me with no read/write access.

> I think this may be relevant to you:
> https://stackoverflow.com/questions/65970214/where-does-bluetooth-and-other-devices-get-its-information-from#comment116656875_65970214

This one is interesting, did the question mean that the product name can be seen like this
[image]
But showing generic name in hid device like this
[image]
Then HidD_GetProductString is provided in Win32API instead of WinRT?
And it is possible that in that question, a BLE device is used which the Microsoft guy replied HidD_GetProductString not supported.

Right now what I'm scared about is did I do something wrong on the device side, so it would be better if anyone can test with their classic Bluetooth HID device.

Youw, 2021-05-28:
To verify your device, you could try using macOS version of the library (you'd need mac of course) or hidraw backend on Linux.
Just to isolate is it Windows-specific or not.

LeeChunHei, 2021-05-28:
> To verify your device, you could try using macOS version of the library (you'd need mac of course) or hidraw backend on Linux.
> Just to isolate is it Windows-specific or not.

What I'm scared of is I did something wrong for what Windows required. I had tested with Linux before and it can even get the product name of the BLE device. So my problem is really Windows specific

DJm00n, 2021-06-02:
Just checked **DualShock4** controller connected **via Bluetooth** and can confirm that it is working:
```
Interface Path:  \\?\hid#{00001124-0000-1000-8000-00805f9b34fb}_vid&0002054c_pid&09cc#8&149c8f5&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
HidD_GetManufacturerString: Sony Interactive Entertainment
HidD_GetProductString: Wireless Controller
HidD_GetSerialNumberString: dcaf687546af
HidD_GetAttributes: VendorID: 0x054c, ProductID: 0x09cc, VersionNumber: 0x0000
```

Also just checked **Xbox Series Controller Model 1914** connected **via Bluetooth LE**: only `HidD_GetAttributes` is working fine with that controller.
```
Interface Path: `\\?\HID#{00001812-0000-1000-8000-00805f9b34fb}&Dev&VID_5e&PID_0b13&REV_0505&987a14b7e3a7&IG_00#9&55c66e1&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}`
HidD_GetManufacturerString: returns empty buffer
HidD_GetProductString: returns empty buffer
HidD_GetSerialNumberString: returns empty buffer
HidD_GetAttributes: VendorID: 0x045e, ProductID: 0x0b13, VersionNumber: 0x0505
```

Tests are done on Windows 10 Version 1909. 

This is how it is seen in Windows "Bluetooth & other devices" screen:
[image]

Here is control panel screenshot:
[image]

Here is joy.cpl:
[image]

Seems there is impossible to get "Xbox Wireless Controller XXXXXX" string for Bluetooth LE device via HID API. Also Bluetooth LE APIs are not exposed via classic IOCTLs and only usable via [Windows.Devices.Bluetooth.GenericAttributeProfile WinRT API](https://docs.microsoft.com/uwp/api/windows.devices.bluetooth.genericattributeprofile). See https://github.com/microsoft/Windows-universal-samples/tree/master/Samples/BluetoothLE/cppwinrt for example code.

Youw, 2021-06-02:
This looks like a potential alternative backend for Windows:
https://docs.microsoft.com/en-us/uwp/api/windows.devices.humaninterfacedevice?view=winrt-20348

I woundner if it works with BluetoothLE devices properly, i.e. so we could use Windows.Devices.HumanInterfaceDevice instead of Windows.Devices.Bluetooth.

LeeChunHei, 2021-06-02:
> This looks like a potential alternative backend for Windows:
> https://docs.microsoft.com/en-us/uwp/api/windows.devices.humaninterfacedevice?view=winrt-20348
>
> I woundner if it works with BluetoothLE [...]

I tried this yesterday when I try to build an hidapi version with winrt API 
Somehow I can't find how to use it to perform the action of hid_read
And I can't use winrt API only for windows version hidapi, as we are missing the winrt to get the serial number and manufacturer string 

So right now I'm trying to build a pull request that uses [winrt API](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformation.name?view=winrt-20348) to get the product string if the HidD_GetProductString return a null pointer

Youw, 2021-06-02:
> I can't find how to use it to perform the action of hid_read

That is `GetInputReportAsync()`

> missing winrt API to get the serial number and manufacturer string

I believe that should be possible using Windows.Devices.<something> API

LeeChunHei, 2021-06-02:
> That is `GetInputReportAsync()`
I think this function is related to hid_get_input_report instead of hid_read?

> I believe that should be possible using Windows.Devices. API
Then I'm not quite sure, as I am using the Windows.Devices.Enumerate.DeviceInformation API
And I not quite understand how the Windows.Device API works

Youw, 2021-06-02:
> > That is GetInputReportAsync()
>
> I think this function is related to hid_get_input_report instead of hid_read?

hid_read reads any _input report_ from async queue, reports that device sends asyncronously (interrupt pipe).

get_input_report(report_id, ...) gets a report with specific report ID from the device on demand.

Functionally, GetInputReportAsync(UInt16) matches hid_get_input_report, and GetInputReportAsync() - hid_read.

> I am using the Windows.Devices.Enumerate.DeviceInformation API

Yeap, this looks right.

LeeChunHei, 2021-06-02:
> hid_read reads any _input report_ from async queue, reports that device sends asyncronously (interrupt pipe).
>
> get_input_report(report_id, ...) gets a report with specific report ID from the device [...]

I see
> > I am using the Windows.Devices.Enumerate.DeviceInformation API
> 
> Yeap, this looks right.

Well, this API didn't provide serial number string and manufacturer name.

Youw, 2021-06-02:
> Well, this API didn't provide serial number string and manufacturer name.

Have you tried to print all of Windows.Devices.Enumerate.DeviceInformation.Properties  ?

LeeChunHei, 2021-06-02:
> > Well, this API didn't provide serial number string and manufacturer name.
> 
> Have you tried to print all of Windows.Devices.Enumerate.DeviceInformation.Properties ?

Yes, somehow it only got these 8 keys
System.Devices.ContainerId
System.Devices.InterfaceClassGuid
System.Devices.DeviceInstanceId
System.Devices.InterfaceEnabled
System.Devices.GlyphIcon
System.Devices.IsDefault
System.Devices.Icon
System.ItemNameDisplay

Youw, 2021-06-02:
That's unfortuante.

I guess as an alternative it should be possible to get/open a Windows.Devices.BluetoothDevice from Windows.Devices.Enumerate.DeviceInformation and ask for its [SerialNumberString](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattcharacteristicuuids.serialnumberstring?view=winrt-20348)

LeeChunHei, 2021-06-02:
> That's unfortuante.
>
> I guess as an alternative it should be possible to get/open a Windows.Devices.BluetoothDevice from Windows.Devices.Enumerate.DeviceInformation and ask for its [...]

I don't think so.
First, we don't know whether we are facing a Bluetooth device or a USB device.
Second, with the path of the hid device, I don't think it can open a Bluetooth device or a USB device.

DJm00n, 2021-06-02:
Also you can try these functions for Bluetooth LE device: https://docs.microsoft.com/windows-hardware/drivers/bluetooth/bluetooth-low-energy-functions

LeeChunHei, 2021-06-02:
> Also you can try these functions for Bluetooth LE device: https://docs.microsoft.com/windows-hardware/drivers/bluetooth/bluetooth-low-energy-functions

Thx, I will check it later.
But first I would like to check if the original code can communicate with ble hid first.
If it is possible, then I may just modify windows/hid.c to use winrt API to get the product name when  HidD_GetProductString failed to do its work.

DJm00n, 2021-06-02:
More investigation gives me more info about HID device:
- if device interface path contains `{00001124-0000-1000-8000-00805f9b34fb}` - then its Bluetooth HID device (HID SERVICE spec at bluetooth website).
- if device interface path contains `{00001812-0000-1000-8000-00805f9b34fb}` - then its Bluetooth LE HID device (HID OVER GATT spec at bluetooth website).

Those UUIDs are listed at https://www.bluetooth.com/specifications/assigned-numbers/ as 16-bit values. See "Service Discovery" spec for clarification.

LeeChunHei, 2021-06-03:
> Thx, I will check it later.
> But first I would like to check if the original code can communicate with ble hid first.
> If it is possible, then I may just modify windows/hid.c to use winrt API to get [...]

Okay, so the current API won't work with ble hid
I am thinking of switching winrt API for this library
However I'm stuck on using winrt API in C
Anyone can give me some guide to do it?

Youw, 2021-06-03:
WinRT is available as part of Windows SDK.
Typical location of the headers:
C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\winrt\windows.devices.enumeration.h

When you build environment is ready, you should nornally do:
`#include <winrt\windows.devices.enumeration.h>`

LeeChunHei, 2021-06-03:
> WinRT is available as part of Windows SDK.
> Typical location of the headers:
> C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\winrt\windows.devices.enumeration.h
> 
> When you build environment is ready, you should nornally do:
> `#include <winrt\windows.devices.enumeration.h>`
> 
> NOTE: I believe WinRT is designed more like C++ API (not a C API), and I'm not sure is it usable from C in general.

Yeah, I know about this
But consider the library we are having now is using C, I can't change it to C++
What I am thinking right now is doing something like [this](https://stackoverflow.com/a/7437006)
But I'm still missing some guide to use these API properly (like how to initiate the API, exit the API)

P.S. We need to keep using C because libraries similar to [this](https://github.com/trezor/cython-hidapi) is not using the dll file, they are using the source file directly

DJm00n, 2021-06-03:
@LeeChunHei you can do it but code will be creepy. Example: https://github.com/libsdl-org/SDL/blob/main/src/joystick/windows/SDL_windows_gaming_input.c

LeeChunHei, 2021-06-03:
> @LeeChunHei you can do it but code will be creepy. Example: https://github.com/libsdl-org/SDL/blob/main/src/joystick/windows/SDL_windows_gaming_input.c

Thanks, this example looks useful to me.
I know it will be creepy, I already know that when I saw those long macro names.

Youw, 2021-06-03:
Since we need to go this way only because WinAPI HID API doesn't work with BluetoorhLE devices, maybe it would be much easier to use this instead:
https://docs.microsoft.com/en-us/windows/win32/api/bluetoothleapis/

LeeChunHei, 2021-06-03:
> Since we need to go this way only because WinAPI HID API doesn't work with BluetoorhLE devices, maybe it would be much easier to use this instead:
> https://docs.microsoft.com/en-us/windows/win32/api/bluetoothleapis/

I think it's a bit hard to use this to communicate with the ble hid device?
As we need to deal with those hid protocol by ourselves.
And we also need to combine it with normal USB hid device.
Anyway, I will check if the winrt HID API can communicate with the BLE device, if everything goes well, I will try to make a pull request about changing to winrt API (May takes forever, as those API in C is so creepy)

Youw, 2021-06-03:
I don't suggest using ble API for everything - only to get Manufacturer/Serial/Product strings if we detect it is a BleLE device.
If I understand it correctly - everything else works fine with standard WinAPI HID API.

Youw, 2021-06-03:
> changing to winrt API

NOTE: we can only accept WinRT API as an _alternative_ backend, not a replacement.

LeeChunHei, 2021-06-03:
> I don't suggest using ble API for everything - only to get Manufacturer/Serial/Product strings if we detect it is a BleLE device.
> If I understand it correctly - everything else works fine with standard WinAPI HID API.

No, from my test, I don't think we can communicate ble hid device with Win32API

Youw, 2021-06-03:
> I don't think we can communicate ble hid device with Win32API

Because you've tried and it doesn't work?

LeeChunHei, 2021-06-03:
> > I don't think we can communicate ble hid device with Win32API
> 
> Because you've tried and it doesn't work?

Yes, I use the python wrapper of hidapi with my esp32 ble hid device
When I try to write the device, it returns -1 instead of how many byte I wrote

LeeChunHei, 2021-06-03:
> > changing to winrt API
> 
> NOTE: we can only accept WinRT API as an _alternative_ backend, not a replacement.

Is it because you want to keep the compatibility with older version windows?
So I should use a macro to separate two backend?

DJm00n, 2021-06-03:
@LeeChunHei I have some working code for BLE HID device info: https://github.com/DJm00n/RawInputDemo/commit/d718a0a613585e2d74bdea33ce6f1e67cdbac91d

Youw, 2021-06-03:
> When I try to write the device, it returns -1 instead of how many byte I wrote

Can you confirm that your device implements a single usage_page/usage pair or that you open a correct device?

> So I should use a macro to separate two backend?

Ideally - a separate .c file, e.g.: winrt/hid.c

LeeChunHei, 2021-06-04:
> > When I try to write the device, it returns -1 instead of how many byte I wrote
> 
> Can you confirm that your device implements a single usage_page/usage pair or that you open a correct device?
> 
> > So I should use a macro to separate two backend?
> 
> Ideally - a separate .c file, e.g.: winrt/hid.c

I can confirm that my device implement correctly, it can be communicated under linux

mcuee, 2021-06-16:
Is Xbox One S Controller a good test device for this case?

mcuee, 2021-06-16:
So we have two issues issues here:
1) BLE device is not working, need new backend for Windows (winrt/hid.c)
2) Bluetooth classic HID Device product string not working for Windows

LeeChunHei, 2021-06-16:
I think @DJm00n tested that Bluetooth classic HID (DualShock4) can have the product string correctly shown.
So maybe there is no problem with classic HID in Windows, maybe is just my esp32 with btstack have problem to show the product string in Windows
But BLE device is for sure that not work with current HID backend API.

mcuee, 2021-06-16:
I see. Thanks for the clarifications.

DJm00n, 2021-06-16:
@mcuee afaik only Xbox Series Controller Model 1914 can work over BLE connection (HID over GATT). Older versions are working via usual Bluetooth 4 (HID over L2CAP).
 
You can see controller model number on the sticker in battery compartment.

mcuee, 2021-06-16:
Thanks. I have ESP32 devices as well but never really play with Bluetooth. Maybe I will try that as well in the future.

mcuee, 2021-06-16:
Thanks. My one is the old model 1708 so it is not compatible with BLE then. 
> @mcuee afaik only Xbox Series Controller Model 1914 seen with VID:045E PID:0B13 (PID:0B12 over USB) can work over BLE connection (HID over GATT). Older versions are working via usual Bluetooth 4 (HID over L2CAP).
> 
> You can see controller model number on the sticker in battery compartment.

mcuee, 2021-06-20:
This seems another difficult question that my question in OSR NTDEV forum got no answers.
https://community.osr.com/discussion/292943/windows-hid-api-and-bluetooth-low-energy

Youw, 2021-06-25:
> > So I should use a macro to separate two backend?

> Ideally - a separate .c file, e.g.: winrt/hid.c

Oh, and BTW: I think there is absolutely no sense implementing winrt backend for HIDAPI in `C`.
It better be in `C++`. It can be compiled by latest MSVC compilers only (with one of the latest Windows SDK).
The only thing we will get by implementing it in `C` - a complicated and unreadable code.

To be clear: it is possible/fairly easy to have both: pure `C` interface and `C++` implementation.

LeeChunHei, 2021-06-25:
> Oh, and BTW: I think there is absolutely no sense implementing winrt backend for HIDAPI in `C`.
> It better be in `C++`. It can be compiled by latest MSVC compilers only (with one of the latest [...]

XD, right now I'm kinda close to finishing the backend with WinRT with c only, but not doing any testing yet since I'm very busy now.
And I would like to keep using c interface such that it is possible for other programs in the c language to build with it?
For example python wrappers like this [https://github.com/trezor/cython-hidapi](https://github.com/trezor/cython-hidapi).

Youw, 2021-06-25:
> To be clear: it is possible/fairly easy to have both: pure C interface and C++ implementation.

It will allows other languages to integrate just fine.

Youw, 2021-06-25:
> I'm kinda close to finishing the backend with WinRT with c only

Do you have a WIP branch or something?
I can take a quick look and show my suggestions regarding using C++ as the implementation.

LeeChunHei, 2021-06-25:
> > I'm kinda close to finishing the backend with WinRT with c only
> 
> Do you have a WIP branch or something?
> I can take a quick look and show my suggestions regarding using C++ as the implementation.

Ah, no
Because I just do this coding by myself and I'm kinda lazy to do the git stuff.
Maybe I setup it tmr and tells you the branch.

LeeChunHei, 2021-06-26:
@Youw 
This is the code I'm working on now [https://github.com/LeeChunHei/hidapi](https://github.com/LeeChunHei/hidapi)
It's in the winrt folder under master branch as I forgot to create a new branch.
I currently test the code by compiling the dll and paste it into the python installation directory and testing its functionality with [this python package](https://github.com/apmorton/pyhidapi).
I found that if I use the WinRT HID API, a few devices cannot be open (maybe those keyboard mouse stuff?), so there will be less device listed if you call hid_enumerate compare to the original API

DJm00n, 2021-07-22:
PR #309 should fix this issue.

DJm00n, 2021-07-23:
@LeeChunHei could you please test my PR #309?

mcuee, 2021-07-24:
> @Youw
> This is the code I'm working on now https://github.com/LeeChunHei/hidapi
> It's in the winrt folder under master branch as I forgot to create a new branch.
> I currently test the code by compiling [...]

This seems to work reasonably well based on the outputs of hidtest.

mcuee, 2021-07-24:
@LeeChunHei

Which ESP32 BLE HID FW are you using? I found a few here but I have never used Arduino (haha, do not laugh at me) or Espressif's ESP-IDF before.
https://github.com/lemmingDev/ESP32-BLE-Gamepad (and the links the author refers to)
https://github.com/asterics/esp32_mouse_keyboard 
https://github.com/olegos76/nimble_kbdhid_example

LeeChunHei, 2021-07-24:
@mcuee 
I use the BLE hid example inside Espressif's ESP-IDF to build the ESP32 firmware
So i'm not sure about how to use micropython or Arduino to make that firmware

LeeChunHei, 2021-07-24:
@DJm00n 
Sure, but I will need to find some spare time first

mcuee, 2021-07-25:
> I use the BLE hid example inside Espressif's ESP-IDF to build the ESP32 firmware
> So i'm not sure about how to use micropython or Arduino to make that firmware

Thanks. I just downloaded ESP-IDF onto my Mac Mini M1 and then I can build the default BLE HID example. Unfortunately it does not seem to work on my cheap ESP32 board -- my Mac can not connected to it.

```
I (560) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (649) BTDM_INIT: BT controller compile version [0e42119]
I (649) system_api: Base MAC address is not set
I (649) system_api: read default base MAC address from EFUSE
I (659) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (1359) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
I (1389) HID_DEV_DEMO: START
I (823139) HID_DEV_DEMO: CONNECT
W (823289) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x13
E (823289) BT_BTM: Device not found

E (823289) BT_APPL: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 5c:f3:70:8a:6d:48
I (823299) HID_DEV_DEMO: DISCONNECT: TERMINATE_PEER_USER
E (823299) ESP_HID_GAP: BLE GAP AUTH ERROR: 0x66

```

mcuee, 2021-07-25:
@LeeChunHei I have changed the title to remove the "product name for Bluetooth classic HID in Windows 10" portion, as it seems to work well with the current HIDAPI codes, based on the testing results from @DJm00n . You can refer to my Logitech Bluetooth Mouse output which also shows that the current HIDAPI code is working. So that portion is probably just related to ESP32.

mcuee, 2021-07-25:
@LeeChunHei Interestingly the ESP32 BLE HID device has no issues under my Windows 10 20H2 Dell Laptop.

@DJm00n Output from #309 seems to show that it is working.

```
Device Found
  type: 16c0 05df
  path: \\?\hid#{00001812-0000-1000-8000-00805f9b34fb}_dev_vid&0216c0_pid&05df_rev&0100_30aea44c4efa#9&38bb9e5a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 30aea44c4efa
  Manufacturer: Espressif
  Product:      ESP BLE HID2
  Release:      100
  Interface:    -1
  Usage (page): 0x100 (0xff00)
  Bus type: 2

Device Found
  type: 16c0 05df
  path: \\?\hid#{00001812-0000-1000-8000-00805f9b34fb}_dev_vid&0216c0_pid&05df_rev&0100_30aea44c4efa#9&f97cf65&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 30aea44c4efa
  Manufacturer: Espressif
  Product:      ESP BLE HID2
  Release:      100
  Interface:    -1
  Usage (page): 0x1 (0xc)
  Bus type: 2

```

mcuee, 2021-07-25:
@LeeChunHei Your WinRT branch seems to work fine as well.

```
Device Found
  type: 16c0 05df
  path: \\?\HID#{00001812-0000-1000-8000-00805f9b34fb}_Dev_VID&0216c0_PID&05df_REV&0100_30aea44c4efa#9&38bb9e5a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer:
  Product:      ESP BLE HID2
  Release:      100
  Interface:    0
  Usage (page): 0x100 (0xff00)

Device Found
  type: 16c0 05df
  path: \\?\HID#{00001812-0000-1000-8000-00805f9b34fb}_Dev_VID&0216c0_PID&05df_REV&0100_30aea44c4efa#9&f97cf65&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer:
  Product:      ESP BLE HID2
  Release:      100
  Interface:    0
  Usage (page): 0x1 (0xc)

```

PR #309 (2021-09-06): RFC: Bluetooth LE device info on Windows
This PR fixes issue with empty `manufacturer_string`/`product_string`/`serial_number` in `hid_device_info` struct with Bluetooth LE devices on Windows systems.

Fixes: #282
commit 11cc0e8a (2021-09-06): Fetch Bluetooth LE HID device info from Windows devnode properties (#309)

As per MS, HidD_GetProductString/HidD_GetManufacturerString/HidD_GetSerialNumberString [doesn't work](https://docs.microsoft.com/en-us/answers/questions/401236/hidd-getproductstring-with-ble-hid-device.html) with Bluetooth LE (BLE) devices.

Add special handling for BLE devices and fill `manufacturer_string`/`product_string`/`serial_number` using BLE-specific API.

Fixes: #282
closed 2021-09-06 by Youw