libusb/hidapi #283: Is possible an software damage an controller using HIDAPI ?

Augusto7743, 2021-06-12:
Hello.
Using Ubuntu 21.04 64 bits and PS3 DS3 controller in wired.
The problem is that was used an controller in Ubuntu using HIDAPI being that before had worked fine without problems in windows and now the controller has strange issues.
If the controller is connected for more than 20 minutes automatically is disabled and power off and not is possible enable the controller in wireless mode pressing PS button.
I need to wait some hours to work again.
Another problem is analog right is totally to up even not moving the controller.
Both problem had happen only after using the controller using Linux with HIDAPI.
I had tested with another controller and was the same problem with analog right and not trying to use thus avoiding problems.

Is possible and software having wrong code send command to PS3 controller using HIDAPI and change data in controller EEPROM ?
Please only waiting an information to understand where is the problem an not demading fix.

Have an nice week.

mcuee, 2021-06-12:
Yes technically that is possible, if there are EEPROMs inside the controller and allowed to be written by user applications. Then a mal-formed command may make the EEPROMs corrupted.

Augusto7743, 2021-06-12:
Thanks very much for that information.
Simply 2 controllers having strange issues how if having anything done inside creating problems.
Is possible fix an corrupted eeprom ? you known any information where is possible rewrite the eeproms with original code ?
Since that the controller was used in Linux using hidapi or hidraw problems at point any damage was done.

Thanks for your reply.

Youw, 2021-06-13:
Even if it _is_ possible to fix broken EEPROM, only a device's vendor can help you with that.

mcuee, 2021-06-13:
I do not know even the PS3 DualStrike 3 is using EEPROM or not. I was just replying your questions. You should ask in the PS3 Linux community to seek help. I will close the issue here.

> Thanks very much for that information.
> Simply 2 controllers having strange issues how if having anything done inside creating problems.
> Is possible fix an corrupted eeprom ? you known any [...]

closed 2021-06-13 by mcuee