libusb/hidapi #284: RFC: remove FOX based testgui

mcuee, 2021-06-15:
I think the Fox based testgui is not worth maintaining.

mcuee, 2021-06-15:
We can point users to hidapitester and hidpytoy.

hidapitester command line tool
https://github.com/todbot/hidapitester

hidpytoy: GUI
https://github.com/todbot/hidpytoy

Youw, 2021-06-15:
> I think the Fox based testgui is not worth maintaining.

https://github.com/libusb/hidapi/pull/220/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R62

mcuee, 2021-06-16:
@Youw Great. I am not a fan of auto-tools and I think CMake will be a bit better for HIDAPI (but I know libusb project rejected CMake ideas since auto-tools works fine)

Youw, 2021-06-16:
> (but I know libusb project rejected CMake ideas since auto-tools works fine)

Yeah, "If it is not broke, don’t fix it" - has its pros and cons

mcuee, 2021-07-13:
Close this since the latest README already mentioned that Test GUI is not actively being developed.

closed 2021-07-13 by mcuee

mcuee, 2026-04-25:
Interestingly TestGUI can still be built under the latest Ubuntu 26.04 once I install the dependency.
```
sudo apt install libfox-1.6-dev
```

Full build log.
```
mcuee  …/hidapi   master     v15.2.0  16:29  ./bootstrap 
+ autoreconf --install --verbose --force
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: aclocal --force -I m4
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:20: installing './compile'
configure.ac:18: installing './missing'
hidtest/Makefile.am: installing './depcomp'
autoreconf: Leaving directory '.'

mcuee  …/hidapi   master ?  16:30  mkdir build_autotools_gui

mcuee  …/hidapi   master ?  16:30  cd build_autotools_gui/
./  ../

mcuee  …/hidapi/build_autotools_gui   master ?  16:32  ../configure --enable-testgui   
This build script for HIDAPI is deprecated.
Consider using CMake instead.
checking for a BSD-compatible install... /usr/bin/install -c
checking whether sleep supports fractional seconds... yes
checking filesystem timestamp resolution... 0.01
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking xargs -n works... yes
checking whether UID '1000' is supported by ustar format... yes
checking whether GID '1000' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking build system type... x86_64-pc-linux-gnu
[... 212 lines cut ...]
make[2]: Entering directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui/hidtest'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui/hidtest'
make[1]: Leaving directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui/hidtest'
Making install in testgui
make[1]: Entering directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui/testgui'
make[2]: Entering directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui/testgui'
 /usr/bin/mkdir -p '/usr/local/bin'
  /bin/bash ../libtool   --mode=install /usr/bin/install -c hidapi-hidraw-testgui hidapi-libusb-testgui '/usr/local/bin'
libtool: install: /usr/bin/install -c .libs/hidapi-hidraw-testgui /usr/local/bin/hidapi-hidraw-testgui
libtool: install: /usr/bin/install -c .libs/hidapi-libusb-testgui /usr/local/bin/hidapi-libusb-testgui
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui/testgui'
make[1]: Leaving directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui/testgui'
make[1]: Entering directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui'
make[2]: Entering directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui'
make[2]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/usr/local/share/doc/hidapi'
 /usr/bin/install -c -m 644 ../README.md ../AUTHORS.txt ../LICENSE-bsd.txt ../LICENSE-gpl3.txt ../LICENSE-orig.txt ../LICENSE.txt '/usr/local/share/doc/hidapi'
 /usr/bin/mkdir -p '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 pc/hidapi-hidraw.pc pc/hidapi-libusb.pc '/usr/local/lib/pkgconfig'
make[2]: Leaving directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui'
make[1]: Leaving directory '/home/mcuee/build/libusb/hidapi/build_autotools_gui'
```