libusb/hidapi #285: RFC: change hidtest to an example and potentially include hidapitester

mcuee, 2021-06-15:
hidtest is using a test device based on modified Microchip Simple HID demo. It can be kept as an example but not that useful as a test tool for the project.

hidapitester is an excellent tool for testing of HID devices and HIDAPI. It will be good to include it in HIDAPI.
https://github.com/todbot/hidapitester

mcuee, 2021-06-15:
Only issue to include hidapitester is probably the GPL 3 license, which may or may not be a problem. I do not see it as a problem. But if some people see it as a problem, we can ask @todbot to see if he is willing to change the license.

mcuee, 2021-06-16:
@todbot How do you like the idea of including hidapitester in HIDAPI project?

todbot, 2021-06-16:
Sounds great to me

todbot, 2021-06-16:
I can change the license to whatever is most useful

mcuee, 2021-06-16:
HIDAPI license:
https://github.com/libusb/hidapi/blob/master/LICENSE.txt

hidapitester is already licensed under GPL3, just need to add the other two licenses.

mcuee, 2021-06-17:
Just FYI on the original hidtest, it is based on Microchip Demo FW (part of Microchip USB Stack) but I do not know if Alan Ott published the modified FW or not.

With the original FW (04d8:003f), here is the output.
```
Device Found
  type: 04d8 003f
  path: \\?\hid#vid_04d8&pid_003f#e&24ba9f4d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: Microchip Technology Inc.
  Product:      Simple HID Device Demo
  Release:      2
  Interface:    -1
  Usage (page): 0x1 (0xff00)

Manufacturer String: Microchip Technology Inc.
Product String: Simple HID Device Demo
Unable to read serial number string
Serial Number String: (0)
Indexed String 1: Microchip Technology Inc.
Unable to send a feature report.
Unable to get a feature report.
Incorrect function.waiting...
Data read:
   81 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Press any key to continue . . .
```

mcuee, 2021-07-24:
With the enhancement for hidtest, it is of course worth to keep it now. But still I think hidapitester is a worthy add-on for the hidapi project as it is useful for different type of HID devices.

mcuee, 2021-08-28:
The thing is that I am using the same version of hidapi here, the latest git. Test GUI is also from the latest git.

Supposedly the off-by-one has been fixed.
https://github.com/libusb/hidapi/pull/232
https://github.com/libusb/hidapi/commit/6fcb0bb2282dbb505561a53689eba0c0536570d9

mcuee, 2021-08-28:
It could be an hidapi issue under Windows in the end. Here is the test result from the original generic HID example here:
http://janaxelson.com/hidpage.htm

The C# based GUI application works fine. And the libusb1 hid backend works fine with the off-by-one fix https://github.com/libusb/libusb/pull/986

Host libusb1 codes:
https://github.com/mcuee/picusb/blob/master/lvr_genhid/libusb1_lvrhid4.c
```
C:\work\picusb\picusb\lvr_genhid [master ≡ +17 ~2 -0 !]> .\libusb_lvrhid4.exe
Successfully find the LVR Generic HID device
Successfully set usb configuration 1
Successfully claimed interface
Testing control transfer using loop back test of feature report
20, 20; 21, 21;
Testing interrupt transfer using loop back test of input/output report
40, 40; 41, 41;
```

mcuee, 2021-08-28:
C# GUI is working fine.
[image]

mcuee, 2021-08-28:
hidapitester output: total number of read bytes seems to be correct (2 bytes). But the data read seems to be not consistent between input report and feature report. Maybe this is expected behavior.

```
C:\work\hid\hidapitester2 [master ≡]> .\hidapitester.exe --vidpid 0925:7001 --open --send-feature 0,0x08,0x88 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 08 88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 0...read 2 bytes:
 00 08 88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
C:\work\hid\hidapitester2 [master ≡]> .\hidapitester.exe --vidpid 0925:7001 --open --send-output 0,0x08,0x88 --read-input
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 64-bytes...wrote 3 bytes:
 00 08 88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte input report 0, 250 msec timeout...read 2 bytes:
 08 88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
```

mcuee, 2021-08-28:
The Test GUI indeed has an issue to cut the feature in report. But the question is that should we report reading 3 bytes or 2 bytes? I think we probably need to report reading 3 bytes, including the report id 0. In that case the GUI will work fine for the reading of feature report.

[image]

mcuee, 2021-08-28:
@todbot In the end I think hidapitester is good. 
The issue is probably for feature report under Windows. Shall we report one more byte so that data will not be truncated?

mcuee, 2021-08-28:
Now I start to question the following commit which I proposed in #232. It may be correct when the report id is not zero but may not be correct when the report id is zero.
https://github.com/libusb/hidapi/commit/6fcb0bb2282dbb505561a53689eba0c0536570d9

mcuee, 2021-08-28:
@Youw : please take a look as well for the feature report. Thanks.

Ref: 
https://github.com/libusb/hidapi/issues/229#issuecomment-810780640
https://github.com/liquidctl/liquidctl/issues/312

mcuee, 2021-08-28:
My recommendation is to discard the report id 0 for feature report under Windows, to be consistent with input report handing under Windows, and also other OS.

mcuee, 2021-08-28:
Output under Linux:
```
hidapitester on  master [?] ❯ sudo ./hidapitester_hidraw --vidpid=0925:7001 --open --send-feature 0,0x40,0x41 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 0...read 3 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
hidapitester on  master [?] ❯ sudo ./hidapitester_hidraw --vidpid=0925:7001 --open --send-output 0,0x80,0x81 --read-input    
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 64-bytes...wrote 64 bytes:
 00 80 81 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte input report 0, 250 msec timeout...read 2 bytes:
 80 81 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device

```

mcuee, 2021-08-28:
Linux test gui is consistent with hidapitester
[image]

mcuee, 2021-08-28:
Hopefully other Windows users can chime in and test latest git of hidapi to see if there is a real issue for the feature report or not.

mcuee, 2021-08-28:
Just tried hidapitester with hidapi-0.10.1 release and hidapi git head, with regard to input report, the result is the same since it is using hid_read_timeout(). So #232 does not matter in that case.
https://github.com/todbot/hidapitester/blob/master/hidapitester.c#L379

With regard to feature report, then the result will be different as it uses hid_get_feature_report().
https://github.com/todbot/hidapitester/blob/master/hidapitester.c#L403

$ ./hidapitester_release0101.exe --vidpid=0925:7001 --open --send-feature 0,0x40,0x41 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 0...read 3 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device

$ ./hidapitester_git4d63a50.exe --vidpid=0925:7001 --open --send-feature 0,0x40,0x41 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 0...read 2 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device

mcuee, 2021-08-28:
For those with report id, commit https://github.com/libusb/hidapi/commit/6fcb0bb2282dbb505561a53689eba0c0536570d9 seems to be correct.

```
C:\work\hid\hidapitester [master ≡ +1 ~1 -0 !]> .\hidapitester_release0101.exe --vidpid 0925:7001 --open --send-feature 3,0x08,0x88 --read-feature 4
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 03 08 88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 4...read 4 bytes:
 04 08 88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
C:\work\hid\hidapitester [master ≡ +1 ~1 -0 !]> .\hidapitester_git4d63a50.exe --vidpid 0925:7001 --open --send-feature 3,0x08,0x88 --read-feature 4
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 03 08 88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 4...read 3 bytes:
 04 08 88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
```

mcuee, 2021-08-30:
Here is a quick hack of hidapitester with libusb backend (https://github.com/todbot/hidapitester/issues/8)

With patched libusb-1.0.dll (https://github.com/libusb/libusb/pull/986), it works fine.

```
C:\work\hid\hidapitester [master ≡ +3 ~0 -0 !]> .\hidapitester_libusb.exe --vidpid=0925:7001 --open --send-feature 0,0x40,0x41 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 0...read 3 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
C:\work\hid\hidapitester [master ≡ +3 ~0 -0 !]> .\hidapitester_libusb.exe --vidpid=0925:7001 --open --send-output 0,0x80,0x81 --read-input
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 64-bytes...wrote 4 bytes:
 00 80 81 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte input report 0, 250 msec timeout...read 3 bytes:
 80 81 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
```

[hidapitester_mingw64_libusb.zip](https://github.com/libusb/hidapi/files/7077458/hidapitester_mingw64_libusb.zip)

mcuee, 2021-08-30:
All in all, hidapitester is okay. But we may need to fix hidapi under Windows, for reading of feature report. Therefore I have created an issue #328.

mcuee, 2022-04-25:
I will close this as hidapitester is already mentioned in hidapi README.md.
https://github.com/libusb/hidapi/blob/master/README.md

closed 2022-04-25 by mcuee