libusb/hidapi #290: Need binary data transfer

drmcnelson, 2021-06-24:
The recv call returns a list of ints, one int for each byte in the data received from the device.

Consider the case where the data is a structure of strings floats and ints, or pixel data from an imaging sensor that you intend to pass on to other hardware.  

Your hardware sent binary data, and the python hidapi gives it to your as list ints, an int per byte, which you now have to convert back to a bytearray, at the cost of copying and repacking the data.

While the list of ints might be okay for some use cases,  it is not faithful to the data and by doing that, ignores and even vicerates the more general set of use cases.

Since,  the authors are unwilling to correct this api because (on the thesis that it is already deployed), the next best solution is provide another call that does return the data faithful to what was sent.

Please think about this seriously.   There is no point to defending a broken api.   Users will simply replace it.

mcuee, 2021-06-25:
Just wondering which API are you referring to. Are you talking about cython-hidapi or hidapi itself? I will say the C HIDAPI APIs talk about unsigned char *data.

https://github.com/libusb/hidapi/blob/master/hidapi/hidapi.h
```
int  HID_API_EXPORT HID_API_CALL hid_write(hid_device *dev, const unsigned char *data, size_t length);
int HID_API_EXPORT HID_API_CALL hid_read_timeout(hid_device *dev, unsigned char *data, size_t length, int milliseconds);
int  HID_API_EXPORT HID_API_CALL hid_read(hid_device *dev, unsigned char *data, size_t length);
int HID_API_EXPORT HID_API_CALL hid_send_feature_report(hid_device *dev, const unsigned char *data, size_t length);
int HID_API_EXPORT HID_API_CALL hid_get_feature_report(hid_device *dev, unsigned char *data, size_t length);
int HID_API_EXPORT HID_API_CALL hid_get_input_report(hid_device *dev, unsigned char *data, size_t length);

```

mcuee, 2021-06-25:
cython-hidapi is a different project.
https://github.com/trezor/cython-hidapi

drmcnelson, 2021-06-25:
I was referring to the python hidapi

mcuee, 2021-06-25:
In that case, I will close the issue here and please raise the issue to cython hidapi.

closed 2021-06-25 by mcuee

drmcnelson, 2021-06-25:
Thank you, sorry for the mispost