libusb/hidapi #292: How does the versioning scheme apply to ABI compatibility?

hacker1024, 2021-06-26:
As far as I can tell, the versioning scheme is not documented anywhere.
Assuming that semantic versioning is in use, I'd expect the following:

Pre `1.0.0`: `0.y.a` is ABI compatible with `0.y.b`
Post `1.0.0`: `x.a.b` is ABI compatible with `x.c.d`

Youw, 2021-06-27:
HIDAPI is being under version 0.x.y since forever, and by far I don't think we're close to 1.x.y.
Practically speaking, each new HIDAPI release always been API/ABI backward compatible with previouse releases.
There has been only one patch release (I guess current HIDAPI versioning can matches the [semantic versioning](https://semver.org/)) so far, and next release is going to be a minor version upgrade as well.

hacker1024, 2021-06-28:
Thanks for the info.
Should this be documented somewhere (perhaps in the wiki)?

Youw, 2021-07-03:
Added it to our [Wiki](https://github.com/libusb/hidapi/wiki)

closed 2021-07-03 by Youw