libusb/hidapi #298: hid_enumerate with hidraw backend returns lots of duplicate hid_device_infos

Be-ing, 2021-07-04:
Calling `hid_enumerate` with the hidraw Linux backend returns lots of duplicate `hid_device_info`s for the same device. I have to filter out duplicates by comparing the paths of the devices: https://github.com/mixxxdj/mixxx/pull/4054/files/6dd77b1fa61a2101e2dfac7cb7af85bf5eab168b#r663541357

Youw, 2021-07-04:
Do those devices have the same usage/usage_page pairs?

Be-ing, 2021-07-05:
Some do, some do not. Here is the output from Mixxx:
```
info [Controller] Scanning USB HID devices
info [Controller] Found HID device: { 17cc:1720 r60 | Usage: ff01:0000 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0001 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0001 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0030 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0001 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0030 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0030 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0030 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0080 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0080 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0080 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:0080 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:00d8 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:00d8 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:00f0 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:00f0 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:00e0 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Duplicate HID device, excluding { 17cc:1720 r60 | Usage: ff01:00e0 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 | S/N: 69DDA7A8 }
info [Controller] Found HID device: { 056a:5110 r0 | Usage: 000d:0004 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 000d:0022 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 000d:0022 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 000d:0022 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 000d:0022 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 000d:0022 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: ff11:0011 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 000d:000e | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 000d:0023 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 000d:0002 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 000d:0020 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: ff11:0002 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: ff11:0020 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 0001:0002 | Product: WCOM5110:00 056A:5110 }
info [Controller] Duplicate HID device, excluding { 056a:5110 r0 | Usage: 0001:0001 | Product: WCOM5110:00 056A:5110 }
info [Controller] Found HID device: { 06cb:7f8b r0 | Usage: 0001:0002 | Product: SYNA2B31:00 06CB:7F8B }
info [Controller] Duplicate HID device, excluding { 06cb:7f8b r0 | Usage: 0001:0001 | Product: SYNA2B31:00 06CB:7F8B }
[... cut ...]
info [Controller] Duplicate HID device, excluding { 06cb:7f8b r0 | Usage: ff00:0001 | Product: SYNA2B31:00 06CB:7F8B }
info [Controller] Found HID device: { 046d:b020 r0 | Usage: 0001:0002 | Product: MX Vertical | S/N: ed:39:60:f6:c3:e4 }
info [Controller] Duplicate HID device, excluding { 046d:b020 r0 | Usage: 0001:0001 | Product: MX Vertical | S/N: ed:39:60:f6:c3:e4 }
info [Controller] Duplicate HID device, excluding { 046d:b020 r0 | Usage: ff43:0202 | Product: MX Vertical | S/N: ed:39:60:f6:c3:e4 }
```

Be-ing, 2021-07-05:
I have not tested these devices on Windows.

Youw, 2021-07-05:
> some do not

This is unexpected.

> Some do

That is expected. That is the only way to check supported usage/usage_page-s of the device - report same device multiple times with the same path but different usage/usage_page pair

> I have not tested these devices on Windows.

I see.

mcuee, 2021-07-13:
Interesting, I have not seen such issue under Windows 10 Dell PC and macOS (Mac Mini M1). But I have not done much testing on Linux side (my Raspberry Pi 400 is my Linux computer now).

Be-ing, 2021-08-01:
This seems to occur on Windows too:
https://mixxx.discourse.group/t/cant-find-midi-output-controller-to-link-to-qlc/22751/4

mcuee, 2021-08-02:
@Be-ing It will be good to post the output of hidtest and the report descriptors (eg: under Linux or macOS). That may give some clues about the devices. Windows will create multiple devices with each for TLC (top level collections) and may even have some virtual device created on top.

Windows: https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/top-level-collections

mcuee, 2021-08-02:
@Be-ing It seems that it is not really an issue under Windows to have "multiple individual enabled devices for my singular mouse and keyboard". That is kind of expected under windows with devices with multiple TLCs, same for hidraw and macOS.

Maybe Mixxx code has wrong assumptions in this asepct.

Relevant discussion: https://github.com/libusb/hidapi/issues/287

Be-ing, 2021-08-02:
> Maybe Mixxx code has wrong assumptions in this asepct.

Perhaps, but besides potential functional implications, there is an issue of user confusion showing the same device multiple times.

mcuee, 2021-08-02:
> > Maybe Mixxx code has wrong assumptions in this asepct.
> 
> Perhaps, but besides potential functional implications, there is an issue of user confusion showing the same device multiple times.

That you can not change. hidapi is designed this way. You have to change mixxx.

Be-ing, 2021-08-02:
I already did: https://github.com/mixxxdj/mixxx/pull/4054/commits/5c582575a06f28e9c42b1261f5b475024c0f186b

But I wasn't sure if the problem was in Mixxx's use of hidapi or in hidapi.

mcuee, 2021-08-02:
In order to debug, please post the output of hidapi and the report descriptors.

mcuee, 2021-08-02:
@Be-ing As @Youw mentioned, if the device do not have the "same usage/usage_page pairs", then the behavior is not expected with Linux hidraw backend. Then that could be a hidapi bug. Please post the output of hidtest and the report descriptor to debug the issue.

mcuee, 2021-08-02:
Similar for Windows, if the device is listed in Windows Device Manager as different device due to fact that is has multiple TLCs, then it will be listed by HIDAPI as multiple devices and that is expected. If that is not the case, then it could be a HIDAPI bug. Again please post the output of hidtest under Windows and the HID report descriptors of the device (using Linux or macOS as it is difficult to do under Windows)

Be-ing, 2021-08-02:
I did not notice the issue on Windows; a user on the Mixxx forum did. I think it's too much to ask that user to build hidtest to investigate this. It would help if there were builds of that available.

mcuee, 2021-08-02:
I just created a snapshot here with VS2019 and MinGW-w64 (MSYS2 distro), both 32bit and 64bit binaries of hidtest and libhidapi.dll are included.
https://github.com/mcuee/hidapi/releases/tag/V27July2021

Be-ing, 2021-08-02:
It would be nice to automate that with GitHub Actions.

mcuee, 2021-08-02:
@Be-ing Good idea. 
@Youw please check if this can be done for different OS. Thanks.

mcuee, 2021-08-02:
On the other hand, under Windows, it is not easy to dump the hid report descriptors, so the user may need to use Linux or macOS. But anyway, another simple method is to check Windows Device Manager to see if the device is listed in multiple entries, then compare with the hidtest output.

Youw, 2021-08-02:
> @Youw please check if this can be done for different OS. Thanks.

#315

Youw, 2021-08-02:
> I already did: mixxxdj/mixxx@5c58257

If you want that to work any platform (Windows) you should not assume device uniqueness by the device path, but rather by the device VID+PID+USAGE+USAGE_PAGE.

mcuee, 2021-09-04:
@Be-ing @uklotzde
https://github.com/mixxxdj/mixxx/pull/4054#issuecomment-912897365

@uklotzde Just wondering if you can comment on this hidapi issue with regard to the Duplicate HID device under Linux hidraw backend. Please post the output of "lsusb -vvv" and "hidtest" (from hidapi) under Linux about your device (Traktor Kontrol S4 MK3). Thanks.

Be-ing, 2021-09-04:
lsusb output for Kontrol S4 Mk3: https://github.com/systemd/systemd/issues/20533#issuecomment-907358491

mcuee, 2021-09-04:
@Be-ing Unfortunately the lsusb output is not good enough as the HID report descriptor is not listed.

```
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface             11 Traktor Kontrol S4 MK3 HID
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     673
         Report Descriptors: 
           ** UNAVAILABLE **
```

Be-ing, 2021-09-04:
HID report descriptors for Kontrol S4 Mk3:
```
❯ usbhid-dump -d17cc:1720 | grep -v : | xxd -r -p | hidrd-convert -o spec
Usage Page (FF01h),                     ; FF01h, vendor-defined
Usage (00h),
Collection (Application),
    Usage (01h),
    Collection (Logical),
        Report ID (1),
        Usage (02h),
        Logical Minimum (0),
        Logical Maximum (1),
        Report Size (1),
        Report Count (136),
        Input (Variable),
        Usage (07h),
        Logical Minimum (0),
        Logical Maximum (1),
        Report Size (1),
        Report Count (16),
        Input (Variable),
        Usage (03h),
        Logical Minimum (0),
        Logical Maximum (15),
        Report Size (4),
        Report Count (6),
        Input (Variable),
    End Collection,
    Usage (01h),
    Collection (Logical),
        Report ID (2),
        Usage (05h),
        Logical Minimum (0),
        Logical Maximum (4095),
        Report Size (16),
        Report Count (7),
        Input (Variable),
        Usage (04h),
        Logical Minimum (0),
        Logical Maximum (4095),
        Report Size (16),
        Report Count (32),
        Input (Variable),
    End Collection,
    Usage (30h),
    Collection (Logical),
        Report ID (3),
        Usage (31h),
        Logical Minimum (0),
        Logical Maximum (255),
        Report Size (8),
        Report Count (3),
        Input (Variable),
        Usage (31h),
        Logical Minimum (0),
        Logical Maximum (255),
        Report Size (8),
        Report Count (28),
        Input (Variable),
        Usage (31h),
        Logical Minimum (0),
        Logical Maximum (255),
        Report Size (8),
        Report Count (28),
        Input (Variable),
    End Collection,
    Usage (01h),
    Collection (Logical),
        Report ID (4),
        Usage (31h),
[... 233 lines cut ...]
        Logical Maximum (255),
        Report Size (8),
        Report Count (1),
        Feature (Constant, Variable),
        Usage (E7h),
        Logical Minimum (0),
        Logical Maximum (100),
        Report Size (8),
        Report Count (1),
        Feature (Variable),
        Usage (E6h),
        Logical Minimum (0),
        Logical Maximum (100),
        Report Size (8),
        Report Count (1),
        Feature (Variable),
        Usage (E8h),
        Logical Minimum (0),
        Logical Maximum (1),
        Report Size (8),
        Report Count (1),
        Feature (Variable),
    End Collection,
End Collection
```

mcuee, 2021-09-04:
@Be-ing Thanks. Now we need the output of "hidtest-hidraw" of this device to see what is the problem.

Be-ing, 2021-09-04:
I have been working on reverse engineering the HID traffic between the Kontrol S4 Mk3 and Traktor this week in a Windows VM so I have that set up. I installed the [latest Mixxx main branch](https://github.com/mixxxdj/mixxx/commit/31b5e6c9641fee5072a4579b5f18a3c09e88a6fa) build and forwarded the Kontrol S4 Mk3 to the VM. It appears this issue does not affect Windows. This is what I see in the log regarding HID devices:

```
Info [Controller] Found HID device: { 17cc:1720 r60 | Usage: ff01:0000 | Interface: #3 | Manufacturer: Native Instruments | Product: Traktor Kontrol S4 MK3 HID | S/N: 69DDA7A8 }
Info [Controller] Excluding HID device { 0627:0001 r0 | Usage: 0001:0002 | Manufacturer: QEMU | Product: QEMU USB Tablet | S/N: 28754-0000:00:02.1:00.0-1 }
```

Be-ing, 2021-09-04:
```
❯ build/hidtest/hidtest_hidraw 
hidapi test/example tool. Compiled with hidapi version 0.11.0, runtime version 0.11.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x4 (0xd)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x11 (0xff11)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0xe (0xd)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x23 (0xd)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x2 (0xd)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x20 (0xd)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x2 (0xff11)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x20 (0xff11)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x2 (0x1)

Device Found
  type: 056a 5110
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: 
  Product:      WCOM5110:00 056A:5110
  Release:      0
  Interface:    -1
  Usage (page): 0x1 (0x1)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x2 (0x1)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x1 (0x1)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x5 (0xd)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0xe (0xd)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 06cb 7f8b
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      SYNA2B31:00 06CB:7F8B
  Release:      0
  Interface:    -1
  Usage (page): 0x1 (0xff00)

Device Found
  type: 046d b020
  path: /dev/hidraw2
  serial_number: ed:39:60:f6:c3:e6
  Manufacturer: 
  Product:      MX Vertical
  Release:      0
  Interface:    -1
  Usage (page): 0x2 (0x1)

Device Found
  type: 046d b020
  path: /dev/hidraw2
  serial_number: ed:39:60:f6:c3:e6
  Manufacturer: 
  Product:      MX Vertical
  Release:      0
  Interface:    -1
  Usage (page): 0x1 (0x1)

Device Found
  type: 046d b020
  path: /dev/hidraw2
  serial_number: ed:39:60:f6:c3:e6
  Manufacturer: 
  Product:      MX Vertical
  Release:      0
  Interface:    -1
  Usage (page): 0x202 (0xff43)

unable to open device
```

Be-ing, 2021-09-04:
```
hidapi on  fix_include_path [?] via △ v3.20.5 
❯ build/hidtest/hidtest_libusb 
hidapi test/example tool. Compiled with hidapi version 0.11.0, runtime version 0.11.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 17cc 1720
  path: 1-7.2.1:1.3
  serial_number: 69DDA7A8
  Manufacturer: Native Instruments
  Product:      Traktor Kontrol S4 MK3
  Release:      60
  Interface:    3
  Usage (page): 0x0 (0x0)

unable to open device
```

Be-ing, 2021-09-04:
Here is the hidraw output on Windows:
```
PS C:\Users\Be\Documents\hidapi_win_bin\hidapi_win_bin\VS2019\64bit> .\hidtest.exe
hidapi test/example tool. Compiled with hidapi version 0.11.0, runtime version 0.11.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 17cc 1720
  path: \\?\hid#vid_17cc&pid_1720&mi_03#8&15e53da9&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 69DDA7A8
  Manufacturer: Native Instruments
  Product:      Traktor Kontrol S4 MK3 HID
  Release:      60
  Interface:    3
  Usage (page): 0x0 (0xff01)

Device Found
  type: 0627 0001
  path: \\?\hid#vid_0627&pid_0001#7&143c9699&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 28754-0000:00:02.1:00.0-1
  Manufacturer: QEMU
  Product:      QEMU USB Tablet
  Release:      0
  Interface:    -1
  Usage (page): 0x2 (0x1)

unable to open device
```

mcuee, 2021-09-04:
@Be-ing Your hidtest-hidraw output is good. Indeed it shows the problems.

Eg: two duplicate entries here.
```
Device Found
  type: 17cc 1720
  path: /dev/hidraw3
  serial_number: 69DDA7A8
  Manufacturer: Native Instruments
  Product:      Traktor Kontrol S4 MK3
  Release:      60
  Interface:    3
  Usage (page): 0x1 (0xff01)

Device Found
  type: 17cc 1720
  path: /dev/hidraw3
  serial_number: 69DDA7A8
  Manufacturer: Native Instruments
  Product:      Traktor Kontrol S4 MK3
  Release:      60
  Interface:    3
  Usage (page): 0x1 (0xff01)
```

And 4 duplicate entries here:
```
Device Found
  type: 17cc 1720
  path: /dev/hidraw3
  serial_number: 69DDA7A8
  Manufacturer: Native Instruments
  Product:      Traktor Kontrol S4 MK3
  Release:      60
  Interface:    3
  Usage (page): 0x30 (0xff01)

Device Found
  type: 17cc 1720
  path: /dev/hidraw3
  serial_number: 69DDA7A8
  Manufacturer: Native Instruments
  Product:      Traktor Kontrol S4 MK3
  Release:      60
  Interface:    3
  Usage (page): 0x30 (0xff01)

Device Found
  type: 17cc 1720
  path: /dev/hidraw3
  serial_number: 69DDA7A8
  Manufacturer: Native Instruments
  Product:      Traktor Kontrol S4 MK3
  Release:      60
  Interface:    3
  Usage (page): 0x30 (0xff01)

Device Found
  type: 17cc 1720
  path: /dev/hidraw3
  serial_number: 69DDA7A8
  Manufacturer: Native Instruments
  Product:      Traktor Kontrol S4 MK3
  Release:      60
  Interface:    3
  Usage (page): 0x30 (0xff01)
```

mcuee, 2021-09-04:
@Be-ing 
As for the hidapi-libusb output under Linux, it seems to be a bit strange:
```Usage (page): 0x0 (0xff01)``` but that is not related to this issue.

As for your device under Windows, indeed I think it will not be affected as Windows treat the whole device (with single top level collection based on the report descriptor dump) as a single HID device.

Ref: Top-Level Collections under Windows: Microsoft defines a top-level collection as a HID collection that is not nested within another collection.
https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/top-level-collections

Be-ing, 2021-09-04:
This seems to be affecting macOS too:

https://github.com/mixxxdj/mixxx/pull/4243#issuecomment-911503659

mcuee, 2021-09-04:
> This seems to be affecting macOS too:
> 
> [mixxxdj/mixxx#4243 (comment)](https://github.com/mixxxdj/mixxx/pull/4243#issuecomment-911503659)

It will be good to ask the user to post the output of "hidtest" and "USB Prober". USB Prober can be used to dump the hid report descriptor under macOS.

mcuee, 2021-09-04:
USB Prober App download: https://ww5.touch-base.com/documentation/USBMacProber.htm
The 10.7.3 version still works on my Mac Mini M1 (Big Sur ARM64).
https://ww5.touch-base.com/downloads/MacProber/USB_Prober_Mac_OS_X_10.7.3.zip

mcuee, 2021-09-04:
Last version from IOUSBFamily 10.9.4
[USB Prober_10_9_4.app.zip](https://github.com/libusb/hidapi/files/7109465/USB.Prober_10_9_4.app.zip)
https://newbedev.com/how-can-i-install-usb-prober-from-the-developer-sdk-on-mac-os-x

mcuee, 2021-09-04:
https://newbedev.com/how-can-i-install-usb-prober-from-the-developer-sdk-on-mac-os-x
IORegistryExplorer.app is more update and you can download. 
Lastest release version is from Additional_Tools_for_Xcode_12.5. It can be used to dump the HID Report Descriptor but it does not have the capability to parse the report.

mcuee, 2021-09-13:
@Youw Now that the info under Linux is provided by Be-ing. Please take a look at the info and see how to further debug the issue under Linux.

Youw, 2023-03-18:
I apologise for not getting back to this for so long.
I've checked the logs and have a working theory of what's wrong going on here (spoiler - a HIDAPI issue).
To check my theory, would be nice to get the output of `hidtest` on _macOS_ as well for the same device (17cc:1720).
@Be-ing any chance you could provide that now?

Be-ing, 2023-03-18:
Sorry, I don't have access to any computers running macOS currently.

Youw, 2023-03-18:
What about Windows?
Will you be able to use the latest master's `hidtest` (and you'd have to modify it a bit) to print the output of [hid_get_report_descriptor](https://github.com/libusb/hidapi/blob/bd6be4d83b36d0dcd5401fb335384586b066a2eb/hidtest/test.c#L67) ? On Windows it would likely give a bit different output than you've already provied from a Linux machine (due to Windows Report Recostruction mechanism from PP Data).

And the HID Descriptor [output](https://github.com/libusb/hidapi/issues/298#issuecomment-912898563) you've provided earlier - if you could provide it in raw hex format I'd appreciate it, so I can experiment the fix on it.

Be-ing, 2023-03-18:
I have a Windows VM I could use to run `hidtest` if you make a build with the modifications you alluded to.

Youw, 2023-03-19:
I've just checked - no changes to hidtest required. I've already applied those some while ago.
Simply run the `hidtest` from latest master.

mcuee, 2023-05-01:
> I have a Windows VM I could use to run `hidtest` if you make a build with the modifications you alluded to.

@Be-ing 
Any update on this topic?

Youw, 2023-07-23:
@Be-ing see if you could test #601

commit 3ef6cf4d (2023-07-23): hidraw: report only Top-Level Usage_page/Usage pairs for Collections

To be consistent with Windows/macOS (OS-level implementations).

Fixes: #298
commit 8edfc528 (2023-07-23): hidraw: report only Top-Level Usage_page/Usage pairs for Collections

To be consistent with Windows/macOS (OS-level implementations).

Fixes: #298
commit 78c6adec (2023-07-23): hidraw: report only Top-Level Usage_page/Usage pairs for Collections

To be consistent with Windows/macOS (OS-level implementations).

Fixes: #298
PR #601 (2023-08-26): hidraw: report only Top-Level Usage_page/Usage pairs as 'unique' devices 
To be consistent with Windows/macOS (OS-level implementations).

Fixes: #298
Fixes: #454
closed 2023-08-26 by Youw