libusb/hidapi #300: Problem with writing data to different usage pages on mac?

anjisZhou, 2021-07-05:
HI,I have a **hid** device, one of its **hid** interface has 2 **usage pages**, under **windows**, the path of each **usage page** is different, so you can find that this hid device has 2 paths, I choose one path to open it Open and can write data in, but under the **mac**, the path of the two **usage pages** is the same, so I can’t choose one of them to open, so I always can’t write data in, because the **outputreportlength** of the first **usage page** is 0. Is there any solution?

Youw, 2021-07-05:
> under windows, the path of each usage page is different

It is handled by WinAPI - different Top Level Collections (usage/usage_page pairs) are split into different ligical devices.

> under the mac, the path of the two usage pages is the same

Strictly speaking, on macOS there is only one device. HIDAPI explicitly duplicates it (reports it multiple times), as right now it is the only way to report all supported usage/usage_page pairs.

> I can’t choose one of them to open

Technically you can open either one of them, with the same effect.

> so I always can’t write data in, because the outputreportlength of the first usage page is 0

Not sure what do you mean by this. The length of the output report for each report can be unique (regardless under wich usage/usage_page it is defined).

anjisZhou, 2021-07-06:
Sorry, the previous description is not so good. I found 4 usage/usage pages using hidapi. Under windows, I choose usage page=65424, and data can be written, but under mac, it always returns -1 .
[image]

todbot, 2021-07-06:
@anjisZhou is there a driver installed for this device on the Mac? That will supersede any access by hidapi.

anjisZhou, 2021-07-06:
@todbot No installation, I just tested plugging and unplugging the usb device, using hidapi to link the usb device, you can write once, but you can't always write later, I guess the first write failed. When I read the'in report', the data read is always the data with reportid 1. This is the hardware status, and hid actively reports it, but I need to read and write the data with reportid 0x55, and I don’t know how to do it.

todbot, 2021-07-06:
The first byte of the report buffer to send is the reportId.

anjisZhou, 2021-07-06:
I have added and written data like this, **byte data[65]={0x55,0x50};**

anjisZhou, 2021-07-08:
I have found the reason, it is a problem with our hardware, thanks

closed 2021-07-08 by anjisZhou