libusb/hidapi #307: How to use hidapi library which has a different hid report format

crazoes, 2021-07-19:
I would like to replicate the following Windows based code to set the device in Upgrade mode into Linux but I do not have the details of functions sendData and readData.
https://gist.github.com/crazoes/e33146f4a4ea55f8e4b67997f43a20e6

AFAIK, the first parameter in the report descriptor is the report ID, but as you can see in the Windows code, there is no report ID, instead they are having some header bytes.

Will it be possible to use hidapi library in my case and if yes then can you please explain how?

Youw, 2021-07-19:
Do you even know that your device is a HID device, and `SendData`/`ReadData` are making a HID requests?

mcuee, 2021-07-20:
@crazoes Please post the output of the "lsusb -vvv" result of your device first to confirm that it is an HID device (assume it is a USB device).

crazoes, 2021-07-20:
@Youw , @mcuee The touchscreen device is connected using I2C and it uses i2c-hid and hid-multitouch drivers.
The device by default starts in HID mode but I can also flip it to i2c mode through userspace using some i2ctransfer commands.

lsusb won't list this device since it is an HID I2C based device and is not connected using USB.
I am assuming here that SendData and ReadData are making HID request since I don't see device being flipped to i2c mode anywhere in the code.

mcuee, 2021-07-20:
What is the output of hidtest (from hidapi)?

crazoes, 2021-07-20:
@mcuee I get the following output when I run hidtest ( without the set and get feature report code )

`hidapi test/example tool. Compiled with hidapi version 0.11.0, runtime version 0.11.0.
Compile-time version matches runtime version of hidapi.

hidapi test/example tool. Compiled with hidapi version 0.11.0, runtime version 0.11.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 046d c316
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: Logitech
  Product:      Logitech USB Keyboard
  Release:      2800
  Interface:    0
  Usage (page): 0x6 (0x1)

Device Found
  type: 046d c316
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: Logitech
  Product:      Logitech USB Keyboard
  Release:      2800
  Interface:    1
  Usage (page): 0x1 (0xc)

Device Found
  type: 046d c316
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: Logitech
  Product:      Logitech USB Keyboard
  Release:      2800
  Interface:    1
  Usage (page): 0x80 (0x1)

Device Found
  type: 2808 1015
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      FTS3528:00 2808:1015
  Release:      0
  Interface:    -1
  Usage (page): 0x4 (0xd)

Device Found
  type: 2808 1015
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      FTS3528:00 2808:1015
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 2808 1015
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      FTS3528:00 2808:1015
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 2808 1015
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      FTS3528:00 2808:1015
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 2808 1015
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      FTS3528:00 2808:1015
  Release:      0
  Interface:    -1
  Usage (page): 0x22 (0xd)

Device Found
  type: 2808 1015
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      FTS3528:00 2808:1015
  Release:      0
  Interface:    -1
  Usage (page): 0x1 (0xff00)

Device Found
  type: 2808 1015
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      FTS3528:00 2808:1015
  Release:      0
  Interface:    -1
  Usage (page): 0x0 (0xd)

Device Found
  type: 2808 1015
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: 
  Product:      FTS3528:00 2808:1015
  Release:      0
  Interface:    -1
  Usage (page): 0x20 (0xd)

Manufacturer String: 
Product String: FTS3528:00 2808:1015
Serial Number String: (0) 
Unable to read indexed string 1
Indexed String 1: 
`

I also tried to run this hid-example.c without the set and get feature reports
https://github.com/torvalds/linux/blob/master/samples/hidraw/hid-example.c

I get the following output for it
`Report Descriptor Size: 543
Report Descriptor:
5 d 9 4 a1 1 85 1 9 22 a1 2 9 42 15 0 25 1 75 1 95 1 81 2 95 1 81 3 95 6 81 3 75 8 9 51 95 1 81 2 5 1 26 20 3 75 10 55 e 65 11 9 30 35 0 46 b a 95 2 81 2 35 1 35 0 46 aa 5 26 0 5 9 31 81 2 35 1 35 0 5 d 26 ff 0 46 ff 0 95 1 9 48 81 2 9 49 81 2 c0 9 22 a1 2 9 42 15 0 25 1 75 1 95 1 81 2 95 1 81 3 95 6 81 3 75 8 9 51 95 1 81 2 5 1 26 20 3 75 10 55 e 65 11 9 30 35 0 46 b a 95 2 81 2 35 1 35 0 46 aa 5 26 0 5 9 31 81 2 35 1 35 0 5 d 26 ff 0 46 ff 0 95 1 9 48 81 2 9 49 81 2 c0 9 22 a1 2 9 42 15 0 25 1 75 1 95 1 81 2 95 1 81 3 95 6 81 3 75 8 9 51 95 1 81 2 5 1 26 20 3 75 10 55 e 65 11 9 30 35 0 46 b a 95 2 81 2 35 1 35 0 46 aa 5 26 0 5 9 31 81 2 35 1 35 0 5 d 26 ff 0 46 ff 0 95 1 9 48 81 2 9 49 81 2 c0 9 22 a1 2 9 42 15 0 25 1 75 1 95 1 81 2 95 1 81 3 95 6 81 3 75 8 9 51 95 1 81 2 5 1 26 20 3 75 10 55 e 65 11 9 30 35 0 46 b a 95 2 81 2 35 1 35 0 46 aa 5 26 0 5 9 31 81 2 35 1 35 0 5 d 26 ff 0 46 ff 0 95 1 9 48 81 2 9 49 81 2 c0 5 d 27 ff ff 0 0 75 10 95 1 9 56 81 2 9 54 25 10 95 1 75 8 81 2 85 2 9 55 95 1 25 a b1 2 6 0 ff 9 c5 85 5 15 0 26 ff 0 75 8 96 0 1 b1 2 c0 6 0 ff 9 1 a1 1 15 0 26 ff 0 75 8 85 6 95 3f 9 1 81 2 9 1 91 2 c0 5 d 9 0 a1 1 85 7 9 20 a1 2 9 42 9 44 9 45 15 0 25 1 75 1 95 3 81 2 95 1 81 3 9 32 81 2 95 3 81 3 5 1 55 e 65 11 35 0 75 10 95 2 46 7c b 26 ff f 9 30 81 2 46 7c 6 26 ff f 9 31 81 2 5 d 9 30 15 1 26 ff 0 75 8 95 1 81 2 c0 c0 

Raw Name: FTS3528:00 2808:1015
Raw Phys: i2c-FTS3528:00
Raw Info:
	bustype: 24 (Other)
	vendor: 0x2808
	product: 0x1015
`

mcuee, 2021-07-20:
The following is the HID report descriptor parser output. Now you have to carry out your own research. I will close the issue for now.

https://eleccelerator.com/usbdescreqparser/

```
0x05, 0x0D,        // Usage Page (Digitizer)
0x09, 0x04,        // Usage (Touch Screen)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x09, 0x22,        //   Usage (Finger)
0xA1, 0x02,        //   Collection (Logical)
0x09, 0x42,        //     Usage (Tip Switch)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x75, 0x01,        //     Report Size (1)
0x95, 0x01,        //     Report Count (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //     Report Count (1)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x06,        //     Report Count (6)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x75, 0x08,        //     Report Size (8)
0x09, 0x51,        //     Usage (0x51)
0x95, 0x01,        //     Report Count (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x26, 0x20, 0x03,  //     Logical Maximum (800)
0x75, 0x10,        //     Report Size (16)
0x55, 0x0E,        //     Unit Exponent (-2)
0x65, 0x11,        //     Unit (System: SI Linear, Length: Centimeter)
0x09, 0x30,        //     Usage (X)
0x35, 0x00,        //     Physical Minimum (0)
0x46, 0x0B, 0x0A,  //     Physical Maximum (2571)
0x95, 0x02,        //     Report Count (2)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x35, 0x01,        //     Physical Minimum (1)
0x35, 0x00,        //     Physical Minimum (0)
0x46, 0xAA, 0x05,  //     Physical Maximum (1450)
0x26, 0x00, 0x05,  //     Logical Maximum (1280)
0x09, 0x31,        //     Usage (Y)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x35, 0x01,        //     Physical Minimum (1)
0x35, 0x00,        //     Physical Minimum (0)
0x05, 0x0D,        //     Usage Page (Digitizer)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x46, 0xFF, 0x00,  //     Physical Maximum (255)
0x95, 0x01,        //     Report Count (1)
0x09, 0x48,        //     Usage (0x48)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x49,        //     Usage (0x49)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x09, 0x22,        //   Usage (Finger)
0xA1, 0x02,        //   Collection (Logical)
0x09, 0x42,        //     Usage (Tip Switch)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x75, 0x01,        //     Report Size (1)
0x95, 0x01,        //     Report Count (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //     Report Count (1)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x06,        //     Report Count (6)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x75, 0x08,        //     Report Size (8)
0x09, 0x51,        //     Usage (0x51)
0x95, 0x01,        //     Report Count (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x26, 0x20, 0x03,  //     Logical Maximum (800)
[... 170 lines cut ...]
0x55, 0x0E,        //     Unit Exponent (-2)
0x65, 0x11,        //     Unit (System: SI Linear, Length: Centimeter)
0x35, 0x00,        //     Physical Minimum (0)
0x75, 0x10,        //     Report Size (16)
0x95, 0x02,        //     Report Count (2)
0x46, 0x7C, 0x0B,  //     Physical Maximum (2940)
0x26, 0xFF, 0x0F,  //     Logical Maximum (4095)
0x09, 0x30,        //     Usage (X)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x46, 0x7C, 0x06,  //     Physical Maximum (1660)
0x26, 0xFF, 0x0F,  //     Logical Maximum (4095)
0x09, 0x31,        //     Usage (Y)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x0D,        //     Usage Page (Digitizer)
0x09, 0x30,        //     Usage (Tip Pressure)
0x15, 0x01,        //     Logical Minimum (1)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection

// 543 bytes
```

closed 2021-07-20 by mcuee

mcuee, 2021-07-20:
Google shows this touchscreen controller works under Linux. So you may get better help from relevant Linux community for the communication protocol.
https://ubuntu-mate.community/t/ubuntu-mate-on-a-hybrid-lenovo-miix-320/22042
https://gist.github.com/mcxiaoke/9485c7f6978326553269decbcc6beca3