libusb/hidapi #317: 'undefined reference' when compiling test code

Ruffunaru, 2021-08-04:
Hello,

I want to use this library so i downloaded the package with :
`sudo apt install libhidapi-dev`

Then I tried to compile the test code from the front page with a simple command :
`gcc TestHID.c -o TestHID`

But I got this error
```
TestHID.c:6:10: fatal error: hidapi.h: No such file or directory
```
I fixed it by changing `#include "hidapi.h"` with `#include "hidapi/hidapi.h"`

Then, I find another output :
```
/usr/bin/ld: /tmp/cc7q90gQ.o: in function `main':
TestHID.c:(.text+0x2c): undefined reference to `hid_init'
/usr/bin/ld: TestHID.c:(.text+0x46): undefined reference to `hid_open'
/usr/bin/ld: TestHID.c:(.text+0x6b): undefined reference to `hid_get_manufacturer_string'
/usr/bin/ld: TestHID.c:(.text+0xaa): undefined reference to `hid_get_product_string'
/usr/bin/ld: TestHID.c:(.text+0xe9): undefined reference to `hid_get_serial_number_string'
/usr/bin/ld: TestHID.c:(.text+0x12f): undefined reference to `hid_get_indexed_string'
/usr/bin/ld: TestHID.c:(.text+0x173): undefined reference to `hid_write'
/usr/bin/ld: TestHID.c:(.text+0x19c): undefined reference to `hid_write'
/usr/bin/ld: TestHID.c:(.text+0x1bd): undefined reference to `hid_read'
/usr/bin/ld: TestHID.c:(.text+0x217): undefined reference to `hid_close'
/usr/bin/ld: TestHID.c:(.text+0x21c): undefined reference to `hid_exit'
collect2: error: ld returned 1 exit status
```

I tried to find a solution on google but I did'nt find anything.
Is there something I forget ? I feel I miss something.

Youw, 2021-08-04:
If that is the direction you want to go, then you need to **link** against one of the backends (there is [2](https://github.com/libusb/hidapi#about) on Linux) - `hidapi-hidraw` or `hidapi-libusb`, e.g.:
`gcc TestHID.c -o TestHID -lhidapi-libusb`.

Secondly, `hidapi` currently provides `pkg-config` package, and there is known ways to use `pkg-config` for dependencies: https://people.freedesktop.org/~dbn/pkg-config-guide.html#faq (Q1 gives a simple answer of "how".)

---
Please NOTE: those are a very basics of using 3rdparty libraries with C/C++. This is a development issue tracker, not a support forum. Stackoverflow is where you should normally ask questions like this.

closed 2021-08-04 by Youw

mcuee, 2021-08-06:
It is also okay to ask in libusb-devel mailing list for HIDAPI related questions.