libusb/hidapi #318: hid_open() doesn't open USB device, but once device is open hid_read() works fine

Pinkers, 2021-08-05:
Hi, I am working with an HID device (a CO2 monitor) and I am trying to use the hidapi library to connect to it. When I plug in the device and run the following function, hid_open(), it returns 'unable to connect to HID device'. Obviously, if I run hid_read, no data is returned. However, if I open the proprietary software for my co2meter which can be found here: //www.zyaura.com/support/support_software.htm the outcome is different. In this case, hid_open() has the same effect, but hid_read() returns data. Does anyone have ANY idea what is happening here? If so, some knowledge would be very appreciated.

In the init (for context):
`
self._h = hid.device()
self._handle = 1
'
the method:
'
def hid_open(self, send_magic_table=True):
        handle = self._h.open(_CO2MON_HID_VENDOR_ID, _CO2MON_HID_PRODUCT_ID)
        if not handle:
            print('unable to connect to HID device')
        else:
            self._handle = handle
            print('device connected')`

mcuee, 2021-08-06:
You have to reverse engineering the protocol. Apparently the vendor software put the device into some mode (eg: by sending an Out report or Feature report to the device) which then enables you to use hid_read() to read the Input report.

A USB analyzer may help here. Or if you use Linux, then you can set up a Windows VM and then use usbmon to capture the USB traffic.

mcuee, 2021-08-06:
I will close this for now. You can use libusb-devel mailing list or website like stackoveflow to ask technical questions. Thanks.

mcuee, 2021-08-06:
Please also check if the device changes the PID after the vendor software opens. By right you should be able to open the device if it is an HID device.

mcuee, 2021-08-06:
You can also post the output of hidtest about your device. hidtest is here. You can build it with Visual Studio or MinGW-w64(eg: MSYS2 distribution).
https://github.com/libusb/hidapi/tree/master/hidtest

closed 2021-08-06 by mcuee

mcuee, 2021-08-06:
I will reopen this issue if hidtest also can not open your device.