libusb/hidapi #320: Could not determine the host path corresponding to '/work/windows/.libs'

tresf, 2021-08-11:
This is a request for help, not a bug report.

When cross-compiling `hidapi` using `llvm-mingw` wrapper (a mingw-w64 wrapper around clang, capable of producing ARM64 binaries for Windows, it shows the following error:

Bad output from `llvm-mingw` build logs:
```diff
  make[2]: Entering directory '/work/hidtest'
    CCLD     hidtest.exe
- libtool:   error: Could not determine the host path corresponding to
- libtool:   error:   '/work/windows/.libs'
  libtool:   error: Continuing, but uninstalled executables may not work.
  libtool:   error: Could not determine the host path corresponding to
  libtool:   error:   '/work/windows/.libs:/usr/local/lib:/usr/local/bin'
  libtool:   error: Continuing, but uninstalled executables may not work.
```

I assume this is due to `/work/hidtest/hidtest.exe` not being knowledgeable of `/work/windows/.libs`, but if that's the case, why does this NOT fail when building using mingw-w64 directly?

Good output from `mingw-w64` build logs:
```
make[2]: Entering directory '/work/hidtest'
  CC       test.o
  CCLD     hidtest.exe
make[2]: Leaving directory '/work/hidtest'
make[2]: Entering directory '/work'
make[2]: Leaving directory '/work'
make[1]: Leaving directory '/work'
```

Naturally, the build produces usable binaries, only the `hidtest.exe` seems to fail, which is benign to this particular build system.

Questions:
1. Can I configure make to skip building `hidtest.exe` to eliminate the error?
2. What types of things can I look at to patch this?  Why would the build system fail for this environment only?

Note 1: As a reminder, `llvm-mingw` uses clang under the covers, so if there are any GNU-specific assumptions in the compiler, I'd be happy to dig deeper.  Any help is appreciated!

Note 2: I've searched before asking and found  #187, #68.  Neither seem to be the same exact problem, but I wanted to link them incase they are. 🍻

mcuee, 2021-08-12:
Try using the latest git head and CMake. You may get better results here.

mcuee, 2021-08-12:
BTW, #187 and #68 talk about the legacy Fox based Test GUI application, it has nothing to do with hidtest.

mcuee, 2021-08-12:
Another thing, I believe clang is not the issue here. After all, there is no issue to build hidapi under macOS which use clang.

Youw, 2021-08-12:
> Can I configure make to skip building hidtest.exe to eliminate the error?

autotools doesn't have such option, but you could remove [this](https://github.com/libusb/hidapi/blob/8741697359b8dd0d73702a69c54e64ffc552d3cf/configure.ac#L245) and [this](https://github.com/libusb/hidapi/blob/8741697359b8dd0d73702a69c54e64ffc552d3cf/Makefile.am#L42) lines to skip hidtest build

---
We're not looking to support nor improve autotools build system. If you come up with a clean working patch - we will accept it, but otherwise prefer using CMake instead.

tresf, 2021-08-12:
> remove this 

Thanks, that's helpful, I'll see what's going on with that additional makefile.

> If you come up with a clean working patch

That's what I'm hoping.

> but otherwise prefer using CMake instead.

I'm normally not afraid to using CMake, but this particular downstream project uses autotools currently for 10 separate targets, so I'll take a swing at a quick patch first,.

tresf, 2021-08-12:
> latest git head

For clarification, it is the latest HEAD from master branch.

tresf, 2021-08-12:
Using this bug report as a scratch pad, I hope that's ok?  So for starters, the linking in the `hidtest.exe` is as follows:

```make
LINK = $(LIBTOOL) $(AM_V_lt) --tag=CC $(AM_LIBTOOLFLAGS) \
	$(LIBTOOLFLAGS) --mode=link $(CCLD) $(AM_CFLAGS) $(CFLAGS) \
	$(AM_LDFLAGS) $(LDFLAGS) -o $@
```

The first `AM_V_lt` is verbosity.  Digging into this shows I can call `make V=1` to increase verbosity of the automake system, which successfully outputs a more verbose command:

```bash
/bin/bash ../libtool  --tag=CC   --mode=link /usr/xcc/aarch64-w64-mingw32-cross/bin/aarch64-w64-mingw32-gcc  
   -g -O2  -no-undefined -o hidtest.exe test.o ../windows/libhidapi.la -lsetupapi

libtool: link: /usr/xcc/aarch64-w64-mingw32-cross/bin/aarch64-w64-mingw32-gcc -g -O2 -o 
.libs/hidtest.exe test.o  ../windows/.libs/libhidapi.dll.a -lsetupapi -L/usr/local/lib
```

... which seems to be properly referencing the archive by relative path.

What's particularly interesting is that the `libtool` command is properly linking `../windows/.libs/`, however according to [`libtool`'s documentation](https://www.gnu.org/software/libtool/manual/html_node/File-Name-Conversion-Failure.html), may simply be a file name conversion failure?  Maybe not though... 

First thing that comes to mind is the `lib` suffix, but I don't see how this could cause issues, I'd expect this to fail on mingw too.  Next, I stumbled upon this similar post: https://github.com/VirusTotal/yara/issues/933#issuecomment-441192312, which mentions this could go away if the required library is installed.  Oddly, it doesn't.

```
libtool: install: aarch64-w64-mingw32-ranlib /usr/local/lib/libhidapi.a
 /bin/mkdir -p '/usr/local/include/hidapi'
 /usr/bin/install -c -m 644 ../hidapi/hidapi.h '/usr/local/include/hidapi'
```

Same error:

```
libtool: link: /usr/xcc/aarch64-w64-mingw32-cross/bin/aarch64-w64-mingw32-gcc -g -O2 -o .libs/hidtest.exe test.o  ../windows/.libs/libhidapi.dll.a -lsetupapi -L/usr/local/lib
libtool:   error: Could not determine the host path corresponding to
libtool:   error:   '/work/windows/.libs'
libtool:   error: Continuing, but uninstalled executables may not work.
libtool:   error: Could not determine the host path corresponding to
libtool:   error:   '/work/windows/.libs:/usr/local/lib:/usr/local/bin'
libtool:   error: Continuing, but uninstalled executables may not work.
```

Then I stumbled upon [this post](https://lists.gnu.org/archive/html/libtool/2008-10/msg00012.html) which offers some additional troubleshooting commands such as `objdump` and `(prefix)-nm -B -f POSIX -A libhidapi.a`, but this didn't really offer any more insight.  The files looked fine from what I can tall.

... but this looks like a linker issue, so I decided to take one final look, run `make V=1` and see if I could observe any obvious differences between `libtool` calls for mingw "proper" and mingw "wrapper".  The main difference I can find is this:

* The "wrapper" makes use of a dedicated `aarch64-w64-mingw32-ranlib`, whereas `mingw` uses the system-provided one.   This is likely to do with the `clang` dependency of the wrapper.

That's it.  I can't find any discernible differences between mingw proper and the clang wrapper tool.   This leads me to believe that the error is actually a misconfiguration with the mingw wrapper and autotools, specifically the function that tries to convert unix paths to w32 paths, `func_convert_file_nix_to_w32`, but the `libtool` code between builds is identical.

So, what else could be different?  Well, I stumbled up on the `BUILD.autotools.md` file which states the following config option is available: 

> `--disable-shared`
> &nbsp;&nbsp;&nbsp;&nbsp; By default, both shared and static libraries are going to be built/installed.
> &nbsp;&nbsp;&nbsp;&nbsp; This option disables shared library build, if only static library is required.

So I took a second look at `libtool` and spotted the following differences:

mingw "proper" versus "wrapper":

```diff
  # Whether or not to build shared libraries.
- build_libtool_libs=yes
+ build_libtool_libs=no
```

Is this just a matter of the wrapper I'm using defaulting to building shared libraries?  If so, is that something the environment does automagically or something that `hidapi` has defined somewhere in an automake file?

So long story short, this seems to fix it:
```bash
./configure --disable-shared 
```

tresf, 2021-08-12:
Sorry, had to edit out some stuff above.  Hid it for historical purposes.

Anyway... I think I found something that fixes this, but I'm not sure if it's welcome in the codebase:

`hidtest/Makefile.am`:
```diff
  hidtest_SOURCES = test.c
+ hidtest_LDFLAGS = -all-static
  hidtest_LDADD = $(top_builddir)/$(backend)/libhidapi.la
```

I still have a sneaking suspicion something is toggling this on for mingw, I just don't know where to look. 😅

tresf, 2021-08-12:
WIthout changing the `hidtest/Makefile.am`, `LDFLAGS="-static"` or `CFLAGS="-static"` seems to fix this as well.  I'm not really sure the impact of this, so I'll test the resulting binary on a Windows ARM64 computer to make sure it still works.

tresf, 2021-08-12:
The produced binary works just fine in my tests using `./configure LDFLAGS="-static"`, so I'm closing this out as a configuration issue (my problem)/possible upstream issue (not hidapi's problem, perhaps `llvm-mingw`)

Anyway, I hope the information above is useful to someone else running into a similar problem down the road.  If anyone has more information to add, please let me know. :)

closed 2021-08-12 by tresf

mcuee, 2021-08-13:
No idea why it works for you for cross build. But anyway, if that works for you, it is good.

With normal MSYS2 MinGW-w64 build, `./configure LDFLAGS="-static"` is basically the same as `./configure --disable-shared --enable-static".

```
$ ./configure LDFLAGS="-static"
configure: loading site script /etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking build system type... x86_64-w64-mingw32
checking host system type... x86_64-w64-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe
checking if the linker (C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /mingw64/bin/nm -B
checking the name lister (/mingw64/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-w64-mingw32 file names to x86_64-w64-mingw32 format... func_convert_file_msys_to_w32
checking how to convert x86_64-w64-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /mingw64/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for dlfcn.h... no
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -DDLL_EXPORT -DPIC
checking if gcc PIC flag -DDLL_EXPORT -DPIC works... yes
[... 67 lines cut ...]
make[2]: Entering directory '/c/work/hid/hidapi/hidtest'
  CC       test.o
  CCLD     hidtest.exe
make[2]: Leaving directory '/c/work/hid/hidapi/hidtest'
make[2]: Entering directory '/c/work/hid/hidapi'
make[2]: Leaving directory '/c/work/hid/hidapi'
make[1]: Leaving directory '/c/work/hid/hidapi'

$ make
make  all-recursive
make[1]: Entering directory '/c/work/hid/hidapi'
Making all in windows
make[2]: Entering directory '/c/work/hid/hidapi/windows'
  CC       hid.lo
  CCLD     libhidapi.la
make[2]: Leaving directory '/c/work/hid/hidapi/windows'
Making all in hidtest
make[2]: Entering directory '/c/work/hid/hidapi/hidtest'
  CC       test.o
  CCLD     hidtest.exe
make[2]: Leaving directory '/c/work/hid/hidapi/hidtest'
make[2]: Entering directory '/c/work/hid/hidapi'
make[2]: Leaving directory '/c/work/hid/hidapi'
make[1]: Leaving directory '/c/work/hid/hidapi'
```