libusb/hidapi #321: macOS hid.c created a separate thread for runloop

mcuee, 2021-08-12:
Ref: https://github.com/cvuchener/hidpp/issues/17#issuecomment-895392608

cvuchener (author of hidpp) seems to say this may not be a necessary and he thinks there is a potentially better way to use mach ports.

I do not understand this well. Just wondering why the macOS codes create a seperate thread for runloop.

Youw, 2021-08-12:
HIDAPI needs a runloop to function. A runloop blocks the thread execution where it is invoked, and only processes "asyncronous" messages sent to objects that "live" in that runloop. Each macOS HID object has to live in a runloop as it is the only way to get HID interruptions.

In theory - yes, HIDAPI could leave without a separate thread, but only if a calling code guarantees that in each thread where `hid_init/hid_enumerate/hid_open` (maybe not all of those, I'm not sure, but it doesn't really matter) is called - there is a runloop created and that it will be active for the whole hid_device lifetime.

This makes a limitation on the usage of the HIDAPI, and makes it really hard to use HIDAPI in simple UNIX-like console application, which knows nothing of the environment they run in (macOS)/runloops/etc.

Youw, 2021-08-12:
What is done right now - a thread is created per-device.
In theory this can be optimised to create a thread per-context (we have only one context as per current implementation) for all devices.
But I think that is as much as hidapi could go to preserve the API and simple usage.

noah-nuebling, 2021-08-12:
I’m currently working on the macOS port for HIDPP and my understanding is that, in the case of HIDPP all interaction with the device is blocking, and not asynchronous. That’s why it was unnecessary to create a separate thread to run the runLoop on.

As soon as you want any asynchronous interaction with the device like having a `Device::registerCallback(callback)` function that returns right away, you’d need a separate thread.

Youw, 2021-08-12:
> As soon as you want any asynchronous interaction with the device

and HIDAPI does that [1](https://github.com/libusb/hidapi/blob/8741697359b8dd0d73702a69c54e64ffc552d3cf/mac/hid.c#L809), [2](https://github.com/libusb/hidapi/blob/8741697359b8dd0d73702a69c54e64ffc552d3cf/mac/hid.c#L812)

mcuee, 2021-08-13:
Close this as it is answered.

closed 2021-08-13 by mcuee

noah-nuebling, 2025-03-07:
I think I see what @cvuchener might have meant now.

hidapi uses an [`IOHIDDevice`](https://developer.apple.com/documentation/iokit/iohiddevice_h_user-space), which is mostly a wrapper around [`IOHIDDeviceDeviceInterface`](https://developer.apple.com/documentation/iokit/iohiddevicedeviceinterface) (Source: [IOHIDDevice.c](IOHIDDeviceDeviceInterface))

`IOHIDDeviceDeviceInterface` has a [`getAsyncEventSource()`](https://developer.apple.com/documentation/iokit/iohiddevicedeviceinterface/1395681-getasynceventsource) member function which gives you a [`CFRunLoopSourceRef`](https://developer.apple.com/documentation/corefoundation/cfrunloopsource?language=objc)

This `CFRunLoopSourceRef` is most likely a wrapper around a mach port. If so, it should be possible to extract the mach port by first calling [`CFRunLoopSourceGetContext()`](https://developer.apple.com/documentation/corefoundation/1541961-cfrunloopsourcegetcontext?language=objc) to get a [`CFRunLoopSourceContext1`](https://developer.apple.com/documentation/corefoundation/cfrunloopsourcecontext1?language=objc) struct, and then calling the [`getPort()`](https://developer.apple.com/documentation/corefoundation/cfrunloopsourcecontext1/1542846-getport?language=objc) member function of that struct to retrieve a `mach_port_t`

As far as I know, mach ports are sort of like a message queue for messages that a process received from the kernel.

With the way hidapi currently works, this message queue is only looked at by the `IOHIDDevice`s runLoop. But if you directly look at the message queue, perhaps you could get the device's response faster and avoid having to create runLoops and report queues at all, similar to the Linux implementation.

But this seems sort of hacky, and not the intended way to use these APIs. So I'd say it's probably better to just stick with the current solution, as it seems to work well.

The only good reason I could see to do it this way is perhaps for performance optimization.

noah-nuebling, 2025-03-07:
Another option might be to only run the runLoop temporarily inside `hid_read()`

If the the mach port is really a message *queue* then the runLoop doesn't have to be running for the kernel to add new messages to the port. You could then run the runLoop after the fact to clear any of the enqueued messages.

I'm not sure if this works, but it should be worth a test, and it's probably a bit easier to implement than extracting the `mach_port_t` and reading from it directly.

Youw, 2025-03-08:
> aside from simplifying/unifying the code between platforms?

Mostly this.

> Based on that, I assume that queued up messages are never automatically deleted. So that's good.

Doesn't look like a good thing... At least because it doesn't match the behavior on other platforms. And because it might not make sense, specially for casses when user code doesn't read every input report, but only sometime, depending on some external conditions.

noah-nuebling, 2025-03-09:
I did some testing in the [cvuchener/hidpp](https://github.com/cvuchener/hidpp) macOS backend which I wrote a few years ago. 

The backed I wrote used a separate thread for listening to input reports, just like the hidapi backend.

I removed the separate thread, instead simply running the runLoop for a single iteration in the `readReport()` function, just like we discussed above.

After the refactor, I tested cvucheners `hidpp-list-devices` and `hidpp-list-features` tools and they both still seem to work correctly with my MX Master!

I also tried adding a 500ms sleep inside `readReport()`, to test whether it's actually returning old reports that have been queued up by the mach port, or whether we just happen to catch all the reports only because the runLoop is always started *before* the device can respond – And it looks like we *are* getting old, queued up reports from the mach port!

When you click a bunch of buttons while `hidpp-list-features` reads the features, the tool will still work, it will only delay the next output by 500ms every time you click a button. This strongly suggests that more than one report can be queued up by the mach port, and they are all returned in order – one for every time that `readReport()` is called.

So this seems very promising.

You can check out the code [here](https://github.com/noah-nuebling/hidpp/commit/47e78c3ebdba4ed8c1a8289b621f0d38d29f249e) in case your'e interested.

> Doesn't look like a good thing... At least because it doesn't match the behavior on other platforms.

Interesting point, yes the current macOS backend limits the queued up reports [to 30](https://github.com/libusb/hidapi/blob/122ecb023a129d70eac57a41bc93e57ca43fdb8a/mac/hid.c#L908).

Youw, 2025-03-10:
> Interesting point, yes the current macOS backend limits the queued up reports [to 30](https://github.com/libusb/hidapi/blob/122ecb023a129d70eac57a41bc93e57ca43fdb8a/mac/hid.c#L908).

So does [Windows](https://github.com/libusb/hidapi/blob/122ecb023a129d70eac57a41bc93e57ca43fdb8a/windows/hid.c#L1014C7-L1014C30) backend (even though the constant is different - but it is explicit).

So does [libusb](https://github.com/libusb/hidapi/blob/122ecb023a129d70eac57a41bc93e57ca43fdb8a/libusb/hid.c#L985) backend.

It is unclear to me what hidraw backend does - HIDAPI don't explicitly set the queue size (if there is one), so it behaves according to the default behavior of kernel/driver/etc.