libusb/hidapi #325: hid_enumerate method issue on windows

JunzheFan, 2021-08-18:
Hi, I am not sure this hid_enumerate function can't find one of my HID devices somehow under Window platform, then I transferred the exact same code to my Linux machine using Hidapi backend, this device successfully got detected, any idea why?

Youw, 2021-08-18:
Imagine yourself on my place here and read your own message:

---

"It works on Linux, but doesn't work on Windows. Help."

---

Do you think you've provided enough information for anyone to help you?

JunzheFan, 2021-08-18:
@Youw sry bro, my bad, I just tried using libusb of windows version to open my device, it still can't be detected.

here is the code I used for both libusb and hidapi:

 `hid_init(); `
     `hid_enumerate(0x0D8C,0x0018); `
/////////////////////////////////////////////
     `int rc; `
    ` rc=libusb_init(nullptr); `
     `qDebug()<<rc; `
     `libusb_device_handle* ptr= nullptr; `
     `ptr=libusb_open_device_with_vid_pid(nullptr,0x0D8C,0x0018); `
     `if(ptr){
        qDebug()<<"found"; }`

here is my usb device info:

        ------------------ Device Descriptor ------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x110 (USB Version 1.10)
bDeviceClass             : 0x00 (defined by the interface descriptors)
bDeviceSubClass          : 0x00
bDeviceProtocol          : 0x00
bMaxPacketSize0          : 0x10 (16 bytes)
idVendor                 : 0x0D8C (C-MEDIA ELECTRONICS INC.)
idProduct                : 0x0018
bcdDevice                : 0x0000
iManufacturer            : 0x01 (String Descriptor 1)
 Language 0x0409         : "C-Media"
iProduct                 : 0x02 (String Descriptor 2)
 Language 0x0409         : "USB Device"
iSerialNumber            : 0x00 (No String Descriptor)
bNumConfigurations       : 0x01

        ------------------ String Descriptors -----------------
             ------ String Descriptor 0 ------
bLength                  : 0x04 (4 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language ID[0]           : 0x0409 (English - United States)
             ------ String Descriptor 1 ------
bLength                  : 0x10 (16 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "C-Media"
             ------ String Descriptor 2 ------
bLength                  : 0x16 (22 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "USB Device"

      ---------------- Configuration Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x02 (Configuration Descriptor)
wTotalLength             : 0x0022 (34 bytes)
bNumInterfaces           : 0x01
bConfigurationValue      : 0x01
iConfiguration           : 0x00 (No String Descriptor)
bmAttributes             : 0x80
 D7: Reserved, set 1     : 0x01
 D6: Self Powered        : 0x00 (no)
 D5: Remote Wakeup       : 0x00 (no)
 D4..0: Reserved, set 0  : 0x00
MaxPower                 : 0x32 (100 mA)

        ---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x00
bAlternateSetting        : 0x00
bNumEndpoints            : 0x01 (1 Endpoint)
bInterfaceClass          : 0x03 (HID - Human Interface Device)
bInterfaceSubClass       : 0x00 (None)
bInterfaceProtocol       : 0x00 (None)
iInterface               : 0x00 (No String Descriptor)

        ------------------- HID Descriptor --------------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x21 (HID Descriptor)
bcdHID                   : 0x0100 (HID Version 1.00)
bCountryCode             : 0x00 (00 = not localized)
bNumDescriptors          : 0x01
Descriptor 1:
bDescriptorType          : 0x22 (Class=Report)
wDescriptorLength        : 0x001A (26 bytes)
Error reading descriptor : ERROR_INVALID_PARAMETER

        ----------------- Endpoint Descriptor -----------------
bLength                  : 0x07 (7 bytes)
bDescriptorType          : 0x05 (Endpoint Descriptor)
bEndpointAddress         : 0x87 (Direction=IN EndpointID=7)
bmAttributes             : 0x03 (TransferType=Interrupt)
wMaxPacketSize           : 0x0010 (16 bytes)
bInterval                : 0x01 (1 ms)

JunzheFan, 2021-08-18:
and I am not a advanced user of dealing with hid devices, can you instruct what I should do for debugging if you need more information?thanks

Youw, 2021-08-18:
Firstly - you definitely don't need to use `libusb` for your HID devices. That's what HIDAPI for.

> hid_init();
auto devices = hid_enumerate(0x0D8C,0x0018);

Does this works on Linux? With HIDRAW or LIBUSB backed of HIDAPI?

---
Does your "Device Manager" on Windows shows your HID device in the list?
[image]
(Excuse me for my Ukrainian sample. I don't have an English localisation on Windows right now).

---
Does the output of `hidtest` tool contains your device in the list?

JunzheFan, 2021-08-18:
@Youw 

- Does this works on Linux? With HIDRAW or LIBUSB backed of HIDAPI?
        Ans: Yes, I only tried with libusb backend and device can be found 

- Does your "Device Manager" on Windows shows your HID device in the list?
      Ans: I tried un-plugging and re-plugging my device, the device tree is the same which is weired, does this imply it's not a HID device? But I used the authentic "usbTreeViewTool" and I can see the device successfully got listed there, and according to the interface descriptor shown above, it says  it belongs to HID class 

-Does the output of hidtest tool contains your device in the list?
  Yes . I downloaded the zip and run the .exe in the zip via command prompt, the device got detected. 
  here is the output :
E:\libusb>hidtest-libusb.exe
hidapi test/example tool. Compiled with hidapi version 0.11.0, runtime version 0.10.1.
Compile-time version is different than runtime version of hidapi.
]nDevice Found
  type: 0d8c 0018
  path: 0002:0030:00
  serial_number: (null)
  Manufacturer: (null)
  Product:      (null)
  Release:      0
  Interface:    0
  Usage (page): 0x0 (0x0)

Device Found
  type: 30fa 0400
  path: 0002:002b:00
  serial_number: (null)
  Manufacturer: (null)
  Product:      USB OPTICAL MOUSE
  Release:      100
  Interface:    0
  Usage (page): 0x0 (0x0)

unable to open device

Youw, 2021-08-18:
> Yes . ... the device got detected.

> /******my device *******/

So you have a working [sample](https://github.com/libusb/hidapi/blob/master/hidtest/test.c) of the app that successfully enumerates your device.

I can only suggest to compare the implementation or try using debugger and go by-step to figure out the step where it fails.

JunzheFan, 2021-08-18:
@Youw I wrote a simple separated script using the link u provide, but why I am getting different list of output without un-plugging any of my device :
```
#include "hidapi.h"
#include <iostream>
using namespace std;

int main(){

   hid_init();

   auto ptr=hid_enumerate(0x0, 0x0);
   while(ptr){
       printf("Device Found\n  type: %04hx %04hx\n  path: %s\n  serial_number: %ls", ptr->vendor_id, ptr->product_id,    ptr->path, ptr->serial_number);
       printf("\n");
       printf("  Manufacturer: %ls\n", ptr->manufacturer_string);
       printf("  Product:      %ls\n", ptr->product_string);
       printf("  Release:      %hx\n", ptr->release_number);
       printf("  Interface:    %d\n",  ptr->interface_number);
       printf("  Usage (page): 0x%hx (0x%hx)\n", ptr->usage, ptr->usage_page);
       printf("\n");
       ptr = ptr->next;

   }

}
'''

```
output :

Device Found
  type: 06cb 000f
  path: \\?\hid#synhidmini&col01#1&b12c6d1&a&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer: Synaptics Inc.
  Product:      HID Miniport Device
  Release:      102
  Interface:    -1
  Usage (page): 0x2 (0x1)

Device Found
  type: 06cb 000f
  path: \\?\hid#synhidmini&col02#1&b12c6d1&a&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer: Synaptics Inc.
  Product:      HID Miniport Device
  Release:      102
  Interface:    -1
  Usage (page): 0x5 (0xd)

Device Found
  type: 06cb 000f
  path: \\?\hid#synhidmini&col03#1&b12c6d1&a&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer: Synaptics Inc.
  Product:      HID Miniport Device
  Release:      102
  Interface:    -1
  Usage (page): 0xe (0xd)

Device Found
  type: 06cb 000f
  path: \\?\hid#synhidmini&col04#1&b12c6d1&a&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer: Synaptics Inc.
  Product:      HID Miniport Device
  Release:      102
  Interface:    -1
  Usage (page): 0x1 (0xff00)

Device Found
  type: 30fa 0400
  path: \\?\hid#vid_30fa&pid_0400#6&27bb2781&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer:
  Product:      USB OPTICAL MOUSE
  Release:      100
  Interface:    -1
  Usage (page): 0x2 (0x1)

Process finished with exit code 0
```

JunzheFan, 2021-08-18:
does this mean there is a conflict between that Synaptics touchpad driver and my HID device that HIDAPI can resolve ?

Youw, 2021-08-18:
That is a possible reason.

HIDAPI doesn't handle any drivers on Windows. It just uses WinAPI, and it is up to Windows runtime to provide the list of the devices and its capabilities.

A fautly 3rdparty driver can affect devices it is not directly related to, and in such cases there is practically nothing a user application can do about it.

JunzheFan, 2021-08-19:
@Youw  ok I see . And I am wondering what is the back end of this https://github.com/libusb/hidapi/files/6966661/hidapi_libusb2.zip test .exe using ?  is it possible I can use libusb as back-end instead of winAPI on Windows ? If so how can I specify it in CMAKE script , thanks.

Youw, 2021-08-23:
> is it possible I can use libusb as back-end instead of winAPI on Windows ?

It is theoretically possible, but you'd have to replace a standard HID driver for you device with a WinUSB or UsbDk driver so the device will be available via libusb.

mcuee, 2021-08-25:
> @Youw ok I see . And I am wondering what is the back end of this https://github.com/libusb/hidapi/files/6966661/hidapi_libusb2.zip test .exe using ? is it possible I can use libusb as back-end instead of winAPI on Windows ? If so how can I specify it in CMAKE script , thanks.

That file is just a quick hack to have the HIDAPI libusb backend "working" under Windows. If you did not replace the driver, it still use the Windows HID driver and native Windows HID API. If you replace the HID driver with WinUSB, then it will use the WinUSB backend of libusb.

It is interesting that you can find your device with the hack libusb backend.

mcuee, 2021-08-25:
@JunzheFan Can you post your Windows Device Manager screenshot about your device? And post the output of USBTreeView. Thanks.

And you can run the "hidtest.exe" from here (or you can build your own from latest git head source codes) and post the output?
https://github.com/mcuee/hidapi/releases/tag/V27July2021 (https://github.com/mcuee/hidapi/releases/download/V27July2021/hidapi_win_bin.zip)

mcuee, 2021-08-25:
I have encountered issues with some applications using Windows HID API (Microchip PICKit 2 with the old PC Application using C#)  since the application seems to have some conflict with Microsoft/Logitech mouse, but I have no issues with hidapi.

Ref: https://www.microchip.com/forums/m1146140.aspx

mcuee, 2021-08-25:
BTW, USB Device Tree Viewer will tell you the driver used.
https://www.uwe-sieber.de/usbtreeview_e.html