libusb/hidapi #327: Next release 0.11

mcuee, 2021-08-29:
From the github Projects page, #133 and #146 are planned. Probably they can be delayed.
https://github.com/libusb/hidapi/projects/2

mcuee, 2021-08-29:
#280 is ready.
#309 is more or less ready.
#164 is said to need a bit of work.

mcuee, 2021-08-29:
Reference: https://github.com/trezor/cython-hidapi/issues/116
A new release will help downstream project.

Youw, 2021-09-04:
Good summary, thanks. I'll try to get my hands to it by the end of the week.

Youw, 2021-09-22:
Release draft is ready.
As soon as we check/merge #334 - I'll attach Windows binaries to the release and will fire a v0.11.0,

mcuee, 2021-09-28:
Looks like we are ready now that #334 is merged.

Youw, 2021-09-28:
Done: https://github.com/libusb/hidapi/releases/tag/hidapi-0.11.0

closed 2021-09-28 by Youw