libusb/hidapi #328: Missing one byte feature report read under Windows for devices without report id

mcuee, 2021-08-30:
Reference here:
https://github.com/libusb/hidapi/issues/285

Now I start to question the commit https://github.com/libusb/hidapi/commit/6fcb0bb2282dbb505561a53689eba0c0536570d9 which I proposed in #232 with regard to feature report. It seems to be correct when the report id is not zero. However, it seems to report one less byte when there is no report id. It seems to me for Windows, we may need to report one more byte for reading of feature report when the report id is zero.

mcuee, 2021-08-30:
https://github.com/libusb/hidapi/issues/285#issuecomment-907570047

Just tried hidapitester with hidapi-0.10.1 release and hidapi git head, with regard to input report, the result is the same since it is using hid_read_timeout(). So #232 does not matter in that case.
https://github.com/todbot/hidapitester/blob/master/hidapitester.c#L379

With regard to feature report, then the result will be different as it uses hid_get_feature_report(). So commit https://github.com/libusb/hidapi/commit/6fcb0bb2282dbb505561a53689eba0c0536570d9 does matter for getting the feature report. In this particular case without report id (report id zero), it seems to me the commit is not correct.
https://github.com/todbot/hidapitester/blob/master/hidapitester.c#L403

```
$ ./hidapitester_release0101.exe --vidpid=0925:7001 --open --send-feature 0,0x40,0x41 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 0...read 3 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device

$ ./hidapitester_git4d63a50.exe --vidpid=0925:7001 --open --send-feature 0,0x40,0x41 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 0...read 2 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
```

Youw, 2021-09-04:
Do you think we need to do the same as we do for [macOS](https://github.com/libusb/hidapi/blob/4d63a5085ac6e6a5be6f0f29a42a4e739da30881/mac/hid.c#L940)?

mcuee, 2021-09-05:
> Do you think we need to do the same as we do for [macOS](https://github.com/libusb/hidapi/blob/4d63a5085ac6e6a5be6f0f29a42a4e739da30881/mac/hid.c#L940)?

Yes I think that will fix this issue.

Youw, 2021-09-22:
I think this one is worth fixing before releasing v0.11.0

Youw, 2021-09-22:
After reading the documentation for `IOCTL_HID_GET_INPUT_REPORT` and `IOCTL_HID_GET_FEATURE` one more time, combined with your test results, I believe `input_report` is as broken as `feature_report`

mcuee, 2021-09-23:
> After reading the documentation for `IOCTL_HID_GET_INPUT_REPORT` and `IOCTL_HID_GET_FEATURE` one more time, combined with your test results, I believe `input_report` is as broken as `feature_report`

Let me test out #334 over the weekend.

mcuee, 2021-09-27:
Firstly for feature report for device without report ID.

I think#334 is correct since it reports reading 3 bytes.

```
$ ./hidapitester_release0101.exe --vidpid=0925:7001 --open --send-feature 0,0x40,0x41 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 0...read 3 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device

$ ./hidapitester_hidapi_master.exe --vidpid=0925:7001 --open --send-feature 0,0x40,0x41 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 0...read 2 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device

$ ./hidapitester_hidapi_fix_feature.exe --vidpid=0925:7001 --open --send-feature 0,0x40,0x41 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 0...read 3 bytes:
 00 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device

```

mcuee, 2021-09-27:
Secondly for devices with report ID. No regression from the current master. Release 0.10.1 is not correct and it was fixed by https://github.com/libusb/hidapi/commit/6fcb0bb2282dbb505561a53689eba0c0536570d9 .

```
$ ./hidapitester_release0101.exe --vidpid=0925:7001 --open --send-feature 3,0x40,0x41 --read-feature 4
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 03 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 4...read 4 bytes:
 04 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device

$ ./hidapitester_hidapi_master.exe --vidpid=0925:7001 --open --send-feature 3,0x40,0x41 --read-feature 4
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 03 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 4...read 3 bytes:
 04 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device

$ ./hidapitester_hidapi_fix_feature.exe --vidpid=0925:7001 --open --send-feature 3,0x40,0x41 --read-feature 4
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 03 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte feature report, report_id 4...read 3 bytes:
 04 40 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device

```

mcuee, 2021-09-27:
hdiapitester can not be used to test #334 though as it uses hid_read_timeout().
https://github.com/libusb/hidapi/issues/285#issuecomment-907570047

mcuee, 2021-09-27:
> After reading the documentation for `IOCTL_HID_GET_INPUT_REPORT` and `IOCTL_HID_GET_FEATURE` one more time, combined with your test results, I believe `input_report` is as broken as `feature_report`

Even though I can not use hidapitester to test the input report, but I agree with you after reading the documentation one more time.

commit 19e809ef (2021-09-22): Windows: hid_get_feature/input_report return size fix (#334)

- It appears when the numbered reports aren't used,
WinAPI returns size of the report excluding the data[0] which contains 0,
as an indication that numbered reports aren't used;
Explicity count that byte in the result.

Fixes: #328
commit ad360652 (2021-09-22): Windows: hid_get_feature/input_report return size fix (#334)

- It appears when the numbered reports aren't used,
WinAPI returns size of the report excluding the data[0] which contains 0,
as an indication that numbered reports aren't used;
Explicity count that byte in the result.

Fixes: #328
PR #334 (2021-09-27): Windows: hid_get_feature/input_report return size fix
- It appears when the numbered reports aren't used,
WinAPI returns size of the report excluding the data[0] which contains 0,
as an indication that numbered reports aren't used;
Explicity count that byte in the result.

Fixes: #328
commit b923aeda (2021-09-27): Windows: hid_get_feature/input_report return size fix (#334)

- It appears when the numbered reports aren't used,
WinAPI returns size of the report excluding the data[0] which contains 0,
as an indication that numbered reports aren't used;
Explicity count that byte in the result.

Fixes: #328
closed 2021-09-27 by Youw