libusb/hidapi #329: Story: WinRT backend for Windows

Youw, 2021-09-06:
Current Windows backend is WinAPI based.
There is also an alternative - [WinRT API](https://docs.microsoft.com/uk-ua/uwp/api/windows.devices.humaninterfacedevice?view=winrt-20348), and it actually uses a different implementation.

More discussion about it: #282.
Having an alternative backend might be useful in the future. Leaving this ticket to follow up on it if someone ever gets to finish it.
Some attempt to move in this direction:
https://github.com/LeeChunHei/hidapi

A general recommentation is to have this backend implemented in C++, due to way simpler implementation, when it comes to WinRT, compared to C.

Youw, 2023-03-26:
I have had a little "weekend codding" session, and tried to implement the WinRT backend implementation for HIDAPI, and here is the result: https://github.com/libusb/hidapi/compare/winrt

I'm not even sure if it work making a PR into master as of yet, but here are some statements about the implementation:
- it is written in pure C++/WinRT, i.e. it can be use in Windows Store applications (I believe the WinAPI HIDAPI backend uses some APIs that are not available from UWP/Windows Store environment);
- this is an implementation from scratch; no existing code from WinAPI backend is used/copied (I guess `hid_error` looks very familiar, but that's not the point);
- it inherits all of the limitations of the [`Windows.Devices.HumanInterfaceDevice`](https://learn.microsoft.com/uwp/api/windows.devices.humaninterfacedevice?view=winrt-22621) subsystem as of this moment, e.g.:
    - Serial Number is not available;
    - Manufacturer String is not available;
    - Vid/Pid/UsagePage/Usage - available only for devices that can be opened, i.e. not availbale for Keyboars/Mouses/etc. (even though enumerated and "Name" is available);

And I wasn't able to test it intensively, keeping it in the branch for now. Just in case someone would want to play with it.

JoergAtGithub, 2023-03-26:
> * it is written in pure C++/WinRT, i.e. it can be use in Windows Store applications (I believe the WinAPI HIDAPI backend uses some APIs that are not available from UWP/Windows Store environment);

The Microsoft Store is no longer restricted to UWP: https://developer.microsoft.com/en-us/microsoft-store/desktop-apps

Youw, 2023-03-26:
I believe there're still restrictions on WinAPI usage (and HIDAPI/WinAPI doesn't follow them).
I'll search for link with description later.

mcuee, 2023-04-16:
This seems to work pretty well, tested under Windows 11 with VS2022.
```
PS C:\work\libusb\hidapi_winrt> .\build_winrt\hidtest\hidtest.exe
hidapi test/example tool. Compiled with hidapi version 0.14.0, runtime version 0.14.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 045e 0916
  path: \\?\HID#{00001812-0000-1000-8000-00805f9b34fb}_Dev_VID&02045e_PID&0916_REV&0110_f338080e09c4&Col01#9&1937f32c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: f338080e09c4
  Manufacturer: Microsoft
  Product:      BluetoothMouse3600
  Release:      110
  Interface:    -1
  Usage (page): 0x2 (0x1)
  Bus type: 2

  Report Descriptor: (100 bytes)
0x05, 0x01, 0x09, 0x02, 0xa1, 0x01, 0x09, 0x02, 0xa1, 0x02,
0x09, 0x01, 0xa1, 0x00, 0x85, 0x1a, 0x05, 0x09, 0x19, 0x01,
0x29, 0x05, 0x15, 0x00, 0x25, 0x01, 0x75, 0x01, 0x95, 0x05,
0x81, 0x02, 0x75, 0x03, 0x95, 0x01, 0x81, 0x03, 0x05, 0x01,
0x09, 0x30, 0x09, 0x31, 0x16, 0x01, 0x80, 0x26, 0xff, 0x7f,
0x75, 0x10, 0x95, 0x02, 0x81, 0x06, 0x09, 0x00, 0xa1, 0x02,
0x09, 0x38, 0x16, 0x01, 0x80, 0x26, 0xff, 0x7f, 0x75, 0x10,
0x95, 0x01, 0x81, 0x06, 0xc0, 0x09, 0x00, 0xa1, 0x02, 0x05,
0x0c, 0x0a, 0x38, 0x02, 0x16, 0x01, 0x80, 0x26, 0xff, 0x7f,
0x75, 0x10, 0x95, 0x01, 0x81, 0x06, 0xc0, 0xc0, 0xc0, 0xc0,
Device Found
  type: 045e 0916
  path: \\?\HID#{00001812-0000-1000-8000-00805f9b34fb}_Dev_VID&02045e_PID&0916_REV&0110_f338080e09c4&Col02#9&1937f32c&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: f338080e09c4
  Manufacturer: Microsoft
  Product:      BluetoothMouse3600
  Release:      110
  Interface:    -1
  Usage (page): 0x1 (0xc)
  Bus type: 2

  Report Descriptor: (58 bytes)
0x05, 0x0c, 0x09, 0x01, 0xa1, 0x01, 0x85, 0x27, 0x06, 0x00,
0xff, 0x0a, 0x0a, 0xfa, 0x15, 0x00, 0x26, 0xff, 0x00, 0x75,
0x08, 0x95, 0x13, 0x81, 0x02, 0x85, 0x22, 0x0a, 0x0a, 0xfa,
0x15, 0x00, 0x26, 0xff, 0x00, 0x75, 0x08, 0x95, 0x13, 0xb1,
0x02, 0x85, 0x24, 0x0a, 0x0a, 0xfa, 0x15, 0x00, 0x26, 0xff,
0x00, 0x75, 0x08, 0x95, 0x13, 0xb1, 0x02, 0xc0,
Device Found
  type: 045e 0916
  path: \\?\HID#{00001812-0000-1000-8000-00805f9b34fb}_Dev_VID&02045e_PID&0916_REV&0110_f338080e09c4&Col03#9&1937f32c&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: f338080e09c4
  Manufacturer: Microsoft
  Product:      BluetoothMouse3600
  Release:      110
  Interface:    -1
  Usage (page): 0x1 (0xc)
  Bus type: 2

  Report Descriptor: (78 bytes)
0x05, 0x0c, 0x09, 0x01, 0xa1, 0x01, 0x05, 0x01, 0x09, 0x02,
0xa1, 0x02, 0x85, 0x1f, 0x05, 0x0c, 0x0a, 0x38, 0x02, 0x16,
0x01, 0x80, 0x26, 0xff, 0x7f, 0x75, 0x10, 0x95, 0x01, 0x81,
0x06, 0x85, 0x17, 0x06, 0x00, 0xff, 0x0a, 0x0f, 0xff, 0x15,
0x00, 0x25, 0x01, 0x35, 0x01, 0x45, 0x10, 0x75, 0x02, 0x95,
0x02, 0xb1, 0x02, 0x0a, 0x04, 0xff, 0x15, 0x00, 0x25, 0x01,
0x75, 0x01, 0x95, 0x01, 0x35, 0x00, 0x45, 0x00, 0xb1, 0x02,
0x75, 0x03, 0x95, 0x01, 0xb1, 0x03, 0xc0, 0xc0,
Device Found
  type: 8087 0a1e
  path: \\?\HID#INTC816#3&d2322f2&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
[... 323 lines cut ...]
0x41, 0x15, 0x00, 0x26, 0xff, 0x00, 0x75, 0x08, 0x95, 0x0e,
0x91, 0x00, 0x85, 0x21, 0x09, 0x42, 0x15, 0x00, 0x26, 0xff,
0x00, 0x75, 0x08, 0x95, 0x1f, 0x91, 0x00, 0xc0,
Device Found
  type: 046d c52b
  path: \\?\HID#VID_046D&PID_C52B&MI_00#7&34f0fd76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\KBD
  serial_number:
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      1203
  Interface:    0
  Usage (page): 0x6 (0x1)
  Bus type: 1

  Report Descriptor: (66 bytes)
0x05, 0x01, 0x09, 0x06, 0xa1, 0x01, 0x05, 0x07, 0x19, 0xe0,
0x29, 0xe7, 0x15, 0x00, 0x25, 0x01, 0x75, 0x01, 0x95, 0x08,
0x81, 0x02, 0x75, 0x08, 0x95, 0x01, 0x81, 0x03, 0x19, 0x00,
0x29, 0xff, 0x15, 0x00, 0x26, 0xff, 0x00, 0x75, 0x08, 0x95,
0x06, 0x81, 0x00, 0x05, 0x08, 0x19, 0x01, 0x29, 0x05, 0x15,
0x00, 0x25, 0x01, 0x75, 0x01, 0x95, 0x05, 0x91, 0x02, 0x75,
0x03, 0x95, 0x01, 0x91, 0x03, 0xc0,
unable to open device

```

Youw, 2023-04-16:
> This seems to work pretty well, tested under Windows 11 with VS2022.

I do not believe you've actually used the winrt backend.
You need to pass `-DHIDAPI_WITH_WINRT=ON` and run the `hidtest-winrt` executable.

mcuee, 2023-04-17:
Here is the output.

```
**********************************************************************
** Visual Studio 2022 Developer PowerShell v17.5.4
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
PS C:\Program Files\Microsoft Visual Studio\2022\Community> cd C:\work\
PS C:\work> cd .\libusb\
PS C:\work\libusb> cd .\hidapi_winrt\
PS C:\work\libusb\hidapi_winrt> cmake -B build_winrt -D HIDAPI_BUILD_HIDTEST=1 -D HIDAPI_WITH_TESTS=1 -D HIDAPI_WITH_WINRT=1
-- Building for: Ninja
-- The C compiler identification is MSVC 19.35.32217.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.35.32215/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- hidapi: v0.14.0
-- The CXX compiler identification is MSVC 19.35.32217.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.35.32215/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building hidtest
-- Configuring done
-- Generating done
-- Build files have been written to: C:/work/libusb/hidapi_winrt/build_winrt
PS C:\work\libusb\hidapi_winrt> cmake --build build_winrt
[13/13] Linking C executable hidtest\hidtest-winrt.exe
PS C:\work\libusb\hidapi_winrt> .\build_winrt\hidtest\hidtest-winrt.exe
hidapi test/example tool. Compiled with hidapi version 0.14.0, runtime version 0.14.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 0000 0000
  path: \\?\HID#VID_046D&PID_C52B&MI_00#7&34f0fd76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\KBD
  serial_number: (null)
  Manufacturer: (null)
  Product:      Logitech« Unifying Receiver
  Release:      0
  Interface:    -1
  Usage (page): 0x0 (0x0)
  Bus type: 0

  Report Descriptor: Unable to open device by path
Device Found
  type: 046d c52b
  path: \\?\HID#VID_046D&PID_C52B&MI_02&Col03#7&12bd7e0e&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: (null)
  Product:      Logitech« Unifying Receiver
  Release:      1203
  Interface:    -1
  Usage (page): 0x4 (0xff00)
  Bus type: 0

  Report Descriptor: error getting: HID Report reconstruction is not implemnted for WinRT backend
Device Found
  type: 046d c52b
  path: \\?\HID#VID_046D&PID_C52B&MI_02&Col02#7&12bd7e0e&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: (null)
  Product:      Logitech« Unifying Receiver
  Release:      1203
  Interface:    -1
  Usage (page): 0x2 (0xff00)
  Bus type: 0

[... 182 lines cut ...]
  type: 045e 0916
  path: \\?\HID#{00001812-0000-1000-8000-00805f9b34fb}_Dev_VID&02045e_PID&0916_REV&0110_f338080e09c4&Col02#9&1937f32c&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: (null)
  Product:      Bluetooth Mobile Mouse 3600
  Release:      110
  Interface:    -1
  Usage (page): 0x1 (0xc)
  Bus type: 0

  Report Descriptor: error getting: HID Report reconstruction is not implemnted for WinRT backend
Device Found
  type: 0000 0000
  path: \\?\HID#{00001812-0000-1000-8000-00805f9b34fb}_Dev_VID&02045e_PID&0916_REV&0110_f338080e09c4&Col01#9&1937f32c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: (null)
  Product:      Bluetooth Mobile Mouse 3600
  Release:      0
  Interface:    -1
  Usage (page): 0x0 (0x0)
  Bus type: 0

  Report Descriptor: Unable to open device by path
unable to open device
```

mcuee, 2023-04-17:
Output from the normal Windows backend as a comparison.

```
PS C:\work\libusb\hidapi_winrt> .\build_winrt\hidtest\hidtest.exe
hidapi test/example tool. Compiled with hidapi version 0.14.0, runtime version 0.14.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 045e 0916
  path: \\?\HID#{00001812-0000-1000-8000-00805f9b34fb}_Dev_VID&02045e_PID&0916_REV&0110_f338080e09c4&Col01#9&1937f32c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: f338080e09c4
  Manufacturer: Microsoft
  Product:      BluetoothMouse3600
  Release:      110
  Interface:    -1
  Usage (page): 0x2 (0x1)
  Bus type: 2

  Report Descriptor: (100 bytes)
0x05, 0x01, 0x09, 0x02, 0xa1, 0x01, 0x09, 0x02, 0xa1, 0x02,
0x09, 0x01, 0xa1, 0x00, 0x85, 0x1a, 0x05, 0x09, 0x19, 0x01,
0x29, 0x05, 0x15, 0x00, 0x25, 0x01, 0x75, 0x01, 0x95, 0x05,
0x81, 0x02, 0x75, 0x03, 0x95, 0x01, 0x81, 0x03, 0x05, 0x01,
0x09, 0x30, 0x09, 0x31, 0x16, 0x01, 0x80, 0x26, 0xff, 0x7f,
0x75, 0x10, 0x95, 0x02, 0x81, 0x06, 0x09, 0x00, 0xa1, 0x02,
0x09, 0x38, 0x16, 0x01, 0x80, 0x26, 0xff, 0x7f, 0x75, 0x10,
0x95, 0x01, 0x81, 0x06, 0xc0, 0x09, 0x00, 0xa1, 0x02, 0x05,
0x0c, 0x0a, 0x38, 0x02, 0x16, 0x01, 0x80, 0x26, 0xff, 0x7f,
0x75, 0x10, 0x95, 0x01, 0x81, 0x06, 0xc0, 0xc0, 0xc0, 0xc0,
Device Found
  type: 045e 0916
  path: \\?\HID#{00001812-0000-1000-8000-00805f9b34fb}_Dev_VID&02045e_PID&0916_REV&0110_f338080e09c4&Col02#9&1937f32c&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: f338080e09c4
  Manufacturer: Microsoft
  Product:      BluetoothMouse3600
  Release:      110
  Interface:    -1
  Usage (page): 0x1 (0xc)
  Bus type: 2

  Report Descriptor: (58 bytes)
0x05, 0x0c, 0x09, 0x01, 0xa1, 0x01, 0x85, 0x27, 0x06, 0x00,
0xff, 0x0a, 0x0a, 0xfa, 0x15, 0x00, 0x26, 0xff, 0x00, 0x75,
0x08, 0x95, 0x13, 0x81, 0x02, 0x85, 0x22, 0x0a, 0x0a, 0xfa,
0x15, 0x00, 0x26, 0xff, 0x00, 0x75, 0x08, 0x95, 0x13, 0xb1,
0x02, 0x85, 0x24, 0x0a, 0x0a, 0xfa, 0x15, 0x00, 0x26, 0xff,
0x00, 0x75, 0x08, 0x95, 0x13, 0xb1, 0x02, 0xc0,
Device Found
  type: 045e 0916
  path: \\?\HID#{00001812-0000-1000-8000-00805f9b34fb}_Dev_VID&02045e_PID&0916_REV&0110_f338080e09c4&Col03#9&1937f32c&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: f338080e09c4
  Manufacturer: Microsoft
  Product:      BluetoothMouse3600
  Release:      110
  Interface:    -1
  Usage (page): 0x1 (0xc)
  Bus type: 2

  Report Descriptor: (78 bytes)
0x05, 0x0c, 0x09, 0x01, 0xa1, 0x01, 0x05, 0x01, 0x09, 0x02,
0xa1, 0x02, 0x85, 0x1f, 0x05, 0x0c, 0x0a, 0x38, 0x02, 0x16,
0x01, 0x80, 0x26, 0xff, 0x7f, 0x75, 0x10, 0x95, 0x01, 0x81,
0x06, 0x85, 0x17, 0x06, 0x00, 0xff, 0x0a, 0x0f, 0xff, 0x15,
0x00, 0x25, 0x01, 0x35, 0x01, 0x45, 0x10, 0x75, 0x02, 0x95,
0x02, 0xb1, 0x02, 0x0a, 0x04, 0xff, 0x15, 0x00, 0x25, 0x01,
0x75, 0x01, 0x95, 0x01, 0x35, 0x00, 0x45, 0x00, 0xb1, 0x02,
0x75, 0x03, 0x95, 0x01, 0xb1, 0x03, 0xc0, 0xc0,
Device Found
  type: 8087 0a1e
  path: \\?\HID#INTC816#3&d2322f2&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
[... 364 lines cut ...]
Serial Number String: (0)
  Report Descriptor: (36 bytes)
0x06, 0x00, 0xff, 0x09, 0x01, 0xa1, 0x01, 0x19, 0x01, 0x29,
0x40, 0x15, 0x01, 0x25, 0x40, 0x75, 0x08, 0x95, 0x40, 0x81,
0x00, 0x19, 0x01, 0x29, 0x40, 0x15, 0x01, 0x25, 0x40, 0x75,
0x08, 0x95, 0x40, 0x91, 0x00, 0xc0,
Device Found
  type: 04d8 003f
  path: \\?\HID#VID_04D8&PID_003F#7&aebbb81&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number:
  Manufacturer: Microchip Technology Inc.
  Product:      Simple HID Device Demo
  Release:      2
  Interface:    0
  Usage (page): 0x1 (0xff00)
  Bus type: 1

Indexed String 1: Microchip Technology Inc.
Unable to send a feature report.
Unable to get a feature report: Get Input/Feature Report DeviceIoControl: (0x00000001) Incorrect function.
waiting...
Data read:
   81 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Press any key to continue . . .
```

mcuee, 2023-04-17:
Another computer using Windows 10.

<details>
<summary> click for the run log for WinRT backend </summary>

```
PS C:\work\libusb\hidapi_winrt> .\build_winrt\hidtest\hidtest-winrt.exe
hidapi test/example tool. Compiled with hidapi version 0.14.0, runtime version 0.14.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 0000 0000
  path: \\?\HID#ConvertedDevice&Col01#5&379854aa&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\KBD
  serial_number: (null)
  Manufacturer: (null)
  Product:      1TV4433
  Release:      0
  Interface:    -1
  Usage (page): 0x0 (0x0)
  Bus type: 0

  Report Descriptor: Unable to open device by path
Device Found
  type: 413c b06f
  path: \\?\HID#VID_413C&PID_B06F#d&3624b04c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: (null)
  Product:      Dell dock
  Release:      101
  Interface:    -1
  Usage (page): 0xda (0xffda)
  Bus type: 0

  Report Descriptor: error getting: HID Report reconstruction is not implemnted for WinRT backend
Device Found
  type: 2717 5013
  path: \\?\HID#VID_2717&PID_5013&Col06#6&2a21f7ec&0&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: (null)
  Product:      Mi Wireless Mouse
  Release:      625
  Interface:    -1
  Usage (page): 0x2 (0xff02)
  Bus type: 0

  Report Descriptor: error getting: HID Report reconstruction is not implemnted for WinRT backend
Device Found
  type: 0000 0000
  path: \\?\HID#VID_24AE&PID_4057&MI_01&Col04#e&39a3ff8c&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\KBD
  serial_number: (null)
  Manufacturer: (null)
  Product:      USB Keyboard
  Release:      0
  Interface:    -1
  Usage (page): 0x0 (0x0)
  Bus type: 0

  Report Descriptor: Unable to open device by path
Device Found
  type: 0488 121f
  path: \\?\HID#DELL091A&Col05#5&99b72d3&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: (null)
  Product:      1TV4433
  Release:      40d
  Interface:    -1
  Usage (page): 0xe (0xd)
  Bus type: 0

  Report Descriptor: error getting: HID Report reconstruction is not implemnted for WinRT backend
[... 932 lines cut ...]
0x06, 0xda, 0xff, 0x09, 0xda, 0xa1, 0x01, 0x09, 0xda, 0x15,
0x81, 0x25, 0x7f, 0x75, 0x08, 0x95, 0x01, 0x81, 0x02, 0x19,
0x01, 0x29, 0x0e, 0x15, 0x81, 0x25, 0x7f, 0x75, 0x08, 0x95,
0xbf, 0x81, 0x02, 0x05, 0xda, 0x19, 0x01, 0x29, 0x05, 0x15,
0x81, 0x25, 0x7f, 0x75, 0x08, 0x95, 0xc0, 0x91, 0x02, 0xc0,
Device Found
  type: 045e 0000
  path: \\?\HID#ConvertedDevice&Col01#5&379854aa&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\KBD
  serial_number:
  Manufacturer:
  Product:
  Release:      0
  Interface:    -1
  Usage (page): 0x6 (0x1)
  Bus type: 0

  Report Descriptor: (37 bytes)
0x05, 0x01, 0x09, 0x06, 0xa1, 0x01, 0x85, 0x01, 0x05, 0x07,
0x09, 0x69, 0x09, 0x6a, 0x09, 0x6b, 0x09, 0x6c, 0x09, 0xe3,
0x09, 0x4c, 0x09, 0xe2, 0x09, 0xe0, 0x15, 0x00, 0x25, 0x01,
0x75, 0x01, 0x95, 0x08, 0x81, 0x02, 0xc0,
unable to open device
```

</details>

Youw, 2023-04-17:
Looks about right - consistent with my testing.
All of what is missing - are the limitation of the WinRT API for HID devices.