libusb/hidapi #339: Expose information on which usage-page/usage maps to which report id

dnaq, 2021-09-30:
This is somewhat related to #249 but should be much less of a hassle to implement.

Today when enumerating hid-devices hidapi enumerates all toplevel usage-page/usage pairs and maps them to the device that supports them. But when opening a device and reading from it we will get all input reports from that device, but we have no way of distinguishing them. That information is however available (in linux the hid descriptor is already parsed to find the usage-page/usage pairs), and I believe windows has API-calls to find the report id as well.

This would help immensely when writing host code that uses hidapi to communicate with a hid-device, when that device can send multiple reports.

dnaq, 2021-09-30:
Agreed #249 would be great, I just didn't know how far off that is in time, that's why I opened this issue. Feel free to close if you think #249 is the way forward.

Youw, 2021-09-30:
> if you think #249 is the way forward

It's not the only way - there are other possibilities, e.g. preparse the descriptor on each platform (on Windows we could use WinAPI directly, as there is no HID Descriptor directly available) and return some structure which would include the list of usage/usage_pages/reports and their properties.

But so far #306 is where the most efforts was done, and we better finish off at least something functional.

closed 2021-09-30 by Youw