libusb/hidapi #347: CMake options and variables

karamellpelle, 2021-10-31:
Thank for creating this library :smile:

I think you should specify in [BUILD.cmake.md](https://github.com/libusb/hidapi/blob/master/BUILD.cmake.md#hidapi-as-a-subdirectory) that users needs CMake 3.13 or later (`cmake_minimum_required(VERSION 3.13)`) in  their _CMakeLists.txt_) if they want to use _hidapi_ as a CMake subproject. Because the lines in your example
```
set(HIDAPI_WITH_LIBUSB FALSE) # surely will be used only on Linux
set(BUILD_SHARED_LIBS FALSE) # HIDAPI as static library on all platforms
add_subdirectory(hidapi)
```
will not work if using CMake 3.12 or earlier. This is because `option()` creates _CACHE_ variables and not normal variables, and the behavior in these earlier versions of CMake is to ignore setting the variable (as a normal variable in _CMakeLists.txt_) if this variable is already set as a _CACHE_ variable[^1]. See [_CMP0077](https://cmake.org/cmake/help/latest/policy/CMP0077.html).

Also, I think you should create an option
```
option(HIDAPI_BUILD_SHARED "Build shared version of the libraries, otherwise build statically" ON)
```
instead of using [`BUILD_SHARED_LIBS`](https://cmake.org/cmake/help/latest/variable/BUILD_SHARED_LIBS.html), because setting `BUILD_SHARED_LIBS` before `add_subdirectory()` will affect my _CMakeLists.txt_ too. 

[^1]: CMake 3.12 and earlier would need `set( HIDAPI_WITH_LIBUSB FALSE CACHE BOOL "" FORCE )`

Youw, 2021-11-10:
> specify in BUILD.cmake.md that users needs CMake 3.13 or later

They do not need to have such high version of CMake. HIDAPI will compile and work just fine with CMake `3.6.3` in worst case (on macOS). On Windows CMake `3.1.3` is enough - I've explicitly tested that.

> This is because option() creates CACHE variables and not normal variables

For starters, if you have a relatively large project, with multiple build targets and/or usages of 3rd-party libraries, I wouldn't expect you to to use `BUILD_SHARED_LIBS` as an `option` variable, because you've alreay showed that you need different behavior for different targets, in which case specifying `STATIC`/`SHARED` for each library explicitly is a good way to ensure the behavior you need.

> Also, I think you should create an option

I though about that too.
But that would mean I need to invent my own mechanism (`HIDAPI_BUILD_SHARED`) for something (`BUILD_SHARED_LIBS`) _CMake already has_. And I'd rather not.

> setting BUILD_SHARED_LIBS before add_subdirectory() will affect my CMakeLists.txt too.

That's expected. And also:

1) You can safely reset the value of `BUILD_SHARED_LIBS` to what it was before _after_ including `hidapi` as a subdirectory;
2) You can achieve 1) automatically by having a subdirectory (e.g. `hidapi_3rdparty`) with a `CMakeLists.txt` with a single purpose: set all the configuration variables for HIDAPI and `add_subdirectory(hidapi)`;

---
But since you've pointed to this as an issue, I'll make a remark and point to `CMP0077` for other users to be aware.

karamellpelle, 2021-11-10:
_hidapi_ will build with lower CMake versions, as you say. But the reason to specify `cmake_minimum_required(VERSION 3.13)` for the **user's _CMakeLists.txt_'s** is because I'm pretty sure that setting options (i.e. CACHE variables) before diving into a subdirectory like
```
set(HIDAPI_WITH_LIBUSB FALSE) # surely will be used only on Linux
set(BUILD_SHARED_LIBS FALSE) # HIDAPI as static library on all platforms
add_subdirectory(hidapi)
```
(as described in [BUILD.cmake.md](https://github.com/libusb/hidapi/blob/master/BUILD.cmake.md#hidapi-as-a-subdirectory)) will have any effect for the subdirectory when CMake is 3.12 or earlier.

I remember I had to do `set( <option> <value> CACHE BOOL "" FORCE )` before `add_directory()` for other projects I've previously been working on, but now you don't need this if [CMake is 3.13 or later](https://cmake.org/cmake/help/latest/policy/CMP0077.html).

Youw, 2021-11-11:
> I remember I had to do set( <option> <value> CACHE BOOL "" FORCE ) before add_directory() for other projects I've previously been working on

I believe you're just misusing a CMake in a first place.

`CACHE` variables are meant to be set once and globally for the entire project.
If you need it to have different values for different targes/subfolders, than _it shouldn't be a CACHE variable_ in a first place.

What you're describing is correct, and you're describing a workaround for an issue you've hit.
The initial suggestion is:
> you should specify in BUILD.cmake.md that users needs CMake 3.13 or later

Which rather should be as:
> If you happen to set `BUILD_SHARED_LIBS` as a CACHE variable, then you need CMake 3.13 for the following to work

But if you happen to set `BUILD_SHARED_LIBS` as a CACHE variable and that's causing issues - than a design decicion has been made wrong already.
And the fact that it is no an issue anymore in a newer versions of CMake - doesn't mean we should encourage people to do so (by showing them a workaround).

closed 2021-11-11 by Youw

karamellpelle, 2021-11-16:
I'm not talking specially about the variable `BUILD_SHARED_LIBS`. I'm talking about using _hidapi_ as a subfolder with CMake: Variables declared as `option()` becomes CACHE variables in CMake v3.12 or earlier[^1] , so `set(HIDAPI_WITH_LIBUSB FALSE)` in a user's _CMakeLists.txt_ before `add_subfolder()` will only work when the user use CMake v3.13 or later. [Here is a relevant answer from a guy on StackOverflow](https://stackoverflow.com/a/62542166/753850).

Anyway, thanks for this nice library 😄 

[^1]: Therefore we had to use `set( <option> <value> CACHE BOOL "" FORCE )`

PR #349 (2021-11-11): Explicit remark about BUILD_SHARED_LIBS
And its behavior, as controled by CMP0077.

Closes: #347