libusb/hidapi #354: Unable to enumerate and open all USB devices

arilink, 2021-11-23:
I tried to use hidapi in the Qt project. I first called hid_enumerate(0x0,0x0) to enumerate all USB devices, but as a result, some USB devices were not enumerated, and only devices with hid_enumerate() can be opened, and other devices cannot Open, why is this?

arilink, 2021-11-23:
In the same way, you can get 21 devices using libusb, but hidapi can only get 16 devices.

Youw, 2021-11-23:
Imagine yourself on my place, read your own statement, and give me some suggestion how am I (or someone else) is going to guess what help exactly do you need?

arilink, 2021-11-23:
Sorry, it was my mistake. libusb can list all USB devices. hidapi can only list hid devices.

arilink, 2021-11-23:
On the windows platform, libusb does not support hot-plug monitoring. Hidapi does not have a related api. How can I obtain device plug-in events?

Youw, 2021-11-23:
using WinAPI

closed 2021-11-23 by Youw