libusb/hidapi #355: Mac Monterrey, MacOSX SDK 12.0 linking fails

gerstrong, 2021-12-06:
Unable to build with this library using MacOS Monterrey and SDK 12.0.

```
Apple clang version 13.0.0 (clang-1300.0.29.3)
Target: x86_64-apple-darwin21.1.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
 "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld" -demangle -lto_library /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/libLTO.dylib -dynamic -arch x86_64 -platform_version macos 10.14.0 12.0 -syslibroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk -o myApp -lc++ -search_paths_first -headerpad_max_install_names -lc++abi -lc++ CMakeFiles/myApp.dir/myApp.c.o -framework Foundation -framework CoreFoundation -framework IOKit -lc++ -lSystem /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/13.0.0/lib/darwin/libclang_rt.osx.a
Undefined symbols for architecture x86_64:
  "_NSAppKitVersionNumber", referenced from:
      _hid_init in libtraffix_hidapi.a(hid.c.o)
```

gerstrong, 2021-12-06:
I take it back. I forgot to add the right framework parameters.

closed 2021-12-06 by gerstrong

Youw, 2021-12-06:
I take it as you aren't using our CMake build system, but have your own.

gerstrong, 2021-12-06:
Yes, that's correct. I have plans to cleanup our code and use your cmake files.