libusb/hidapi #356: Enumeration multiple times does not work

arilink, 2021-12-10:
1.I enumerate the USB devices when the software starts, and at this time, I can get the usb device info normally

Youw, 2021-12-10:
Imagine yourself at my place. Read your own issue description and suggest how possibly could I (or someone else) guess what is  the issue that you have.

mcuee, 2021-12-31:
> 1.I enumerate the USB devices when the software starts, and at this time, I can get the usb device info normally 2.When I detect that the USB device is plugged and unplugged, I try to enumerate again, but the list of USB devices cannot be obtained correctly at this time. Why?

You need to clearly state the questions.
1) What is the device?
2) What is the OS and hidapi version you are using?
3) What are you trying to do? 
4) Share the code snipets.

Maybe reading the following can help you.
http://www.catb.org/~esr/faqs/smart-questions.html

mcuee, 2021-12-31:
Further discussions here.
https://github.com/libusb/hidapi/issues/354

closed 2021-12-31 by mcuee