libusb/hidapi #359: hid_write() on windows: "A device attached to the system is not functioning"

stolk, 2021-12-19:
I have  device (Arduino Pro Micro atmega32u4 acting as raw HID) that I can write to using libhidapi_hidraw under linux.

When I try to do the same under Windows, the hid_write() fails with:
```
Examining arduino devices...
type: 2341 8037
  path: \\?\hid#vid_2341&pid_8037&mi_02#7&20cef2ab&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: C
  Manufacturer: GSAS Inc
  Product:      Turbo LEDz 88s
  Release:      100
  Interface:    2
  Usage (page): 0xc00 (0xffc0)
Detected model: 88s
Found 1 Turbo LEDz devices.
Opened hid device at \\?\hid#vid_2341&pid_8037&mi_02#7&20cef2ab&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Opened 1 devices.
hid_write for 2 bytes failed with: A device attached to the system is not functioning.
```

I tried running as Administrator too, but with the same results.

Opening the device succeeds, using:
```
handle = hid_open_path( filenames[i] );
if (!handle)
{
    fprintf(stderr,"Error: hid_open_path() on %s failed : %ls\n", fname, hid_error(handle));
    exit(EX_IOERR);
}
fprintf(stderr,"Opened hid device at %s\n", fname);
// We like to be blocked.
hid_set_nonblocking(handle, 0);
```

Writing to it, succeeds under Linux, but fails under Windows, using:
```
uint8_t rep[2] = { 0x00, bars | 0x80 };
const int written = hid_write( hd, rep, sizeof(rep) );
if ( written < 0 )
{
    fprintf( stderr, "hid_write for %zu bytes failed with: %ls\n", sizeof(rep), hid_error(hd) );
    cleanup();
    exit(EX_IOERR);
}
```

todbot, 2021-12-19:
What does the HID Report Descriptor look like for your device? If it’s standard “Raw HID” the report sizes are 64 bytes with no reportID. The send buffer must be exactly the same size as the report, us one for the reportID. Windows enforces this, Linux does not. 

Try making your “rep” buffer be 65 bytes.

stolk, 2021-12-19:
Ah, thank you very much! Good to know that Windows and Linux treat this differently. I didn't know that.

My microcontroller indeed does:
```
#define RAWHID_SIZE (USB_EP_SIZE)
```
...which is set to 64, like you said.

However, sending 64 bytes instead, gives me the same error, as does 65 bytes.

The descriptor that I use (single report)
```
static const uint8_t  _hidReportDescriptorRawHID[] PROGMEM = {
	/*    RAW HID */
    0x06, lowByte(RAWHID_USAGE_PAGE), highByte(RAWHID_USAGE_PAGE),      /* 30 */
    0x0A, lowByte(RAWHID_USAGE), highByte(RAWHID_USAGE),

    0xA1, 0x01,                  /* Collection 0x01 */
    // RawHID is not multireport compatible.
    // On Linux it might work with some modifications,
    // however you are not happy to use it like that.
    //0x85, HID_REPORTID_RAWHID,			 /* REPORT_ID */
    0x75, 0x08,                  /* report size = 8 bits */
    0x15, 0x00,                  /* logical minimum = 0 */
    0x26, 0xFF, 0x00,            /* logical maximum = 255 */

    0x95, RAWHID_TX_SIZE,        /* report count TX */
    0x09, 0x01,                  /* usage */
    0x81, 0x02,                  /* Input (array) */

    0x95, RAWHID_RX_SIZE,        /* report count RX */
    0x09, 0x02,                  /* usage */
    0x91, 0x02,                  /* Output (array) */
    0xC0                         /* end collection */ 
};
```

todbot, 2021-12-19:
What is the value of “HID_REPORTID_RAWHID” and “HID_RX_SIZE”?  The reportID and report size must match.

stolk, 2021-12-19:
HID_REPORTID_RAWHID Is not used… it is commented out of the code.

RX and TX sizes are both 64.

The author of arduino rawhid, Nico Hood, warns about multireport, hence he commented it out.

i’ve followed his sample code https://github.com/NicoHood/HID which works fine w a linux host pc.

stolk, 2021-12-19:
I stepped with the debugger into the `hid_write()` function, and it turns out that `GetLastError()` returns `ERROR_IO_PENDING`

Later, `GetOverlappedResult()` returns 0.

This is with the first write after opening the device, so I don't understand that IO could be pending at that point?

stolk, 2021-12-19:
Maybe 64 is an edge-case, because I changed my microcontroller code to use a report size of 8 bytes.

And then have my Windows PC send 9 bytes, which ends up working.

Whereas a report size of 64 and sending 65 would not work for me.

Youw, 2021-12-19:
> I changed my microcontroller code to use a report size of 8 bytes

> And then have my Windows PC send 9 bytes, which ends up working.

This makes perfect sense

stolk, 2021-12-20:
>This makes perfect sense

Including sending 65 bytes for 64 report not working?

Youw, 2021-12-20:
I think we need to add a section in the documentation somewhere:
"Sending data of size that doesn't match the report size as per descriptor - is unspecified behavior".

Strictly speaking HIDAPI doesn't perform any checks (except those that are strictly required by the undelying API).

On Linux with hidraw/libusb backends we simply send all the data to the device as is, and it is up to FW to respond to it well or not.
On Windows we're using WinAPI and it has much more restrictions, e.g.: WinAPI always parses and validates all of the reports.
Apparently in case when you send more data than it is expected - WinAPI doesn't respond to it well.

stolk, 2021-12-20:
I think you misunderstood.

Before, I had the firmware expect a 64 byte report, and using @todbot's hint when I was complying with that by sending 64+1, it did not work. I also tried 64+0.

Only after I changed my FW to expect 8 byte report, and sending 8+1, I got it to work.

So even sending the size that was expected, did not work for me in all cases! It could be I made some other mistake, of course, or there could actually be some hard-coded 64byte special case at the PC or at the Arduino side.

Youw, 2021-12-20:
> So even sending the size that was expected, did not work for me in all cases!

That I cannot know why. Yes, sometime there are limitations, e.g. the max size of a report may be additionally limited if you're using USB2.0 vs USB1.1 etc.

The HIDAPI library is agnostic of those details.

mcuee, 2022-02-11:
> So even sending the size that was expected, did not work for me in all cases! It could be I made some other mistake, of course, or there could actually be some hard-coded 64byte special case at the PC or at the Arduino side.

I tened to think that there is a FW bug. You have to debug the FW. 

You can also try other HID related library to see if works or not. For example, pywinusb is another possible tool under Windows. Or use the Windows API directly if you think HIDAPI is the problem.

pywinusb: https://github.com/rene-aguirre/pywinusb
Windows HID native API examples: http://janaxelson.com/hidpage.htm

mcuee, 2022-02-11:
> i’ve followed his sample code https://github.com/NicoHood/HID which works fine w a linux host pc.

Have you tried using his host side example to see if that works or not.
Ref: https://github.com/NicoHood/HID/tree/master/extras/rawhid

mcuee, 2022-05-04:
Close this for now. Please reopen if there are new data to show there are issues with HIDAPI.

closed 2022-05-04 by mcuee