libusb/hidapi #367: How files from libraries from zip archive shall be used within hid package?

andkirby, 2022-01-10:
I downloaded libraries.

And what's next?

Could you please provide a short manual?

Youw, 2022-01-10:
Next you do the same thing as you'd do with any other 3rdparty shared library: use provided headers during the compilation of your project, and link angainst provided libraries during the link time.

This is a HIDAPI issue tracker - not a support forum. Use platforms like stackoverflow for discussion topics that are not related directly to the HIDAPI library development/bugs (if any).

Besides, there is already some amount of [readme](https://github.com/libusb/hidapi/blob/master/BUILD.cmake.md#using-hidapi-in-a-cmake-project).

closed 2022-01-10 by Youw

andkirby, 2022-01-11:
Got it. Thanks!