libusb/hidapi #371: Mac Privacy/Input Monitoring

meng0609, 2022-01-23:
When I use hidapi, "Mac Privacy/Input Monitoring" pops up, how to not display it,
I use other keyboard software, and I can send data to the keyboard without this prompt.

Youw, 2022-01-23:
HIDAPI cannot be used with keyboards/mouses on macOS and Windows.
HIDAPI is not designed to be used with keyboards/mouses at all.

closed 2022-01-23 by Youw

meng0609, 2022-01-24:
My software sends commands to switch the keyboard light through HIDAPI

On a Mac system, the "Mac Privacy/Input Monitoring" option must be turned on to send data successfully

meng0609, 2022-01-24:
@Youw I need your help

Youw, 2022-01-24:
It is impossible to use a HID device on macOS that is also a Keyboard without explicit "Mac Privacy/Input Monitoring" enablement for the application.
If this is your own keyboard, and you're developing lighting API for it - make a _different USB HID interface_ for controlling lighting, and make sure do not define a Desktop/Keyboard usage/usage_page for that interface.