libusb/hidapi #376: Test matrix for hidapi

mcuee, 2022-02-06:
It seems to me it is a good idea to have the test matrix, similar to the following for libusb-1.0.25rc1.
* https://github.com/libusb/libusb/issues/1046

mcuee, 2022-02-09:
macports seem to provide a good one for macOS support.
Ref: https://ports.macports.org/port/hidapi/details/

From here:
https://ports.macports.org/port/hidapi/details/

- [x] Monterey (arm64)
- [x]  Monterey (x86_64)
- [x]  Big Sur (arm64)
- [x]  Big Sur (x86_64)
- [x]  Catalina
- [x]  Mojave
- [x]  High Sierra
- [x]  Sierra
- [x]  El Capitan
- [x]  Yosemite
- [x]  Mavericks
- [x]  Mountain Lion
- [x]  Lion
- [x]  Snow Leopard (x86_64)
- [x]  Snow Leopard (i386)
- [ ]  Leopard (ppc legacy) -->???

Youw, 2022-02-09:
A question here - what do we test by this?
If just to verify that it builds - Github Actions / Appveyor / builds.sr.ht is an indication as well.

mcuee, 2022-02-09:
The idea is not only to carry out build test, but also test with a few real devices (USB/Bluetooth) using hidtest (hidapitester is another good one) and other applications dependant on hidapi.

Ref: https://github.com/libusb/libusb/issues/1052

It will be good to document some known good devices to carry out the tests.

mcuee, 2022-02-09:
The other thing is to see the operating systems the contributors are able to test: eg: Windows versions, macOS versions, Linux and FreeBSD.

For example, does it make sense to still support Windows XP and Vista? Or old macOS versions? Or very old Linux versions?

Youw, 2022-02-09:
> does it make sense to still support Windows XP and Vista? Or old macOS versions? Or very old Linux versions?

For as long as it doesn't require the actually effort to support those.

Youw, 2022-05-02:
I'm ready to release a `v0.12.0`.

How do we want to proceed with the test matrix for it?

mcuee, 2022-05-03:
@Youw I am okay with either of the following.

1) Just release v0.12.0 as it is. Right now I tend to think hidapi is rather stable now and 

2) Release 0.12.0rc1 first and then wait for two weeks to release v0.12.0 so that people can test hidapi on different operating systems and different devices to see if there are regressions. It would also be good to check the major language bindinges (eg: cython-hidapi and node-hid).

mcuee, 2022-05-03:
Reference: libusb 1.0.26-rc1 test matrix
* https://github.com/libusb/libusb/issues/1099

If we go with this idea I should be able to test under the following system.

### macOS 12 ARM64 (mac Mini M1)
  - [ ] build: Apple clang
  - [ ] run:  hidtest and hidapitester

### macOS 12 x86_64 using Rosetta Translation, not native (mac Mini M1)
  - [ ] build: Apple clang + CMake
  - [ ] run: hidtest and hidapitester
  
### Linux x86_64 (Ubuntu 20.04)
  - [ ] build: gcc 9.4
  - [ ] run: hidtest and hidaptester

### Linux ARM32 (Raspberry Pi 400 running Raspberry Pi OS Bullseye 32 bit)
  - [ ] build: gcc 10.2.1
  - [ ] run: hidtest and hidaptester

### Linux ARM64 (Raspberry Pi 400 running Raspberry Pi OS Bullseye 64 bit)
  - [ ] build: gcc 10.2.1
  - [ ] run: hidtest and hidaptester
 
### Windows VS2019 64 bit (Windows 11 x64 laptop)
  - [ ] build: 
  - [ ] run: hidtest and hidaptester
  
### Windows VS2019 32 bit (Windows 11 x64 laptop)
  - [ ] build: 
  - [ ] run: hidtest and hidaptester
 
### Windows MinGW 64 bit (Windows 11 x64 laptop)
  - [ ] build: MSYS2 MinGW-w64 64bit GCC
  - [ ] run: hidtest and hidaptester
  
### Windows MinGW 32 bit (Windows 11 x64 laptop)
  - [ ] build: MinGW-w64 32 bit GCC
  - [ ] run: hidtest and hidaptester

mcuee, 2022-05-03:
I can not test macOS Intel but I am sure there are people who can test it. Same for FreeBSD.

Then I am probably not good at testing cython-hidapi and node-hid but again other people can test. I will try to test them as well.

Youw, 2022-05-03:
https://github.com/libusb/hidapi/issues/407

mcuee, 2022-05-04:
Close this as #407 has been raised.
* #407

closed 2022-05-04 by mcuee